NVIDIA/open-gpu-kernel-modules #524: Sleep (suspend) wakes up, cannot sleep (suspend)

jyblu, 2023-06-25:
### NVIDIA Open GPU Kernel Modules Version

nvidia-open-dkms 535.54.03-3

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

archlinux kde

### Kernel Release

6.3.9-1-clear-x64-v4、linux-zen 6.3.9.zen1-1

### Hardware: GPU

NVIDIA GeForce RTX 4060 Laptop GPU

### Describe the bug

My device is a laptop with an external monitor connected via hdmi.Sleep (suspend) wakes up, cannot sleep (suspend)，When you click sleep (suspend), it will sleep (suspend) for a while, but it will be woken up again in an instant.like this every time.But if the hdmi is unplugged and no external monitor is connected, it can sleep normally (suspend).

### To Reproduce

My device is a laptop with an external monitor connected via hdmi.Sleep (suspend) wakes up, cannot sleep (suspend)，When you click sleep (suspend), it will sleep (suspend) for a while, but it will be woken up again in an instant.like this every time.But if the hdmi is unplugged and no external monitor is connected, it can sleep normally (suspend).

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11860075/nvidia-bug-report.log.gz)

lobroc, 2023-11-15:
This does not alleviate the issue. With both services enabled, and the same exact system configuration, the proprietary modules allow the device to sleep but not the open ones.

xmarda, 2024-04-27:
I think this could also be a bios vendor (lenovo) issue. In my experience they have poor linux support.
Setting a different/recent ACPI _OSI fixed the sleep issues on two laptops with nvidia dGPU for me:

1. run `sudo strings /sys/firmware/acpi/tables/DSDT | grep -i 'windows ' | sort -u` 
2. add `acpi_osi=! "acpi_osi=xxx"` and insert the most recent for xxx. It could likely fix the sleep issues. [source](https://wiki.archlinux.org/title/NVIDIA/Troubleshooting#System_does_not_return_from_suspend)
---
_ps.: not sure if it does change anything for you, but in my experience the nvidia-{suspend,hibernate,resume}.service don't affect dGPU's but primary GPU's instead. Would be nice to know if this is correct/true for you, too._