NVIDIA/open-gpu-kernel-modules #525: get_user_pages signature changed

ElDavoo, 2023-06-28:
### NVIDIA Open GPU Kernel Modules Version

535.54.03

### Operating System and Version

gentoo

### Kernel Release

77b1a7f7a05c

### Build Command

`make -j15 -l16 ARCH=x86 V=1 KBUILD_VERBOSE=1 CONFIG_WERROR= CONFIG_MODULE_SIG_ALL= CONFIG_MODULE_COMPRESS_GZIP= CONFIG_MODULE_COMPRESS_XZ= CONFIG_MODULE_COMPRESS_ZSTD= DEPMOD=: STRIP=: CROSS_COMPILE=x86_64-pc-linux-gnu- HOSTCC=clang HOSTCXX=clang++ 'HOSTCFLAGS=-march=native -fomit-frame-pointer -pipe -flto -O3 -fno-stack-protector -fno-stack-clash-protection -pipe -falign-functions=32' 'HOSTCXXFLAGS=-march=native -fomit-frame-pointer -pipe -flto -O3 -fno-stack-protector -fno-stack-clash-protection -pipe -falign-functions=32' 'HOSTLDFLAGS=-Wl,-O1 -Wl,--as-needed -fuse-ld=lld -rtlib=compiler-rt -O2 -Wl,--as-needed' HOSTPKG_CONFIG=x86_64-pc-linux-gnu-pkg-config CC=/usr/bin/x86_64-pc-linux-gnu-gcc-12 CXX=/usr/bin/x86_64-pc-linux-gnu-g++-12 LD=/usr/bin/x86_64-pc-linux-gnu-ld.bfd AR=/usr/bin/x86_64-pc-linux-gnu-gcc-ar NM=/usr/bin/x86_64-pc-linux-gnu-gcc-nm OBJCOPY=/usr/bin/x86_64-pc-linux-gnu-objcopy OBJDUMP=/usr/bin/x86_64-pc-linux-gnu-objdump READELF=/usr/bin/x86_64-pc-linux-gnu-readelf IGNORE_CC_MISMATCH=yes NV_VERBOSE=1 SYSOUT=/usr/src/linux SYSSRC=/usr/src/linux modules `

### Terminal output/Build Log

```
 CC [M]  /var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/nvidia/nv-vtophys.o
  /usr/bin/x86_64-pc-linux-gnu-gcc-12 -Wp,-MMD,/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/nvidia/.nv-vtophys.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -funsigned-char -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector-strong -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -fomit-frame-pointer -ftrivial-auto-var-init=zero -fno-stack-clash-protection -falign-functions=16 -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -I/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/common/inc -I/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open -Wall -Wno-cast-qual -Wno-error -Wno-format-extra-args -D__KERNEL__ -DMODULE -DNVRM -DNV_VERSION_STRING=\"535.54.03\" -Wno-unused-function -Wuninitialized -fno-strict-aliasing -ffreestanding -DNV_UVM_ENABLE -Werror=undef -DNV_SPECTRE_V2=0 -DNV_KERNEL_INTERFACE_LAYER -I/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/nvidia -DNVIDIA_UNDEF_LEGACY_BIT_MACROS -UDEBUG -U_DEBUG -DNDEBUG  -DMODULE  -DKBUILD_BASENAME='"nv_vtophys"' -DKBUILD_MODNAME='"nvidia"' -D__KBUILD_MODNAME=kmod_nvidia -c -o /var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/nvidia/nv-vtophys.o /var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/nvidia/nv-vtophys.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --orc --static-call --uaccess   --module /var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/nvidia/nv-vtophys.o
In file included from [01m[K./arch/x86/include/asm/preempt.h:7[m[K,
                 from [01m[K./include/linux/preempt.h:78[m[K,
                 from [01m[K./include/linux/spinlock.h:56[m[K,
                 from [01m[K/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/common/inc/nv-lock.h:29[m[K,
                 from [01m[K/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/common/inc/nv-linux.h:32[m[K,
                 from [01m[K/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/nvidia/nv-dma.c:27[m[K:
[01m[K/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/common/inc/nv-mm.h:[m[K In function ‘[01m[KNV_GET_USER_PAGES[m[K’:
[01m[K./arch/x86/include/asm/current.h:44:17:[m[K [01;35m[Kwarning: [m[Kpassing argument 1 of ‘[01m[Kget_user_pages[m[K’ makes integer from pointer without a cast [[01;35m[K]8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wint-conversion-Wint-conversion]8;;[m[K]
   44 | #define current [01;35m[Kget_current()[m[K
      |                 [01;35m[K^~~~~~~~~~~~~[m[K
      |                 [01;35m[K|[m[K
      |                 [01;35m[Kstruct task_struct *[m[K
[01m[K/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/common/inc/nv-mm.h:88:31:[m[K [01;36m[Knote: [m[Kin expansion of macro ‘[01m[Kcurrent[m[K’
   88 |         return get_user_pages([01;36m[Kcurrent[m[K, current->mm, start, nr_pages, write,
      |                               [01;36m[K^~~~~~~[m[K
In file included from [01m[K/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/common/inc/nv-pgprot.h:30[m[K,
                 from [01m[K/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/common/inc/nv-linux.h:33[m[K:
[01m[K./include/linux/mm.h:2408:35:[m[K [01;36m[Knote: [m[Kexpected ‘[01m[Klong unsigned int[m[K’ but argument is of type ‘[01m[Kstruct task_struct *[m[K’
 2408 | long get_user_pages([01;36m[Kunsigned long start[m[K, unsigned long nr_pages,
      |                     [01;36m[K~~~~~~~~~~~~~~^~~~~[m[K
In file included from [01m[K/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/common/inc/nv-linux.h:34[m[K:
[01m[K/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/common/inc/nv-mm.h:88:47:[m[K [01;35m[Kwarning: [m[Kpassing argument 2 of ‘[01m[Kget_user_pages[m[K’ makes integer from pointer without a cast [[01;35m[K]8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wint-conversion-Wint-conversion]8;;[m[K]
   88 |         return get_user_pages(current, current->mm, start, nr_pages, write,
[01m[K./include/linux/mm.h:2408:56:[m[K [01;36m[Knote: [m[Kexpected ‘[01m[Klong unsigned int[m[K’ but argument is of type ‘[01m[Kstruct mm_struct *[m[K’
 2408 | long get_user_pages(unsigned long start, [01;36m[Kunsigned long nr_pages[m[K,
      |                                          [01;36m[K~~~~~~~~~~~~~~^~~~~~~~[m[K
[01m[K/var/tmp/portage/x11-drivers/nvidia-drivers-535.54.03/work/kernel-module-source/kernel-open/common/inc/nv-mm.h:88:60:[m[K [01;35m[Kwarning: [m[Kpassing argument 4 of ‘[01m[Kget_user_pages[m[K’ makes pointer from integer without a cast [[01;35m[K]8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wint-conversion-Wint-conversion]8;;[m[K]
   88 |         return get_user_pages(current, current->mm, start, [01;35m[Knr_pages[m[K, write,
      |                                                            [01;35m[K^~~~~~~~[m[K
      |                                                            [01;35m[K|[m[K
      |                                                            [01;35m[Klong unsigned int[m[K
[... cut ...]
  182 |             return [01;31m[Kget_user_pages[m[K(NULL, mm, start, nr_pages, write, force, pages, vmas);
      |                    [01;31m[K^~~~~~~~~~~~~~[m[K
[01m[K./include/linux/mm.h:2408:6:[m[K [01;36m[Knote: [m[Kdeclared here
 2408 | long [01;36m[Kget_user_pages[m[K(unsigned long start, unsigned long nr_pages,
      |      [01;36m[K^~~~~~~~~~~~~~[m[K
....................
```

### More Info

This change is part of the "remove the vmas parameter from GUP APIs" changeset, which just got merged into Linus' tree.

ttabi, 2023-06-28:
There's no need to post bug reports for recent changes to upstream kernel that break the installer.  We are already aware of this change.  We have developers who monitor the kernel to catch these things.

oscarbg, 2023-07-07:
would be nice if in the meanwhile somebody can provide a patch.. assuming it's a simple change..

ElDavoo, 2023-09-02:
fixed in latest version

closed 2023-09-02 by ElDavoo