NVIDIA/open-gpu-kernel-modules #526: Error! Bad return status for module build on kernel: 6.4.1-zen1-1-zen (x86_64)

alufd19, 2023-07-03:
### NVIDIA Open GPU Kernel Modules Version

390.157

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Arch Linux

### Kernel Release

6.4.1-zen1-1-zen (x86_64)

### Hardware: GPU

NVIDIA Corporation GF108M [GeForce GT 540M] (rev a1)

### Describe the bug

DKMS make.log for nvidia-390.157 for kernel 6.4.1-zen1-1-zen (x86_64)
Mon  3 Jul 18:37:50 BST 2023
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 13.1.1 20230429
  You are using:           cc (GCC) 13.1.1 20230429
  SYMLINK /var/lib/dkms/nvidia/390.157/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/390.157/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: INIT_WORK
 CONFTEST: remap_pfn_range
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: follow_pfn
 CONFTEST: vmap
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: change_page_attr
 CONFTEST: pci_get_class
 CONFTEST: pci_choose_state
 CONFTEST: vm_insert_page
 CONFTEST: acpi_device_id
 CONFTEST: acquire_console_sem
 CONFTEST: console_lock
 CONFTEST: kmem_cache_create
 CONFTEST: on_each_cpu
 CONFTEST: smp_call_function
 CONFTEST: acpi_evaluate_integer
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: acpi_walk_namespace
 CONFTEST: pci_domain_nr
 CONFTEST: dma_mapping_error
 CONFTEST: sg_alloc_table
 CONFTEST: sg_init_table
 CONFTEST: get_num_physpages
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: efi_enabled
 CONFTEST: proc_create_data
 CONFTEST: pde_data
 CONFTEST: PDE_DATA
 CONFTEST: proc_remove
 CONFTEST: pm_vt_switch_required
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: write_cr4
 CONFTEST: of_get_property
 CONFTEST: of_find_node_by_phandle
[... 293 lines cut ...]
sudo dkms status
Result:
nvidia/390.157: added
vboxhost/7.0.8_OSE, 6.4.1-zen1-1-zen, x86_64: installed
vhba-module/20211218, 6.4.1-zen1-1-zen, x86_64: built

sudo dkms install nvidia/390.157
Result:
Sign command: /usr/lib/modules/6.4.1-zen1-1-zen/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j8 IGNORE_PREEMPT_RT_PRESENCE=1 NV_EXCLUDE_BUILD_MODULES='__EXCLUDE_MODULES' KERNEL_UNAME=6.4.1-zen1-1-zen modules.......................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.4.1-zen1-1-zen (x86_64)
Consult /var/lib/dkms/nvidia/390.157/build/make.log for more information.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

Operating System was work properly on kernel 6.3.9-zen1-1-zen, but appeared issue after updating to kernel 6.4.1-zen1-1-zen.

samedovprod, 2023-07-04:
I have the same thing right now, how to solve?

alufd19, 2023-07-04:
After update the kernel 6.4.1-zen1-1-zen to 6.4.1-zen2-1-zen appeared the same error: Error! Bad return status for module build on kernel: 6.4.1-zen2-1-zen (x86_64)

**Describe the bug**

DKMS make.log for nvidia-390.157 for kernel 6.4.1-zen2-1-zen (x86_64)
Tue  4 Jul 20:26:58 BST 2023
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 13.1.1 20230429
  You are using:           cc (GCC) 13.1.1 20230429
  SYMLINK /var/lib/dkms/nvidia/390.157/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/390.157/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: INIT_WORK
 CONFTEST: remap_pfn_range
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: follow_pfn
 CONFTEST: vmap
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: change_page_attr
 CONFTEST: pci_get_class
 CONFTEST: pci_choose_state
 CONFTEST: vm_insert_page
 CONFTEST: acpi_device_id
 CONFTEST: acquire_console_sem
 CONFTEST: console_lock
 CONFTEST: kmem_cache_create
 CONFTEST: on_each_cpu
 CONFTEST: smp_call_function
 CONFTEST: acpi_evaluate_integer
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: acpi_walk_namespace
 CONFTEST: pci_domain_nr
 CONFTEST: dma_mapping_error
 CONFTEST: sg_alloc_table
 CONFTEST: sg_init_table
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: efi_enabled
 CONFTEST: proc_create_data
 CONFTEST: pde_data
 CONFTEST: PDE_DATA
 CONFTEST: proc_remove
 CONFTEST: pm_vt_switch_required
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: write_cr4
 CONFTEST: of_get_property
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: for_each_online_node
 CONFTEST: node_end_pfn
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_remove_bus_device
 CONFTEST: request_threaded_irq
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: backlight_device_register
 CONFTEST: register_acpi_notifier
 CONFTEST: timer_setup
 CONFTEST: kmem_cache_create_usercopy
 CONFTEST: do_gettimeofday
 CONFTEST: dma_direct_map_resource
 CONFTEST: vmf_insert_pfn
[... 267 lines cut ...]
      |                  ^
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/nvidia/390.157/build/nvidia-drm/nvidia-drm-drv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:2024: /var/lib/dkms/nvidia/390.157/build] Error 2
make: *** [Makefile:81: modules] Error 2

**To Reproduce**

sudo dkms status
Result:
nvidia/390.157: added
vboxhost/7.0.8_OSE, 6.4.1-zen2-1-zen, x86_64: installed
vhba-module/20211218, 6.4.1-zen2-1-zen, x86_64: built

sudo dkms install nvidia/390.157
Result:
Sign command: /usr/lib/modules/6.4.1-zen2-1-zen/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j8 IGNORE_PREEMPT_RT_PRESENCE=1 NV_EXCLUDE_BUILD_MODULES='__EXCLUDE_MODULES' KERNEL_UNAME=6.4.1-zen2-1-zen modules......................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.4.1-zen2-1-zen (x86_64)
Consult /var/lib/dkms/nvidia/390.157/build/make.log for more information.

fin444, 2023-07-04:
I have the same problem, 6.4.0 and 6.4.1 non-Zen NixOS.

NVIDIA Corporation GA107M [GeForce RTX 3050 Mobile] (rev a1)

frosth555, 2023-07-04:
keep it mind it's not official fix. 
remove or comment out line with `nv_drm_driver.dumb_destroy` from `.../nvidia-drm/nvidia-drm-drv.c`
and build modules again, 
for archlinux&dkms it can be located at: `/var/lib/dkms/nvidia/XXX.YY.ZZ/source/nvidia-drm/nvidia-drm-drv.c`

[source](http://rglinuxtech.com/?p=3115&utm_source=rss&utm_medium=rss&utm_campaign=kernel-6-4-rc-pre-release-testing-breaks-nvidia-vmware-ok)

alufd19, 2023-07-07:
Package nvidia-390xx-dkms updated 2023-07-06 22:52 (UTC): https://aur.archlinux.org/packages/nvidia-390xx-dkms
Thanks jonathon (vnctdj) for error fixing!!!

closed 2023-07-07 by alufd19

fin444, 2023-07-07:
This issue affects all distros and multiple versions of the drivers. I wouldn't call it fixed just because a patched 390 is on the AUR.

Attigliuzzo, 2023-07-29:
> nv_drm_driver.dumb_destroy

@frosth555 , your fix works GREAT!!!!!! i use Debian Trixie with kernel 6.4.4 and Nvidia driver 390.157. THANKS!!!!!!

qrp73, 2023-09-12:
just updated linux arch and got this error:
```
==> dkms install --no-depmod nvidia/390.157 -k 6.5.2-arch1-1
Error! Bad return status for module build on kernel: 6.5.2-arch1-1 (x86_64)
Consult /var/lib/dkms/nvidia/390.157/build/make.log for more information.
==> WARNING: `dkms install --no-depmod nvidia/390.157 -k 6.5.2-arch1-1' exited 10
```

The fix proposed by @frosth555 doesn't works.

frosth555, 2023-09-12:
mentioned fix was for 6.4+ (aur package has it already), you need another one for 6.5 (look at #558)
and you will need another yet for 6.6 and again and again (untill they release 390 driver cause it is almost year old)