NVIDIA/open-gpu-kernel-modules #534: "Unrecognized AMD processor in cpuidInfoAMD" with AMD Ryzen 9 5950X

notpentadactyl, 2023-07-19:
### NVIDIA Open GPU Kernel Modules Version

535.86.05

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [X] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.4.3-273-tkg-cfs (self build)

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 3080

### Describe the bug

`dmesg` prints `NVRM cpuidInfoAMD: Unrecognized AMD processor in cpuidInfoAMD` on boot when running on a system with an AMD Ryzen 9 5950X CPU.

### To Reproduce

Boot the system with an AMD Ryzen 9 5950X CPU.

### Bug Incidence

Once

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/12093652/nvidia-bug-report.log.gz)

cemkaya-mpi, 2023-08-15:
I have the same error message with a Ryzen 6800H.

s3rgeym, 2023-08-26:
I'm getting random freezes when watching videos on my ryzen 5600h/rtx 3060. I think it's an nvidia bug because the video uses hardware acceleration. And i have a same error

kabasset, 2023-10-20:
I have the same error with an AMD Ryzen 5600 and RTX 3050. The only way to boot is in recovery mode.

notpentadactyl, 2024-02-04:
This is fixed in `550.40.07` (looking at git it was already fixed in `545.23.06`).

closed 2024-02-04 by notpentadactyl