NVIDIA/open-gpu-kernel-modules #537: Can't install nvidia-driver-535-open on Kubuntu 22.04

Kaiyangshi-Ito, 2023-07-26:
### NVIDIA Open GPU Kernel Modules Version

n/a

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Ubuntu 22.04.2 LTS

### Kernel Release

5.19.0-50-generic #50-Ubuntu SMP PREEMPT_DYNAMIC

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 4070 Laptop GPU

### Describe the bug

Can't install the GPU driver, 

```
The following packages have unmet dependencies:
 nvidia-driver-535-open : Depends: libnvidia-gl-535 (= 535.54.03-0ubuntu0.22.04.1) but 535.54.03-0ubuntu1 is to be installed
                          Depends: nvidia-kernel-source-535-open (= 535.54.03-0ubuntu0.22.04.1) but it is not going to be installed
                          Depends: libnvidia-compute-535 (= 535.54.03-0ubuntu0.22.04.1) but 535.54.03-0ubuntu1 is to be installed
                          Depends: libnvidia-extra-535 (= 535.54.03-0ubuntu0.22.04.1) but 535.54.03-0ubuntu1 is to be installed
                          Depends: nvidia-compute-utils-535 (= 535.54.03-0ubuntu0.22.04.1) but 535.54.03-0ubuntu1 is to be installed
                          Depends: libnvidia-decode-535 (= 535.54.03-0ubuntu0.22.04.1) but 535.54.03-0ubuntu1 is to be installed
                          Depends: libnvidia-encode-535 (= 535.54.03-0ubuntu0.22.04.1) but 535.54.03-0ubuntu1 is to be installed
                          Depends: nvidia-utils-535 (= 535.54.03-0ubuntu0.22.04.1) but 535.54.03-0ubuntu1 is to be installed
                          Depends: xserver-xorg-video-nvidia-535 (= 535.54.03-0ubuntu0.22.04.1) but 535.54.03-0ubuntu1 is to be installed
                          Depends: libnvidia-cfg1-535 (= 535.54.03-0ubuntu0.22.04.1) but 535.54.03-0ubuntu1 is to be installed
                          Depends: libnvidia-fbc1-535 (= 535.54.03-0ubuntu0.22.04.1) but 535.54.03-0ubuntu1 is to be installed
                          Recommends: libnvidia-compute-535:i386 (= 535.54.03-0ubuntu0.22.04.1)
                          Recommends: libnvidia-decode-535:i386 (= 535.54.03-0ubuntu0.22.04.1)
                          Recommends: libnvidia-encode-535:i386 (= 535.54.03-0ubuntu0.22.04.1)
                          Recommends: libnvidia-fbc1-535:i386 (= 535.54.03-0ubuntu0.22.04.1)
                          Recommends: libnvidia-gl-535:i386 (= 535.54.03-0ubuntu0.22.04.1)
E: Unable to correct problems, you have held broken packages.

```

### To Reproduce

```sudo apt install nvdiia-driver-535-open```

### Bug Incidence

Always

### nvidia-bug-report.log.gz

no such file since I can't install the driver

birdie-github, 2023-07-30:
If you're using Ubuntu packages, take it to Ubuntu: https://bugs.launchpad.net/ubuntu

elsandosgrande, 2023-07-31:
@Kaiyangshi-Ito That's a packaging issue, not an issue with the project itself. This is not the place to report packaging issues. Also, you should double-check which repositories you have enabled and which packages you have installed that could be causing this conflict.

closed 2023-07-31 by gauravjuvekar

nbollman, 2023-10-11:
I can add as a Ubuntu RTX3090 user I have not had much luck with the Open Kernel drivers... I dont think they are in a state for downloading... I stick with proprietary