NVIDIA/open-gpu-kernel-modules #545: Can't compile with DEBUG flag set on 535.86.05 (error: unknown type name ‘class’)

lachjames, 2023-08-13:
### NVIDIA Open GPU Kernel Modules Version

535.86.05

### Operating System and Version

Ubuntu 22.04

### Kernel Release

5.19.0-32-generic

### Build Command

```sh
make modules DEBUG=1
```

### Terminal output/Build Log

```
lachlan@lachlan-ubuntu-server:~/Downloads/open-gpu-kernel-modules-535.86.05$ make modules DEBUG=1
make -C src/nvidia
make[1]: Entering directory '/home/lachlan/Downloads/open-gpu-kernel-modules-535.86.05/src/nvidia'
 [ nvidia            ]  CC           generated/g_access_cntr_buffer_nvoc.c
 [ nvidia            ]  CC           generated/g_all_dcl_pb.c
 [ nvidia            ]  CC           generated/g_binary_api_nvoc.c
 [ nvidia            ]  CC           generated/g_bindata.c
 [ nvidia            ]  CC           generated/g_ccsl_nvoc.c
 [ nvidia            ]  CC           generated/g_ce_utils_nvoc.c
In file included from generated/g_ce_utils_nvoc.c:8:
generated/g_ce_utils_nvoc.h:169:25: error: unknown type name ‘class’
  169 | NVOC_PREFIX(ceutilsapi) class CeUtilsApi : GpuResource
      |                         ^~~~~
generated/g_ce_utils_nvoc.h:169:42: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘:’ token
  169 | NVOC_PREFIX(ceutilsapi) class CeUtilsApi : GpuResource
      |                                          ^
make[1]: *** [Makefile:196: _out/Linux_x86_64/g_ce_utils_nvoc.o] Error 1
make[1]: Leaving directory '/home/lachlan/Downloads/open-gpu-kernel-modules-535.86.05/src/nvidia'
make: *** [Makefile:34: src/nvidia/_out/Linux_x86_64/nv-kernel.o] Error 2
```

lachjames, 2023-08-13:
The issue seems to be related to this piece of code, conditionally compiled when DEBUG=1 or DEVELOP=1 (fwiw DEVELOP=1 didn't compile either).

```cpp
#if defined(DEBUG) || defined (DEVELOP)
NVOC_PREFIX(ceutilsapi) class CeUtilsApi : GpuResource
{
public:
    NV_STATUS ceutilsapiConstruct(CeUtilsApi *pCeUtilsApi, CALL_CONTEXT *pCallContext,
                                  RS_RES_ALLOC_PARAMS_INTERNAL *pParams)
                                  : GpuResource(pCallContext, pParams);

    void ceutilsapiDestruct(CeUtilsApi *pCeUtilsApi);

    //
    // Below APIs are only provided for SRT testing, thus only available for debug or
    // develop driver builds
    //
    //
    RMCTRL_EXPORT(NV0050_CTRL_CMD_MEMSET, RMCTRL_FLAGS(PRIVILEGED, API_LOCK_READONLY))
    NV_STATUS ceutilsapiCtrlCmdMemset(CeUtilsApi *pCeUtilsApi, NV0050_CTRL_MEMSET_PARAMS *pParams);

    RMCTRL_EXPORT(NV0050_CTRL_CMD_MEMCOPY, RMCTRL_FLAGS(PRIVILEGED, API_LOCK_READONLY))
    NV_STATUS ceutilsapiCtrlCmdMemcopy(CeUtilsApi *pCeUtilsApi, NV0050_CTRL_MEMCOPY_PARAMS *pParams);

    RMCTRL_EXPORT(NV0050_CTRL_CMD_CHECK_PROGRESS, RMCTRL_FLAGS(PRIVILEGED, API_LOCK_READONLY))
    NV_STATUS ceutilsapiCtrlCmdCheckProgress(CeUtilsApi *pCeUtilsApi, NV0050_CTRL_CHECK_PROGRESS_PARAMS *pParams);

    CeUtils *pCeUtils;
};
#endif
```

There's something weird going on, as this is meant to be a C file but it's trying to define a class?

xpahos, 2023-11-12:
Looks like this code should call C++ runtime from C runtime.

xpahos, 2023-11-13:
@aritger sorry for disturbing you. May I ask you or someone else from nvidia to generate `CeUtilsApi` from your sources? I was able to write manually basic methods but I have lack of knowledge how your code translated from C++ to C. I'm not sure that it will work correctly.