NVIDIA/open-gpu-kernel-modules #548: NVIDIA VGPU does not work

sdake, 2023-08-20:
### NVIDIA Open GPU Kernel Modules Version

535.86.10

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [X] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Description:	Debian GNU/Linux 12 (bookworm)

### Kernel Release

Linux wise-a40x1-1 6.1.0-11-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.38-4 (2023-08-08) x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA A30 24G

### Describe the bug

The NVIDIA VGPU software does not function with the `cloud-hypervisor` virtual machine monitor. An extensive analysis has [been completed](https://github.com/cloud-hypervisor/cloud-hypervisor/issues/5319), and a summary [has been produced](https://github.com/cloud-hypervisor/cloud-hypervisor/issues/5319#issuecomment-1685363316).

### To Reproduce

Try to use `nvidia-vgpu` with either a proprietary or open-source driver. In either case, the `nvidia-vgpu-vgpu` `mdev` control plane has odd expectations about the command line for the virtual machine monitor.

A very short summary is that `nvidia-vgpu` is hardcoded to `QEMU`. Most modern accelerated compute startups appear to want to use `cloud-hypervisor` as the technology has superior performance and quality.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

We are all working towards the same goal. I don't have the `nvidia-vgpu` software at this time. I will ask the individual that filed the issue to attach `nvidia-bug-report.sh` output.

@dengxuehua would you be kind enough to follow this issue tracker as well as provide the results of `nvidia-bug-report.sh`?

sdake, 2023-08-20:
Issue within `cloud-hypervisor`: https://github.com/cloud-hypervisor/cloud-hypervisor/issues/5319.

Thank you,
-steve

ttabi, 2023-08-20:
> I confirm that this does **not** happen with the proprietary driver package.

> Try to use nvidia-vgpu with either a proprietary or open-source driver. 

These two statements can't both be true.

sdake, 2023-08-21:
@ttabi A simple thank you for spending 20 minutes of my life reporting a bug against your product offerings would be in order.

Thank you.
-steve

cybik, 2023-09-26:
@aritger for the record, I know it's not the place, but GPU virtualization should REALLY not be locked out of customer boards. At least let us vGPU *one* instance so some of us can virtualize a GPU-enabled Windows VM or something.

aritger, 2023-09-26:
Thanks for your feedback.  I will relay your message to the appropriate team.

closed 2023-09-26 by aritger

sdake, 2023-09-27:
@aritger Did you read my request? I will repeat it so we understand each other.

Your drivers, whether proprietary or open, lack vgpu support for any hypervisor other than QEMU. The broader accelerated computing community much prefer to use modern hypervisors, such as cloud-hypervisor. And unfortunately, the structure of the implementation of vgpu, even if paid, does not work with this hypervisor.

Thank you
-steve

aaronp24, 2023-10-10:
I made sure the vgpu team is aware of your concern. Thank you for bringing this to our attention.

sdake, 2023-10-13:
Cool thanks Aaron! Super appreciate it. Github is a phenomenal tool to
interact with technology suppliers. not sure how the other perspective.

Cheers,
Steve