NVIDIA/open-gpu-kernel-modules #550: building main against current centos stream 8 fails

airlied, 2023-08-23:
### NVIDIA Open GPU Kernel Modules Version

535.104.05

### Operating System and Version

CentOS Stream 8

### Kernel Release

4.18.0-500.el8

### Build Command

just using an internal CI to build against latest centos stream containers.

### Terminal output/Build Log

[rawlog.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/12415692/rawlog.txt)

birdie-github, 2023-08-23:
```
In file included from /builds/airlied/open-gpu-kernel-modules/kernel-open/nvidia-uvm/uvm_gpu.h:31,
                 from /builds/airlied/open-gpu-kernel-modules/kernel-open/nvidia-uvm/uvm_ats_sva.h:27,
                 from /builds/airlied/open-gpu-kernel-modules/kernel-open/nvidia-uvm/uvm_ats_sva.c:24:
/builds/airlied/open-gpu-kernel-modules/kernel-open/nvidia-uvm/uvm_linux.h:157:32: error: expected identifier before numeric constant
 #define MPOL_PREFERRED_MANY    5
                                ^
./include/uapi/linux/mempolicy.h:25:2: note: in expansion of macro 'MPOL_PREFERRED_MANY'
  MPOL_PREFERRED_MANY,
  ^~~~~~~~~~~~~~~~~~~
```

johnhubbard, 2023-08-23:
The build failure below happens because uvm_linux.h defines MPOL_PREFERRED_MANY as a C macro. However, in the Linux kernel, that symbol is a C enum, which will show up as "not #defined" to the preprocessor.

I've filed internal bug 4255212 to track fixing this in our driver.

mmaneetsingh, 2023-09-21:
This issue should be resolved in the latest driver version: 535.113.01.
Closing this issue.

closed 2023-09-21 by mmaneetsingh