NVIDIA/open-gpu-kernel-modules #551: nvidia-smi -gtt doesn't work

kndgs, 2023-08-25:
### NVIDIA Open GPU Kernel Modules Version

535.104

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.4.12-zen1-1-zen

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3060 Ti (UUID: GPU-ba73bc75-4c91-6012-1365-c8e673737f6b)

### Describe the bug

GPU Target Temperature option is ignored. Regardles of value set GPU goes up to maximum temperature. This didn't happen on previous version of the driver.

### To Reproduce

nvidia-smi -gtt 65
run any heavy graphical app

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/12440800/nvidia-bug-report.log.gz)

ttabi, 2023-08-25:
Can you please try this with the proprietary driver of the same version?

ttabi, 2023-08-25:
This issue tracker is only for bugs specific to the open kernel driver.

kndgs, 2023-08-26:
So it happens with both versions. Will report on the nvidia forum I guess.

closed 2023-08-26 by kndgs

birdie-github, 2023-09-11:
> This issue tracker is only for bugs specific to the open kernel driver.

And where exactly should people report issues in the closed driver?

NVIDIA forums have tons of topics with no replies.

An NVIDIA email address acts like a black hole: whatever you sent to it is never replied or acted upon.

Weird.