NVIDIA/open-gpu-kernel-modules #555: allow a way to specify the kernel

vans163, 2023-09-06:
### NVIDIA Open GPU Kernel Modules Version

535.104.05

### Operating System and Version

Ubuntu 23.04

### Kernel Release

Linux 6.5.1 #1 SMP PREEMPT_DYNAMIC Wed Sep  6 18:49:37 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

### Build Command

make modules_install

### Terminal output/Build Log

```
make[1]: Leaving directory '/root/source/open-gpu-kernel-modules/src/nvidia'
cd kernel-open/nvidia/ && ln -sf ../../src/nvidia/_out/Linux_x86_64/nv-kernel.o nv-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/root/source/open-gpu-kernel-modules/kernel-open'
make[2]: Entering directory '/root/source/open-gpu-kernel-modules/kernel-open'
make[2]: *** /lib/modules/6.2.0-26-generic/build: No such file or directory.  Stop.
make[2]: Leaving directory '/root/source/open-gpu-kernel-modules/kernel-open'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/root/source/open-gpu-kernel-modules/kernel-open'
make: *** [Makefile:59: modules] Error 2

```

### More Info

I am running the build inside a systemd-container `make[2]: *** /lib/modules/6.2.0-26-generic/build: No such file or directory.  Stop.` `6.2.0-26-generic` is the output of `uname -r`. The NVIDIA driver correctly detects this and pulls the kernel from inside the systemd-container. the installer here does not give any option to specify the kernel to install the modules to or automatically detect it.

aritger, 2023-09-06:
It should be better documented, but you can use the `SYSSRC` and `SYSOUT` variables to tell the build where to find the kernel sources and build output.  See the logic in kernel-open/Makefile:

```
  ifdef SYSSRC
    KERNEL_SOURCES := $(SYSSRC)
  else
    KERNEL_UNAME ?= $(shell uname -r)
    KERNEL_MODLIB := /lib/modules/$(KERNEL_UNAME)
    KERNEL_SOURCES := $(shell test -d $(KERNEL_MODLIB)/source && echo $(KERNEL_MODLIB)/source || echo $(KERNEL_MODLIB)/build)
  endif

  KERNEL_OUTPUT := $(KERNEL_SOURCES)
  KBUILD_PARAMS :=

  ifdef SYSOUT
    ifneq ($(SYSOUT), $(KERNEL_SOURCES))
      KERNEL_OUTPUT := $(SYSOUT)
      KBUILD_PARAMS := KBUILD_OUTPUT=$(KERNEL_OUTPUT)
    endif
  else
    KERNEL_UNAME ?= $(shell uname -r)
    KERNEL_MODLIB := /lib/modules/$(KERNEL_UNAME)
    ifeq ($(KERNEL_SOURCES), $(KERNEL_MODLIB)/source)
      KERNEL_OUTPUT := $(KERNEL_MODLIB)/build
      KBUILD_PARAMS := KBUILD_OUTPUT=$(KERNEL_OUTPUT)
    endif
  endif
```

Does that help at all?

josephtingiris, 2025-02-21:
I landed here because of `nvidia-open/570.86.15` issues on Fedora 41.  When `dnf upgrade` is run, and it installs a new kernel, then `dkms` will automatically build `nvidia-open` under the running kernel (without explicitly setting a value for `KERNEL_UNAME`).  

That is evident in the `/var/lib/dkms/nvidia-open/*/*/*/log/make.log` file(s).  Notice it enter's the 'wrong' `/usr/src/kernels` directory,  ie:

```
DKMS (dkms-3.1.5) make.log for nvidia-open/570.86.15 for kernel 6.12.15-200.fc41.x86_64 (x86_64)
Fri Feb 21 09:50:50 AM EST 2025
Cleaning build area
# command: 'make' clean
make -C src/nvidia clean
make[1]: Entering directory '/var/lib/dkms/nvidia-open/570.86.15/build/src/nvidia'
rm -f -rf _out/Linux_x86_64
make[1]: Leaving directory '/var/lib/dkms/nvidia-open/570.86.15/build/src/nvidia'
make -C src/nvidia-modeset clean
make[1]: Entering directory '/var/lib/dkms/nvidia-open/570.86.15/build/src/nvidia-modeset'
rm -f -rf _out/Linux_x86_64
make[1]: Leaving directory '/var/lib/dkms/nvidia-open/570.86.15/build/src/nvidia-modeset'
make -C kernel-open clean
make[1]: Entering directory '/var/lib/dkms/nvidia-open/570.86.15/build/kernel-open'
rm -f -r conftest
make[2]: Entering directory '/usr/src/kernels/6.12.10-200.fc41.x86_64'
make[2]: Leaving directory '/usr/src/kernels/6.12.10-200.fc41.x86_64'
make[1]: Leaving directory '/var/lib/dkms/nvidia-open/570.86.15/build/kernel-open'
...
```

Subsequent reboots fail to load the module, eg:

```
kernel: nvidia: version magic '6.12.10-200.fc41.x86_64 SMP preempt mod_unload ' should be '6.12.11-200.fc41.x86_64 SMP preempt mod_unload '
```

Here's a short work-around (for Fedora) that leverages `KERNEL_UNAME`  and properly build `nvidia-open` for the 'latest' kernel.

After a `dnf upgrade`, a variation of following will work as `root`.

```sh
# do this as root
sudo su -

# fyi, version sort installed kernels
rpm -qa kernel | sed -e 's/^kernel-//g' | sort -uV

# fyi, version sort installed dkms module/module-version and kernel/arch
dkms status | sort -uV

export CURRENT_KERNEL="$(uname -r)"; echo "CURRENT_KERNEL=${CURRENT_KERNEL}"
export LATEST_KERNEL="$(rpm -qa kernel | sed -e 's/^kernel-//g' | sort -uV | tail -1)"; echo LATEST_KERNEL=${LATEST_KERNEL} # this matches uname -r

# example rebuild; pick one
export KERNEL_UNAME="6.12.11-200.fc41.x86_64"
export KERNEL_UNAME=${CURRENT_KERNEL}
export KERNEL_UNAME=${LATEST_KERNEL}
echo KERNEL_UNAME=${KERNEL_UNAME}

# set proper values for dkms build, install, etc
export DKMS_ARCH="$(dkms status | grep ${KERNEL_UNAME}, | awk -F, '{print $3}' | awk '{print $1}' | awk -F: '{print $1}')"
export DKMS_KERNEL="$(dkms status | grep ${KERNEL_UNAME}, | awk -F, '{print $2}' | awk '{print $1}')" # should be the same as KERNEL_UNAME
export DKMS_MODULE_VERSION="$(dkms status | grep ${KERNEL_UNAME}, | awk -F, '{print $1}' | awk '{print $1}')"

# manually verify values
echo DKMS_ARCH=${DKMS_ARCH}
echo DKMS_KERNEL=${DKMS_KERNEL}
echo DKMS_MODULE_VERSION=${DKMS_MODULE_VERSION}

# NOTICE! Using the LATEST_KERNEL value is easiest/safest.
# NOTICE! If you're booted with the latest kernel and the modules ARE NOT loaded, then properly rebuilding may immediately load the correct signed module and will likely reset a graphical session.
# IMPORTANT! If you're booted from the kernel you want to 'fix' then do this in a tmux, screen, or from the linux console.
KERNEL_UNAME=${DKMS_KERNEL} dkms uninstall ${DKMS_MODULE_VERSION} -k ${DKMS_KERNEL}/${DKMS_ARCH}
KERNEL_UNAME=${DKMS_KERNEL} dkms build ${DKMS_MODULE_VERSION} -k ${DKMS_KERNEL}/${DKMS_ARCH} --force
KERNEL_UNAME=${DKMS_KERNEL} dkms install ${DKMS_MODULE_VERSION} -k ${DKMS_KERNEL}/${DKMS_ARCH}
KERNEL_UNAME=${DKMS_KERNEL} dkms status ${DKMS_MODULE_VERSION} -k ${DKMS_KERNEL}/${DKMS_ARCH}

# verify the buiuld make.log enters the correct /usr/src/kernels directory ...
less /var/lib/dkms/nvidia-open/*/${DKMS_KERNEL}/${DKMS_ARCH}/log/make.log

systemctl reboot
```

Using `uname -r` in the `Makefile` seems a  bit too convenient for a dynamic kernel module.  I might spend some time writing a PR.

In the meantime, I posted this here hoping it would help someone else.  Good luck.

aritger, 2025-02-21:
Thanks for the post, @josephtingiris.  That definitely looks interesting, but integration with specific distribution package managers (like rpm) and integration with dkms are beyond the scope of the open-gpu-kernel-modules repo, which is distribution-agnostic.  I wonder if something like the script above would better belong as part of the Fedora+nvidia dkms integration?

josephtingiris, 2025-02-21:
> Thanks for the post, [@josephtingiris](https://github.com/josephtingiris). That definitely looks interesting, but integration with specific distribution package managers (like rpm) and integration [...]

I'm not sure and posting over there, too.  There are a tremendous number of threads with issues related to upgrades of nvidia-open.  Fundamentally, I think the issue starts here with `KERNEL_UNAME` in the [Makefile](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/kernel-open/Makefile)

scaronni, 2025-02-22:
@josephtingiris replied here, will be fixed in the new 570 builds: https://github.com/NVIDIA/yum-packaging-nvidia-driver/issues/10#issuecomment-2676145046

scaronni-nvidia, 2025-03-06:
Contained in the latest 570 builds. This can be closed.

closed 2025-03-13 by gauravjuvekar