NVIDIA/open-gpu-kernel-modules #558: Error! Bad return status for module build on kernel: 6.5.2-arch1-1 (x86_64)

qrp73, 2023-09-12:
### NVIDIA Open GPU Kernel Modules Version

390.157

### Operating System and Version

Arch Linux

### Kernel Release

6.5.2-arch1-1 #1 SMP PREEMPT_DYNAMIC x86_64

### Build Command

$ sudo dkms status
nvidia/390.157: added
$ sudo dkms install nvidia/390.157
Building module:
Cleaning build area...
'make' -j4 IGNORE_PREEMPT_RT_PRESENCE=1 NV_EXCLUDE_BUILD_MODULES='__EXCLUDE_MODULES' KERNEL_UNAME=6.5.2-arch1-1 modules.............(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.5.2-arch1-1 (x86_64)
Consult /var/lib/dkms/nvidia/390.157/build/make.log for more information.

### Terminal output/Build Log

```
DKMS make.log for nvidia-390.157 for kernel 6.5.2-arch1-1 (x86_64)
Tue Sep 12 17:30:21 EEST 2023
make[1]: Entering directory '/usr/lib/modules/6.5.2-arch1-1/build'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 13.2.1 20230801
  You are using:           cc (GCC) 13.2.1 20230801
  SYMLINK /var/lib/dkms/nvidia/390.157/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/390.157/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: INIT_WORK
 CONFTEST: remap_pfn_range
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: follow_pfn
 CONFTEST: vmap
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: change_page_attr
 CONFTEST: pci_get_class
 CONFTEST: pci_choose_state
 CONFTEST: vm_insert_page
 CONFTEST: acpi_device_id
 CONFTEST: acquire_console_sem
 CONFTEST: console_lock
 CONFTEST: kmem_cache_create
 CONFTEST: on_each_cpu
 CONFTEST: smp_call_function
 CONFTEST: acpi_evaluate_integer
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: acpi_walk_namespace
 CONFTEST: pci_domain_nr
 CONFTEST: dma_mapping_error
 CONFTEST: sg_alloc_table
 CONFTEST: sg_init_table
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: efi_enabled
 CONFTEST: proc_create_data
 CONFTEST: pde_data
 CONFTEST: PDE_DATA
 CONFTEST: proc_remove
 CONFTEST: pm_vt_switch_required
 CONFTEST: xen_ioemu_inject_msi
[... 315 lines cut ...]
./include/linux/mm.h:2397:6: note: declared here
 2397 | long get_user_pages_remote(struct mm_struct *mm,
      |      ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/390.157/build/nvidia/nv-gpu-numa.c: In function ‘gather_memblock_ids_for_node’:
/var/lib/dkms/nvidia/390.157/build/nvidia/nv-gpu-numa.c:232:47: warning: cast between incompatible function types from ‘int (*)(struct dir_context *, const char *, int,  loff_t,  u64,  unsigned int)’ {aka ‘int (*)(struct dir_context *, const char *, int,  long long int,  long long unsigned int,  unsigned int)’} to ‘bool (*)(struct dir_context *, const char *, int,  loff_t,  u64,  unsigned int)’ {aka ‘_Bool (*)(struct dir_context *, const char *, int,  long long int,  long long unsigned int,  unsigned int)’} [-Wcast-function-type]
  232 |     nv_dir_context_t ats_ctx = { .ctx.actor = (filldir_t)filldir_get_memblock_id };
      |                                               ^
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/nvidia/390.157/build/nvidia/nv-instance.o] Error 1
make[3]: *** Waiting for unfinished jobs....
cc1: some warnings being treated as errors
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/nvidia/390.157/build/nvidia/nv-gpu-numa.o] Error 1
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/nvidia/390.157/build/nvidia/nv-frontend.o] Error 1
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/nvidia/390.157/build/nvidia/nv.o] Error 1
make[2]: *** [/usr/lib/modules/6.5.2-arch1-1/build/Makefile:2032: /var/lib/dkms/nvidia/390.157/build] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.5.2-arch1-1/build'
make: *** [Makefile:81: modules] Error 2
```

### More Info

The error happens after OS update

qrp73, 2023-09-12:
Thanks, [this patch](https://gist.github.com/joanbm/dfe8dc59af1c83e2530a1376b77be8ba#file-nvidia-470xx-fix-linux-6-5-patch) works:
```
$ sudo wget https://gist.githubusercontent.com/joanbm/dfe8dc59af1c83e2530a1376b77be8ba/raw/37ff2b5ccf99f295ff958c9a44ca4ed4f42503b4/nvidia-470xx-fix-linux-6.5.patch
$ sudo mv nvidia-470xx-fix-linux-6.5.patch /var/lib/dkms/nvidia/390.157/source
$ cd /var/lib/dkms/nvidia/390.157/source

$ sudo patch -p1 < nvidia-470xx-fix-linux-6.5.patch
patching file common/inc/nv-mm.h

$ sudo dkms install nvidia/390.157
Building module:
Cleaning build area.....
'make' -j4 IGNORE_PREEMPT_RT_PRESENCE=1 NV_EXCLUDE_BUILD_MODULES='__EXCLUDE_MODULES' KERNEL_UNAME=6.5.2-arch1-1 modules................................................................
Signing module /var/lib/dkms/nvidia/390.157/build/nvidia.ko
Signing module /var/lib/dkms/nvidia/390.157/build/nvidia-uvm.ko
Signing module /var/lib/dkms/nvidia/390.157/build/nvidia-modeset.ko
Signing module /var/lib/dkms/nvidia/390.157/build/nvidia-drm.ko
Cleaning build area...

nvidia.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/6.5.2-arch1-1/updates/dkms/

nvidia-uvm.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/6.5.2-arch1-1/updates/dkms/

nvidia-modeset.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/6.5.2-arch1-1/updates/dkms/

nvidia-drm.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/6.5.2-arch1-1/updates/dkms/
depmod............
```

After reboot it is loaded and at a glance works ok

qrp73, 2023-09-15:
the patch works ok. Could you please apply it to AUR package nvidia-390xx-dkms?

frosth555, 2023-09-15:
@qrp73 I'm not aur maintainer, leave the comment on package site
you can add it to pkgbuild by yourself easly, the same way as previous patches were added by maintainer (you need change path to the patched file only, i guess)

ptr1337, 2023-09-19:
I dont think you should discuss 390xx branch here.

qrp73, 2023-09-20:
@ptr1337, since old video cards such as GTX460 is not supported on current nvidia module, it just don't works, 390xx branch is the only working one, all later versions just don't support many nvidia video cards. 

Where is the proper location to discuss 390xx branch?

ttabi, 2023-09-20:
There is no Open GPU Kernel Module (aka OpenRM) version 390.  The oldest is version 515.  This issue has got nothing to do with OpenRM, because you're not installing OpenRM.

To discuss issues with the standard Nvidia driver, go here: https://forums.developer.nvidia.com/c/gpu-graphics/linux/148

ptr1337, 2023-09-20:
> @ptr1337, since old video cards such as GTX460 is not supported on current nvidia module, it just don't works, 390.157 is the only working one, all later versions just don't support many nvidia video cards.
> 
> Where is the proper location to discuss 390xx branch?

You should use the forum for this. This issuetracker is only for cards, which do support the open kernel module. So, all cards starting from 1660.
ttabi already wrote you the link

ttabi, 2023-09-21:
> This issue tracker is only for cards, which do support the open kernel module. 

Actually, this issue tracker is only for people _using the open kernel module on those cards_.  If you're using a compatible GPU but you're not using the open kernel module, then this issue tracker is still not the right place.

qrp73, 2023-09-24:
ok, sorry for posting issue here, I'm a newbie in linux world. Anyway thanks for patch that solves issue.

closed 2023-09-24 by qrp73