NVIDIA/open-gpu-kernel-modules #569: Wrong version tag?

towo2099, 2023-10-31:
### NVIDIA Open GPU Kernel Modules Version

545.29.02

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

debian unstable

### Kernel Release

6.6.0

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

doesn't matter

### Describe the bug

Nvidia released propritary driver 545.29.02.
The version here is tagged 545.29.02 , so it does not fit.

### To Reproduce

look at the website, the propritary release is versioned 545.29.02

### Bug Incidence

Always

### nvidia-bug-report.log.gz

not needed

aritger, 2023-10-31:
Yes, there were some NVIDIA-internal communication problems that resulted in a disagreement between driver posting and open-gpu-kernel-modules source posting here.  I'll push 545.29.02 here shortly.  Sorry for the confusion.

aritger, 2023-10-31:
545.29.02 is released now: https://github.com/NVIDIA/open-gpu-kernel-modules/releases/tag/545.29.02
I apologize for the mess.

closed 2023-10-31 by aritger