NVIDIA/open-gpu-kernel-modules #571: 545.29.02 VRR HDMI 100% broken

adolfotregosa, 2023-11-01:
### NVIDIA Open GPU Kernel Modules Version

545.29.02

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Manjaro Linux

### Kernel Release

6.5.5

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

3090

### Describe the bug

VRR is not working at least via HDMI. I'm using an LG C1 and besides me clearly knowing when VRR is working or not, the TV built in fps/hz monitor stays put, not showing variations like normal. I have no idea if it is still working via display port, can't test it.

### To Reproduce

Play a game and watch TV built in FPS counter vs game FPS

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/13226496/nvidia-bug-report.log.gz)

aritger, 2023-11-01:
Thank you for the report.  Can you tell if this is a regression with 545.29.02 versus previous open-gpu-kernel-modules releases?  Also, it would help to know if this reproduces for you with the proprietary NVIDIA kernel modules.

adolfotregosa, 2023-11-01:
> Thank you for the report. Can you tell if this is a regression with 545.29.02 versus previous open-gpu-kernel-modules releases? Also, it would help to know if this reproduces for you with the proprietary NVIDIA kernel modules.

regression and yes. I have to warn that this was already reported by someone on the only 545 beta version

aritger, 2023-11-01:
A regression relative to what driver version?  I.e., in your testing, in which driver version does VRR work correctly?

adolfotregosa, 2023-11-01:
> A regression relative to what driver version? I.e., in your testing, in which driver version does VRR work correctly?

Via HDMI ? every single one after G-SYNC/VRR became supported up to the previous driver. 545.29.02, the latest one, is the first where it does not work at all. It kinda tries if you put HDMI frequency at 60Hz but is flickers to black. At 100Hz or 120Hz it is working exactly has a fixed rate signal.

aritger, 2023-11-01:
> 545.29.02, the latest one, is the first where it does not work at all.

Just to double check: can you confirm that 545.23.06 works fine for you in VRR?

adolfotregosa, 2023-11-01:
> > 545.29.02, the latest one, is the first where it does not work at all.
> 
> Just to double check: can you confirm that 545.23.06 works fine for you in VRR?

Sorry I cannot, I forgot about that beta. None of the 545 drivers are working. Only 535 and bellow.

aritger, 2023-11-01:
Okay.  The precise regression window is important, because that will guide the investigation into which particular changes could have broken this.  Can you confirm that 535.129.03 works correctly for you for VRR?

adolfotregosa, 2023-11-01:
> Okay. The precise regression window is important, because that will guide the investigation into which particular changes could have broken this. Can you confirm that 535.129.03 works correctly for you for VRR?

yes 100%, the issue started with 545.xx.xx

aritger, 2023-11-01:
Thanks for confirming.  I see this is also reported on https://forums.developer.nvidia.com/t/545-29-02-vrr-hdmi-100-not-working/271349 and it sounds like this is not specific to the open kernel modules.

I'm going to close this Issue (we're trying to keep the Issues here specific to the open kernel modules) and we can track it in the forum post.

closed 2023-11-01 by aritger

adolfotregosa, 2023-11-01:
> Thanks for confirming. I see this is also reported on https://forums.developer.nvidia.com/t/545-29-02-vrr-hdmi-100-not-working/271349 and it sounds like this is not specific to the open kernel [...]

That was also me that reported. I tested both closed and open and reported the issue on the correct forum for each.

ttabi, 2023-11-01:
> I tested both closed and open and reported the issue on the correct forum for each.

Actually, you didn't.  This forum is only for bug reports that _cannot_ be reproduced with the proprietary driver.  So you're never supposed to report the same bug on both forums.

This is made abundantly clear to you when you report bugs.  The first question you need to answer is "Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver."

ttabi, 2023-11-01:
> Imagine that I refused to use "proprietary" closed source drivers

Verifying that the bug does not occur with the proprietary driver is a requirement before posting a bug report here.  I don't how else to make that clear.

adolfotregosa, 2023-11-01:
yeah that "not" should be **_NOT_**