NVIDIA/open-gpu-kernel-modules #572: No video after waking after screen evergy saving mode (KDE Plasma)

Ponywka, 2023-11-02:
### NVIDIA Open GPU Kernel Modules Version

nvidia-open 535.113.01-8

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux [Rolling]

### Kernel Release

Linux ponyLaptop 6.5.9-arch2-1 #1 SMP PREEMPT_DYNAMIC Thu, 26 Oct 2023 00:52:20 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 4060 Laptop GPU (UUID: GPU-94db6f8b-687a-6ca9-e48f-ebf2d50d6d42)

### Describe the bug

When laptop trying wake after screen energy saving mode nothing happened (black screen) even on external monitors. Sometimes re plugging HDMI cable can help in this situation but not always

It can't be reproduced with "Sleep" button in KDE Menu. They going sleep and going wake back fine

### To Reproduce

Laptop: MSI Bravo 15 C7VF

1. Use "Discrete mode" (only NVIDIA without AMD APU) in MSI Center (application for MS Windows what configure proprietary hidden parameters in BIOS)
2. (Maybe) In laptop BIOS disabled iGPU and enable S3 sleep mode (disable "modern" sleep mode)
4. Use KDE Plasma
5. Lock laptop (Meta+L)
6. Go away from laptop for some time
7. Go back and try to unlock it back
8. Laptop doesn't show anything even on external display

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

I can't send `nvidia-bug-report.log.gz` now because I have no time to reproduce it but I can send `journalctl -o short-precise -k -b -2`: [nvrm_errors_after_wake.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/13237325/nvrm_errors_after_wake.log.gz)

I can send detailed logs later

Ponywka, 2023-11-02:
I tried to reproduce it with via "Sleep" button but it not happens.
Also I tried to get `nvidia-bug-report.log.gz` but I can't connect to laptop via SSH when it happens.
I need more time to give info :(

Ponywka, 2023-11-03:
Update: I attached the `nvidia-bug-report.log.gz` file on main message

NvidiaUser123, 2023-11-03:
[image]

Ponywka, 2023-11-03:
I use X11. Thanks
(I can't remember why I didn't use Wayland but I use X11)

robinhad, 2024-06-09:
Same issue on desktop 4090 on X11 as well

skarpinis, 2025-01-06:
will never expect nvidia fix their drivers for linux