NVIDIA/open-gpu-kernel-modules #573: The brightness control of NVIDIA seems to be broken

lisuke, 2023-11-08:
### NVIDIA Open GPU Kernel Modules Version

545.29.02-2

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [X] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Description:	Arch Linux

### Kernel Release

Linux ArchLinux 6.5.9-arch2-1 #1 SMP PREEMPT_DYNAMIC Thu, 26 Oct 2023 00:52:20 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: Quadro RTX 5000 (UUID: GPU-ed922411-fe5f-d295-686a-62cd62d986a5)

### Describe the bug

```
❯ journalctl -p 3 -b
11月 08 18:10:05 ArchLinux kernel: x86/cpu: SGX disabled by BIOS.
11月 08 18:10:05 ArchLinux kernel:
11月 08 18:10:05 ArchLinux kernel: NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64  545.29.02  Release Build  (archl>
11月 08 18:10:05 ArchLinux kernel: NVRM testIfDsmSubFunctionEnabled: GPS ACPI DSM called before _acpiDsmSupportedFuncCacheIn>
11月 08 18:10:05 ArchLinux systemd-backlight[490]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:10:05 ArchLinux kernel: iwlwifi 0000:52:00.0: BIOS contains WGDS but no WRDS
11月 08 18:10:05 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:10:05 ArchLinux systemd-backlight[540]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:10:05 ArchLinux kernel: ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-95)
11月 08 18:10:06 ArchLinux kernel: ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-95)
11月 08 18:10:06 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:10:06 ArchLinux systemd-backlight[563]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:10:06 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:10:06 ArchLinux systemd-backlight[567]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:10:06 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:10:06 ArchLinux systemd-backlight[573]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:10:06 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:10:06 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:10:06 ArchLinux kernel: ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-95)
11月 08 18:10:06 ArchLinux kernel: ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-95)
11月 08 18:10:07 ArchLinux kernel: Bluetooth: hci0: Malformed MSFT vendor event: 0x02
11月 08 18:10:08 ArchLinux bluetoothd[588]: src/device.c:set_wake_allowed_complete() Set device flags return status: Invalid>
11月 08 18:10:21 ArchLinux (sd-bright)[923]: nvidia_0: Failed to write brightness to device: Invalid argument
11月 08 18:10:21 ArchLinux systemd-backlight[930]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:10:21 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:10:21 ArchLinux systemd-backlight[989]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:10:21 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:10:21 ArchLinux systemd-backlight[1016]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:10:21 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:10:21 ArchLinux systemd-backlight[1041]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:10:21 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:10:21 ArchLinux systemd-backlight[1052]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:10:21 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:10:21 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:12:39 ArchLinux (sd-bright)[3911]: nvidia_0: Failed to write brightness to device: Invalid argument
11月 08 18:12:39 ArchLinux systemd-backlight[3915]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:12:39 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:12:39 ArchLinux systemd-backlight[3935]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:12:39 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:12:39 ArchLinux (sd-bright)[3940]: nvidia_0: Failed to write brightness to device: Invalid argument
11月 08 18:12:39 ArchLinux systemd-backlight[3939]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argument
11月 08 18:12:39 ArchLinux (sd-bright)[3955]: nvidia_0: Failed to write brightness to device: Invalid argument
[... 172 lines cut ...]
11月 08 18:59:46 ArchLinux systemd-backlight[36378]: nvidia_0: Failed to write system 'brightness' attribute: Invalid argume>
11月 08 18:59:46 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
11月 08 18:59:46 ArchLinux systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:nvidia_0.
```

### To Reproduce

```
❯ pkexec brillo -A 5
Error: dprintf: 55
Error: execution failed
```

```
❯ brightnessctl -q set $(brightnessctl -q get)
Failed to set brightness: Protocol error
```

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/13295795/nvidia-bug-report.log.gz)

aritger, 2023-11-08:
Thank you for the report.  We had a brightness control regression in our 545.xx releases.  You can refer to that bug as #4364327.

However, that bug:
- Also reproduced with the proprietary driver.
- Was a regression between the 535.xx drivers and 545.xx drivers.

It is most likely that the issue you are seeing is the same as that.  However, the report here says that the issue you are seeing does not happen with the proprietary driver.
Could you please:

- Double check that the issue really does not reproduce with the proprietary 545.xx driver?
- Check that the issue does not reproduce with the 535.xx driver?

That will help us tell if this is the same as bug 4364327, or a separate problem.

mamaraddio, 2023-11-09:
Also do I noticed the same issue, systemd fails to load/save brightness at startup/shutdown,
and tools like `brightnessctl` doesn't change the brightness level.

@aritger 
> * Check that the issue does not reproduce with the 535.xx driver?

I downgraded all NVIDIA packages I had installed from 454.xx to last 535.xx possible and it now works again.

System:

> OS: Arch Linux x86_64
> Kernel: 6.6.1-arch1-1
> GPU: NVIDIA GeForce GTX 1650 Mobile / Max-Q

Packages:

> libxnvctrl
> nvidia-open-dkms
> nvidia-settings
> nvidia-utils
> opencl-nvidia

aritger, 2023-11-09:
Thanks for confirming, @mamaraddio.  The bug you're seeing is consistent with our internal bug 4364327.

The attached patch (I believe you can apply it with `git am ...`) against 545.29.02 should fix the issue.  I hope we can include this in a future release soon.

[0001-nvkms-initialize-brightnessType-in-_BACKLIGHT_BRIGHT.patch.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/13310810/0001-nvkms-initialize-brightnessType-in-_BACKLIGHT_BRIGHT.patch.txt)

lisuke, 2023-11-10:
> Thanks for confirming, @mamaraddio. The bug you're seeing is consistent with our internal bug 4364327.
>
> The attached patch (I believe you can apply it with `git am ...`) against 545.29.02 should fix [...]

Thx, fixed.

closed 2023-11-10 by lisuke

CosmicFusion, 2023-11-11:
This bug -DOES- happen with the proprietary version of nvidia 545

And the patch to nvidia open fixes everything

aritger, 2023-11-13:
Thanks for confirming.  This should be fixed in an upcoming release.  Thanks.

mmaneetsingh, 2023-11-22:
This is fixed in the latest release, version 545.29.06 (branch 'main').

musamaanjum, 2023-12-22:
@aritger @mmaneetsingh any idea when https://developer.download.nvidia.com/compute/cuda/repos/debian12/x86_64/ will be updated with 545.29.06 as it has only 545.23.08? This release would greatly help a lot of users.

aritger, 2024-01-03:
I'm sorry, I don't have much visibility into which NVIDIA Linux GPU driver releases CUDA consumes.  That might be something to ask on the CUDA forums: https://forums.developer.nvidia.com/c/accelerated-computing/cuda/206

SalahAdDin, 2024-07-21:
It still happens on the new drivers and new kernels.

hoggle64, 2025-02-12:
Still happens to me on nvdia driver 550.120

Byron-Ding, 2025-08-08:
still happens in 575.64.05! 
Brightness control doesn't work at all in HP OMEN MAX with GeForce 5070 Ti