NVIDIA/open-gpu-kernel-modules #574: crypto_shash_ctx_aligned has been removed from the linux kernel in the linux-next version

babariviere, 2023-11-11:
### NVIDIA Open GPU Kernel Modules Version

545.29.02 (sha256-RncPlaSjhvBFUCOzWdXSE3PAfRPCIrWAXyJMdLPKuIU=)

### Operating System and Version

NixOS 23.05

### Kernel Release

Linux 6.6.0-next-20231110 #1-NixOS SMP PREEMPT_DYNAMIC Fri Nov 10 01:06:34 UTC 2023 x86_64 GNU/Linux

### Build Command

```
make -C kernel-open modules
```

### Terminal output/Build Log

```
make -C kernel-open modules
make[1]: Entering directory '/build/source/kernel-open'
make[2]: Entering directory '/nix/store/8vmsmjxz2c547cgmmgnrzava7gglmqhs-linux-6.6.0-next-20231110-dev/lib/modules/6.6.0-next-20231110/source'
make[3]: Entering directory '/nix/store/8vmsmjxz2c547cgmmgnrzava7gglmqhs-linux-6.6.0-next-20231110-dev/lib/modules/6.6.0-next-20231110/build'
  SYMLINK /build/source/kernel-open/nvidia/nv-kernel.o
  SYMLINK /build/source/kernel-open/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: ioremap_driver_hardened
 CONFTEST: ioremap_driver_hardened_wc
 CONFTEST: ioremap_cache_shared
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: pde_data
 CONFTEST: get_num_physpages
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_rebar_get_possible_sizes
 CONFTEST: wait_for_random_bytes
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write_has_pointer_pos_arg
 CONFTEST: dma_direct_map_resource
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
 CONFTEST: flush_cache_all
 CONFTEST: vmf_insert_pfn
 CONFTEST: jiffies_to_timespec
 CONFTEST: ktime_get_raw_ts64
[... 269 lines cut ...]
  CC [M]  /build/source/kernel-open/nvidia/nvlink_linux.o
  CC [M]  /build/source/kernel-open/nvidia/nvlink_caps.o
  CC [M]  /build/source/kernel-open/nvidia/linux_nvswitch.o
  CC [M]  /build/source/kernel-open/nvidia/procfs_nvswitch.o
  CC [M]  /build/source/kernel-open/nvidia/i2c_nvswitch.o
  CC [M]  /build/source/kernel-open/nvidia-uvm/uvm_ats_sva.o
  CC [M]  /build/source/kernel-open/nvidia-uvm/uvm_conf_computing.o
/build/source/kernel-open/nvidia/libspdm_shash.c: In function 'lkca_hmac_duplicate':
/build/source/kernel-open/nvidia/libspdm_shash.c:90:26: error: implicit declaration of function 'crypto_tfm_ctx_aligned'; did you mean 'crypto_tfm_ctx_align'? [-Werror=implicit-function-declaration]
   90 |         char *src_ipad = crypto_tfm_ctx_aligned(&src_tfm->base);
      |                          ^~~~~~~~~~~~~~~~~~~~~~
      |                          crypto_tfm_ctx_align
/build/source/kernel-open/nvidia/libspdm_shash.c:90:26: warning: initialization of 'char *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
/build/source/kernel-open/nvidia/libspdm_shash.c:91:26: warning: initialization of 'char *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
   91 |         char *dst_ipad = crypto_tfm_ctx_aligned(&dst_tfm->base);
      |                          ^~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /build/source/kernel-open/nvidia-uvm/uvm_sec2_test.o
cc1: some warnings being treated as errors
```

### More Info

crypto_shash_ctx_aligned has been removed from this specific commit:

https://github.com/torvalds/linux/commit/acd7799574e57f1e494a5b85741eee78d1e93aca

aritger, 2023-11-13:
Thanks for the report.  I've filed bug 4376379 internally for this.

In the meantime, given the contents of the upstream change, I imagine you could change:
```
- char *src_ipad = crypto_tfm_ctx_aligned(&src_tfm->base);
+ char *src_ipad = crypto_tfm_ctx_align(&src_tfm->base, crypto_tfm_alg_alignmask(&src_tfm->base) + 1);

- char *dst_ipad = crypto_tfm_ctx_aligned(&dst_tfm->base);
+ char *dst_ipad = crypto_tfm_ctx_align(&dst_tfm->base, crypto_tfm_alg_alignmask(&dst_tfm->base) + 1);
```
But I'm not familiar enough with this to judge the motivation for the removal upstream, and if that merits more involved fixes within the NVIDIA driver.

relief-melone, 2024-01-11:
same issue on the unstable channel of nixos (currently 24.05)

closed 2024-06-03 by babariviere