NVIDIA/open-gpu-kernel-modules #581: libGL error: failed to load driver iris; failed to load driver swrast

marek-lach, 2023-12-15:
### NVIDIA Open GPU Kernel Modules Version

545.29.06

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.6.7

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 4050 Max-Q

### Describe the bug

libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: iris
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: iris
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast

### To Reproduce

Launching `glxinfo32` returns:

$ glxinfo32
name of display: :1
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: iris
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: iris
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X Error of failed request: BadValue (integer parameter out of range for operation)
Major opcode of failed request: 152 (GLX)
Minor opcode of failed request: 24 (X_GLXCreateNewContext)
Value in failed request: 0x0
Serial number of failed request: 54
Current serial number in output stream: 55

### Bug Incidence

Once

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/13689624/nvidia-bug-report.log.gz)

ttabi, 2023-12-15:
You are supposed to confirm that does not happen with the proprietary driver.  That's what the first checkmark is for, but you didn't check it.

If you haven't already, please install and test the 545.29.06 proprietary driver and try to reproduce the bug.

marek-lach, 2023-12-15:
> You are supposed to confirm that the bug does not happen with the proprietary driver. That's what the first checkmark is for, but you didn't check it.
> 
> If you haven't already, please install and test the 545.29.06 proprietary driver and try to reproduce the bug.

This happens using the proprietary driver just the same...

ttabi, 2023-12-15:
> This happens using the proprietary driver just the same...

In that case, you need to report the bug on the forum, not here: https://forums.developer.nvidia.com/c/gpu-graphics/linux/148

marek-lach, 2023-12-15:
> > This happens using the proprietary driver just the same...
> 
> In that case, you need to report the bug on the forum, not here: https://forums.developer.nvidia.com/c/gpu-graphics/linux/148

[Did that](https://forums.developer.nvidia.com/t/libgl-error-failed-to-load-driver-iris-failed-to-load-driver-swrast/276284), closing here.

closed 2023-12-15 by marek-lach