NVIDIA/open-gpu-kernel-modules #584: Can't build nvidia driver on Linux 4.19 (Ubuntu 20.04)

zbceric, 2023-12-22:
### NVIDIA Open GPU Kernel Modules Version

545.29.06

### Operating System and Version

Ubuntu 20.04.4 LTS

### Kernel Release

4.19.112-0062 (Custom)

### Build Command

When building the NVIDIA driver, I encountered an error where ld couldn't find a series of .o files. I am using a custom kernel, and the project can be found at https://github.com/Soheil-ab/sage/. I want to install the NVIDIA driver for this kernel. I tried compiling from source code and also attempted to install using NVIDIA-Linux-x86_64-535.146.02.run, but both failed. However, when I switched to the default 5.4.0-169-generic kernel in Ubuntu, I was able to compile the driver successfully. I suspect that there may be an issue with the custom kernel, and I'm wondering if there are any specific kernel requirements for compiling the driver. How can I resolve the aforementioned problem?

### Terminal output/Build Log

  LD [M]  /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia.o
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-pci.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-dmabuf.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-nano-timer.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-acpi.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-cray.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-dma.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-i2c.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-mmap.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-p2p.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-pat.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-procfs.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-usermap.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-vm.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/nv-vtophys.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/os-interface.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/os-mlock.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/os-pci.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/os-registry.o: No such file or directory
ld: cannot find /home/zhangbochun/tools/nvidia/open-gpu-kernel-modules/kernel-open/nvidia/os-usermap.o: No such file or directory

aritger, 2024-01-03:
Could you provide a more complete build log?  I suspect the problem actually happened earlier in the build, resulting in failing to produce these object files, and the link failure is fallout from that.  How are you invoking the build and pointing the build at your custom kernel?

zbceric, 2024-01-03:
[nvidia-installer.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/13815457/nvidia-installer.log)
This is the log file generated when running NVIDIA-Linux-x86_64-535.113.01.run

zbceric, 2024-01-03:
> Could you provide a more complete build log? I suspect the problem actually happened earlier in the build, resulting in failing to produce these object files, and the link failure is fallout from that. How are you invoking the build and pointing the build at your custom kernel?

[nvidia-installer.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/13815466/nvidia-installer.log)

This is the log file generated when running NVIDIA-Linux-x86_64-535.113.01.run. Thanks!