NVIDIA/open-gpu-kernel-modules #588: Flicker on fullscreen OpenGL apps in 545 driver

bitrot-alpha, 2024-01-09:
### NVIDIA Open GPU Kernel Modules Version

545.29.06 (proprietary)

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Fedora 38

### Kernel Release

6.6.9

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

GTX 1070

### Describe the bug

When running an app in exclusive fullscreen, like Minecraft or Zoom, the screen flickers intermittently. When switching to another application, the flickering stops. It only happens when a program is full screen.

### To Reproduce

Run an OpenGL application in full screen.
Change from fullscreen to windowed, or vice versa.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/13881258/nvidia-bug-report.log.gz)

### More Info

I have tried switching my color profiles, swapping between 60 and 75hz refresh. Nothing seemed to change.
My default screen setup is one monitor, 1080p 75hz.

Minecraft seems to flicker between what is supposed to be drawn and the Mojang splash screen on the main menu.
I know that this is supposed to go on the forums since I'm using the proprietary driver, but I didn't want to make an account there.

bitrot-alpha, 2024-01-10:
Adding `MUTTER_DEBUG_FORCE_EGL_STREAM=1` to `/etc/environment` makes the issue go away.
Workaround found here: https://gitlab.freedesktop.org/xorg/xserver/-/issues/1317#note_1270402

ttabi, 2024-01-10:
You wrote "545.29.06 (proprietary)"

Does this mean that the problem happens with the proprietary 545.29.06 driver?

bitrot-alpha, 2024-01-11:
Yes. I am using a Pascal card.

ttabi, 2024-01-11:
Please close this bug as invalid.  Report the bug here instead: https://forums.developer.nvidia.com/c/gpu-graphics/linux/148

1. This issue tracker is only for bugs with the open kernel driver.  In fact, it says so right on the top, in big black letters:

**Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.**

2. Pascal boards are not supported by this driver anyway. See https://github.com/NVIDIA/open-gpu-kernel-modules/issues/19

I don't understand why you're reporting the bug here.  You're not using this driver, and you're not on supported hardware anyway, so what made you think to come here to report the bug?

mAd-DaWg, 2024-01-16:
I don't enjoy making comments like these, however, as stated by aritger in the first reply (of the second link you commented):

"Fortunately, NVIDIA's internal code base is organized such that we share a lot of code between the open kernel modules in this repository and the proprietary kernel modules that are needed for pre-Turing GPUs. So, many/most changes to the open kernel modules will also apply and benefit the proprietary kernel modules. It is not as if the proprietary kernel modules will be ignored and allowed to bitrot.

For the foreseeable future, NVIDIA will continue to support both: the proprietary kernel modules and the open kernel modules. The two sets of kernel modules share a lot of code, they provide the same interfaces to the user-mode driver components like OpenGL, Vulkan, and CUDA, and they will continue to evolve together."

Being a primarily linux motivated project, and given where this project is hosted, it makes sense to most people to report linux specific issues here, as stated above, because this code feeds back to the proprietary driver.

If you rather want people to log the issues [here](https://forums.developer.nvidia.com/c/gpu-graphics/linux/148), then probably ask them do so without the sarcasm.

cwegener, 2024-02-11:
NVIDIA have already proposed a solution for this here: https://gitlab.freedesktop.org/xorg/xserver/-/merge_requests/967

I recommend following the relevant discussions for latest updates: https://gitlab.freedesktop.org/xorg/xserver/-/merge_requests/967#note_2259660

closed 2025-11-01 by Binary-Eater