NVIDIA/open-gpu-kernel-modules #594: 6.8 rc1: Module is failing to build

ptr1337, 2024-01-22:
### NVIDIA Open GPU Kernel Modules Version

545.29.06

### Operating System and Version

CachyOS (ArchLinux based)

### Kernel Release

6.8.0rc1

### Build Command

        cd "${srcdir}/${_nv_pkg}/kernel"
        local MODULE_FLAGS=(
           KERNEL_UNAME="${pkgver}-${pkgsuffix}"
           IGNORE_PREEMPT_RT_PRESENCE=1
           NV_EXCLUDE_BUILD_MODULES='__EXCLUDE_MODULES'
           SYSSRC="${srcdir}/${_srcname}"
           SYSOUT="${srcdir}/${_srcname}"
        )
        make ${BUILD_FLAGS[*]} ${MODULE_FLAGS[*]} -j$(nproc) modules

### Terminal output/Build Log

warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 13.2.1 20240118
  You are using:           cc (GCC) 13.2.1 20240118
  SYMLINK /tmp/makepkg/linux-cachyos-rc/src/NVIDIA-Linux-x86_64-545.29.06/kernel/nvidia/nv-kernel.o
  SYMLINK /tmp/makepkg/linux-cachyos-rc/src/NVIDIA-Linux-x86_64-545.29.06/kernel/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: ioremap_driver_hardened
 CONFTEST: ioremap_driver_hardened_wc
 CONFTEST: ioremap_cache_shared
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_rebar_get_possible_sizes
 CONFTEST: wait_for_random_bytes
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write_has_pointer_pos_arg
 CONFTEST: dma_direct_map_resource
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
[... 674 lines cut ...]
diff --git a/kernel/rcu/tree_plugin.h b/kernel/rcu/tree_plugin.h
index 41021080ad25..72474d8ec180 100644
--- a/kernel/rcu/tree_plugin.h
+++ b/kernel/rcu/tree_plugin.h
@@ -406,7 +406,7 @@ void __rcu_read_lock(void)
                WRITE_ONCE(current->rcu_read_unlock_special.b.need_qs, true);
        barrier();  /* critical section after entry code. */
 }
-EXPORT_SYMBOL_GPL(__rcu_read_lock);
+EXPORT_SYMBOL(__rcu_read_lock);
 
 /*
  * Preemptible RCU implementation for rcu_read_unlock().
@@ -431,7 +431,7 @@ void __rcu_read_unlock(void)
                WARN_ON_ONCE(rrln < 0 || rrln > RCU_NEST_PMAX);
        }
 }
-EXPORT_SYMBOL_GPL(__rcu_read_unlock);
+EXPORT_SYMBOL(__rcu_read_unlock);
 
 /*
  * Advance a ->blkd_tasks-list pointer to the next entry, instead
-- 
2.43.0
```

ptr1337, 2024-01-22:
Okay, it appears that the GPL error is also present on 545, if patching the `DRM_UNLOCKED` issue.

Would be cool, if you would fix this with the upcoming release and including the 4070 Super support so that i can finally use the 545.xx drivers :)

ttabi, 2024-01-22:
I'm sorry, but we don't accept bug reports -rc kernels.  Unfortunately, the template for reporting build bugs is missing this clause and checkmark:

**Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.**
☐ I am running on a stable kernel release.

Please close this bug and try again when 6.8 is released.  Nvidia already does testing with -rc kernels and have fixes for these already in the queue.

ptr1337, 2024-01-22:
Tabi, don't take me wrong but there will be multiple people which will run into this issue, when they are doing the kernel testing.
This issue should be kept open, so other users can see what is going on.

It is good, that NVIDIA is doing internal testing with the RC Kernel, but we don't have access to your roadmap or issuetracker, so it should be good to have this publicly visible.

Also, I have reported in 4 different kernel versions in the past and it just kept be open and it was fine.

Feel free to close it, if you want to do so.

ttabi, 2024-01-22:
The reason we don't want bug reports for -rc kernels is because we already know about the issues.  The problem exists with every kernel, so it's just noise to file these reports.  This bug tracker should be used only for bugs that Nvidia doesn't already know about.

Also, we are in the process of fixing the template.

gulafaran, 2024-01-24:
> The reason we don't want bug reports for -rc kernels is because we already know about the issues. The problem exists with every kernel, so it's just noise to file these reports. This bug tracker [...]

there is so much to to this that i barely know where to begin, so nvidias official take is kernel development is not support on nvidia devices? you either wait until nvidia release a driver 3 months down the road or hunt on obscure reddit posts for user workarounds to get it running, "bugs Nvidia doesnt already know about." well wow if only you didnt keep such things internal and actually disclosed your known bugs or even responded to things in your own dev forum about it. 

this would solve itself in many cases if your bugs wasnt internal so we actually knew what you know...

ttabi, 2024-01-24:
The reason for the policy against reporting bugs on -rc kernels is to avoid pointless arguments like this one.  Reporting bugs like these just waste everyone's time.

Nvidia already aggressively tests -rc kernels and even linux-next.  So we sometimes update our build process for changes that haven't made it into an -rc kernel yet.  However, we can't catch everything, and we do have a development process that we stick to.  So you'll just have to be patient.

amie1972, 2024-01-29:
> Okay, it appears that the GPL error is also present on 545, if patching the `DRM_UNLOCKED` issue.
> 
> Would be cool, if you would fix this with the upcoming release and including the 4070 Super support so that i can finally use the 545.xx drivers :)
Is this patch for the kernel or the driver?
My understanding is that it is for the kernel. Isn't that not good?

ptr1337, 2024-01-29:
> > Okay, it appears that the GPL error is also present on 545, if patching the `DRM_UNLOCKED` issue.
> > Would be cool, if you would fix this with the upcoming release and including the 4070 Super support so that i can
> > finally use the 545.xx drivers
> 
> Is this patch for the kernel or the driver? My understanding is that it is for the kernel. Sorry, I don't think this is a good idea.

The following patch needs to be applied on the kernel, this is required for 545 as well as 550.
In 550 they fixed the DRM_UNLOCKED issue but not the GPL Issue
```
From 44154371304cc8d068dd33c5dec4057926c364af Mon Sep 17 00:00:00 2001
From: Peter Jung <admin@ptr1337.dev>
Date: Mon, 22 Jan 2024 18:26:56 +0100
Subject: [PATCH] Workaround to fix issues when compiling the nvidia driver

---
 arch/x86/kernel/alternative.c | 2 +-
 kernel/rcu/tree_plugin.h      | 4 ++--
 2 files changed, 3 insertions(+), 3 deletions(-)

diff --git a/arch/x86/kernel/alternative.c b/arch/x86/kernel/alternative.c
index cc130b57542a..1d85cb7071cb 100644
--- a/arch/x86/kernel/alternative.c
+++ b/arch/x86/kernel/alternative.c
@@ -403,7 +403,7 @@ noinstr void BUG_func(void)
 {
        BUG();
 }
-EXPORT_SYMBOL_GPL(BUG_func);
+EXPORT_SYMBOL(BUG_func);
 
 #define CALL_RIP_REL_OPCODE    0xff
 #define CALL_RIP_REL_MODRM     0x15
diff --git a/kernel/rcu/tree_plugin.h b/kernel/rcu/tree_plugin.h
index 41021080ad25..72474d8ec180 100644
--- a/kernel/rcu/tree_plugin.h
+++ b/kernel/rcu/tree_plugin.h
@@ -406,7 +406,7 @@ void __rcu_read_lock(void)
                WRITE_ONCE(current->rcu_read_unlock_special.b.need_qs, true);
        barrier();  /* critical section after entry code. */
 }
-EXPORT_SYMBOL_GPL(__rcu_read_lock);
+EXPORT_SYMBOL(__rcu_read_lock);
 
 /*
  * Preemptible RCU implementation for rcu_read_unlock().
@@ -431,7 +431,7 @@ void __rcu_read_unlock(void)
                WARN_ON_ONCE(rrln < 0 || rrln > RCU_NEST_PMAX);
        }
 }
-EXPORT_SYMBOL_GPL(__rcu_read_unlock);
+EXPORT_SYMBOL(__rcu_read_unlock);
 
 /*
  * Advance a ->blkd_tasks-list pointer to the next entry, instead
-- 
2.43.0
```

This one seems to help for the 545 driver and 6.8:
https://github.com/CachyOS/kernel-patches/blob/master/6.8/misc/nvidia/0001-nvidia-545.patch

We didnt check, how they did fix it in the 550 series yet, but the above workaround works.

amie1972, 2024-01-29:
> From 44154371304cc8d068dd33c5dec4057926c364af Mon Sep 17 00:00:00 2001
> From: Peter Jung <admin@ptr1337.dev>
> Date: Mon, 22 Jan 2024 18:26:56 +0100
> Subject: [PATCH] Workaround to fix issues when compiling the nvidia driver
> 
> Signed-off-by: Peter Jung <admin@ptr1337.dev>
> ---
>  arch/x86/kernel/alternative.c | 2 +-
>  kernel/rcu/tree_plugin.h      | 4 ++--
>  2 files changed, 3 insertions(+), 3 deletions(-)
> 
> diff --git a/arch/x86/kernel/alternative.c b/arch/x86/kernel/alternative.c
> index cc130b57542a..1d85cb7071cb 100644
> --- a/arch/x86/kernel/alternative.c
> +++ b/arch/x86/kernel/alternative.c
> @@ -403,7 +403,7 @@ noinstr void BUG_func(void)
>  {
>         BUG();
>  }
> -EXPORT_SYMBOL_GPL(BUG_func);
> +EXPORT_SYMBOL(BUG_func);
>  
>  #define CALL_RIP_REL_OPCODE    0xff
>  #define CALL_RIP_REL_MODRM     0x15
> diff --git a/kernel/rcu/tree_plugin.h b/kernel/rcu/tree_plugin.h
> index 41021080ad25..72474d8ec180 100644
> --- a/kernel/rcu/tree_plugin.h
> +++ b/kernel/rcu/tree_plugin.h
> @@ -406,7 +406,7 @@ void __rcu_read_lock(void)
>                 WRITE_ONCE(current->rcu_read_unlock_special.b.need_qs, true);
>         barrier();  /* critical section after entry code. */
>  }
> -EXPORT_SYMBOL_GPL(__rcu_read_lock);
> +EXPORT_SYMBOL(__rcu_read_lock);
>  
>  /*
>   * Preemptible RCU implementation for rcu_read_unlock().
> @@ -431,7 +431,7 @@ void __rcu_read_unlock(void)
>                 WARN_ON_ONCE(rrln < 0 || rrln > RCU_NEST_PMAX);
>         }
>  }
> -EXPORT_SYMBOL_GPL(__rcu_read_unlock);
> +EXPORT_SYMBOL(__rcu_read_unlock);
>  
>  /*
>   * Advance a ->blkd_tasks-list pointer to the next entry, instead
> -- 
> 2.43.0

Since I can't compile the kernel yet (which is dangerous for a noob like me), I request you to submit this workaround to the official kernel, or patch the nvidia driver itself to fix the problem.

aritger, 2024-01-29:
Correct: the `DRM_UNLOCKED` issue was an NVIDIA bug, and fixed in the most recent NVIDIA driver release.

The `BUG_func` issue is resolved in the mainline Linux kernel:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/arch/x86/kernel/alternative.c?id=f7cfe7017b531e08c108ac6615b1ddedcc892428

Are you still seeing problems with `__rcu_read_unlock`/`__rcu_read_lock`?  I'm not aware of any remaining problems there.

ptr1337, 2024-01-29:
> Correct: the `DRM_UNLOCKED` issue was an NVIDIA bug, and fixed in the most recent NVIDIA driver release.
>
> The `BUG_func` issue is resolved in the mainline Linux kernel: [...]

Hi @aritger ,

yes, i compiled today 6.8 rc2 and the rcu_read_unlock and rcu_read_lock GPL issue still affects the 550 drivers.

DRM_UNLOCKED got fixed in the NVIDIA 550 and the BUG_func got also fixed in rc2.

aritger, 2024-01-29:
Thanks, @ptr1337.

Can you tell if the __rcu_read_unlock/__rcu_read_lock problem is new with 6.8 rc's versus 6.7?  Are you using the same kconfig between 6.7 and 6.8-rcN?  I vaguely recall some scenarios in the past where some debug kconfig knobs (maybe CONFIG_DEBUG_KMEMLEAK?) cause common utility macros to call EXPORT_SYMBOL_GPL __rcu_read_unlock/__rcu_read_lock functions.

If you're hitting the problem that I'm remembering, then the options were:

(a) disable whatever kconfig knob causes the indirect calls to __rcu_read_unlock/__rcu_read_lock.

(b) use the open kernel modules, rather than the closed kernel modules.

If what you're seeing is new with 6.8-rcN, then I'd like to investigate further.
But, if what you're seeing is preexisting (6.7 or earlier) when using the same kconfig, then I fear options (a) and (b) are the only viable choices.

aritger, 2024-01-30:
Oh, sorry: we actually do have a new problem in the 6.8-rcs with __rcu_read_unlock/__rcu_read_lock, due to use of the macro pfn_valid() which in turn calls those EXPORT_GPL_SYMBOLS.  It required a bit of detangling, but our next 550 release should remove the pfn_valid() use.  Thanks for your patience.

ptr1337, 2024-01-30:
> Oh, sorry: we actually do have a new problem in the 6.8-rcs with __rcu_read_unlock/__rcu_read_lock, due to use of the macro pfn_valid() which in turn calls those EXPORT_GPL_SYMBOLS. It required a [...]

Hi!

Great to hear, that it will be fixed in the upcoming 550 Release (likely with the 4080 Super launch?).
Here you can find the used config as well as PKGBUILD:
https://github.com/CachyOS/linux-cachyos/blob/master/linux-cachyos-rc/config

Currently we patch for the RC Kernel `__rcu_read_unlock/__rcu_read_lock` to have compatbility with nvidia for our kernel testing.

Looking forward for the next release.

6.7 is completly fine btw.

piso77, 2024-02-01:
> Oh, sorry: we actually do have a new problem in the 6.8-rcs with __rcu_read_unlock/__rcu_read_lock, due to use of the macro pfn_valid() which in turn calls those EXPORT_GPL_SYMBOLS. It required a [...]

Any ETA for the release with the pfn_valid() fix?

towo2099, 2024-02-01:
Now it happens with linux 6.7.3 too.

ptr1337, 2024-02-01:
@aritger @ttabi 
I really think you should consider to revert this commit https://github.com/NVIDIA/open-gpu-kernel-modules/commit/bb2dac1f20a06f78e028c4acdc4df1c7908dd91c and have here information and discussions about upstream kernel regression.

as you see above it helped already several people and distrubutions to have information about the current state.

smira, 2024-02-01:
Now same problem with Linux 6.1.76 as well.

birdie-github, 2024-02-01:
@aritger 

This commit now breaks binary NVIDIA drivers on Linux 6.7.3, 6.6.15 and 6.1.76:

```
commit 3a01daace71b521563c38bbbf874e14c3e58adb7
Author: Charan Teja Kalla
Date:   Fri Oct 13 18:34:27 2023 +0530

    mm/sparsemem: fix race in accessing memory_section->usage
    commit 5ec8e8ea8b7783fab150cf86404fc38cb4db8800 upstream.
```

A new beta couldn't come earlier, please.

NuLL3rr0r, 2024-02-01:
I can confirm this is reproducible with <code>6.6.15</code> as well. <code>6.6.14</code> builds as expected. I'm using Gentoo ebuilds x11-drivers/nvidia-drivers-545.29.06-r1.

ptr1337, 2024-02-01:
> I can confirm this is reproducible with `6.6.15` as well. `6.6.14` builds as expected. I'm using Gentoo ebuilds `x11-drivers/nvidia-drivers-545.29.06-r1`.

You can "workaround" it with the above patch

NuLL3rr0r, 2024-02-01:
Thank you! I saw the patch. Just wanted to raise the issue for 6.6.x as well since the initial discussion was about not accepting bug reports for -rc kernels.

ionenwks, 2024-02-01:
Kinda off-topic for the open source variant (not affected that I can see), but for those that need a patch *for the drivers, not kernel* using the blob variant, see: https://forums.developer.nvidia.com/t/280908/19

Unsure how NVIDIA is planning to fix this, but that's the simplest patch I could come up with as a quick fix.

birdie-github, 2024-02-03:
A Gentoo developer wrote a patch which doesn't involve patching the kernel (which many can't do and which Fedora/RedHat have outright refused to):

https://bugs.gentoo.org/923456

birdie-github, 2024-02-23:
550.54.14 drivers have just been released:

https://www.nvidia.com/Download/driverResults.aspx/218826/en-us/

I wonder if the issue has been addressed. There's nothing in the release notes.

ptr1337, 2024-02-23:
Yes, it appears to be fixed. Will close this now.

closed 2024-02-23 by ptr1337