NVIDIA/open-gpu-kernel-modules #597: 535, 545, 550 fail to build with GCC14

flukejones, 2024-01-31:
### NVIDIA Open GPU Kernel Modules Version

535, 545, 550

### Operating System and Version

Fedora rawhide (fedora 40 testing)

### Kernel Release

6.6, 6.7

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Build Command

`NVIDIA<version>.run`

### Terminal output/Build Log

```
conftest4394.c: In function 'conftest_drm_gem_object_vmap_has_map_arg':
conftest4394.c:24:46: error: passing argument 2 of 'obj->funcs->vmap' from incompatible pointer type [-Wincompatible-pointer-types]
   24 |                 return obj->funcs->vmap(obj, map);
      |                                              ^~~
      |                                              |
      |                                              struct dma_buf_map *
conftest4394.c:24:46: note: expected 'struct iosys_map *' but argument is of type 'struct dma_buf_map *'
```

### More Info

The bug is in the `conftest.sh`

See https://forums.developer.nvidia.com/t/nvidia-modules-build-failure-with-upcoming-gcc-14-and-recent-kernels-due-to-misfiring-conftest-sh-test-heads-up/279072 for more info as it affect open plus closed driver

aritger, 2024-01-31:
Thanks for the pointer to that forum post.  I've filed NVIDIA internal bug 4478534 to track it.  Hopefully we can resolve this soon.

scaronni, 2024-04-27:
550.76 and 550.78 build fine on my Fedora systems with GCC 14.

flukejones, 2024-05-18:
550.78 is failing with:

```
   /home/luke/Downloads/NVIDIA-Linux-x86_64-550.78/kernel-open/nvidia/os-mlock.c: In function 'nv_follow_pfn':
   /home/luke/Downloads/NVIDIA-Linux-x86_64-550.78/kernel-open/nvidia/os-mlock.c:42:12: error: implicit declaration of function 'follow_pfn'; did you mean 'follow_pte'? [-Wimplicit-function-declaration]
      42 |     return follow_pfn(vma, address, pfn);
         |            ^~~~~~~~~~
         |            follow_pte
```

flukejones, 2024-05-18:
Seems related https://lore.kernel.org/linux-mm/20240324234542.2038726-3-hch@lst.de/T/#mafdd8606dc57d34d7b9c45dbd4160157b2f049c1

flukejones, 2024-05-18:
Related: https://lore.kernel.org/all/20240324234542.2038726-1-hch@lst.de/T/#mdeb04975c27076d6830f96c2041f50ef1d2554fc

follow_pfn is removed.

aritger, 2024-05-20:
Thanks for the pointer on follow_pfn.  We're aware of that and it will be fixed in a future release.

I also think that is unrelated to the originally reported problem here (interaction with `-Werror=incompatible-pointer-types`).

flukejones, 2024-05-21:
Closing as fixed recently

closed 2024-05-21 by flukejones