NVIDIA/open-gpu-kernel-modules #598: [4k] Mesh Odyssey shader toy demo doesn't render

birdie-github, 2024-02-10:
### NVIDIA Open GPU Kernel Modules Version

550.40.07

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Fedora 39

### Kernel Release

6.7.3

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce GTX 1660 Ti

### Describe the bug

https://www.shadertoy.com/view/NdB3zK renders a black screen both in Firefox and Google Chrome.

### To Reproduce

Open the URL.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

Non-applicable

thesword53, 2024-02-10:
The shader from https://www.shadertoy.com/view/NdB3zK seems broken. I even have a black screen on Windows driver and on i915 (Intel HD Graphics 4600).

birdie-github, 2024-02-11:
Indeed, I get a black screen with AMD/Mesa as well.

closed 2024-02-11 by birdie-github