NVIDIA/open-gpu-kernel-modules #601: Nvidia Linux Driver v550.54.14 Docker transcoding broken

ich777, 2024-02-27:
### NVIDIA Open GPU Kernel Modules Version

550.54.14

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Unraid 6.12.8

### Kernel Release

Linux Test 6.1.74-Unraid #1 SMP PREEMPT_DYNAMIC Fri Feb  2 11:06:32 PST 2024 x86_64 AMD Ryzen 3 3200G with Radeon Vega Graphics AuthenticAMD GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA T400 (UUID: GPU-09e16239-57bc-2ca8-39ca-c72ed08bac48)

### Describe the bug

Transcoding fails using Docker container (official Jellyfin/Plex container).

### To Reproduce

Start a transcode in Docker with Jellyfin or Plex, transcoding fails on Jellyfin, transcoding falls back to Software transcoding on Plex. Even when tryting to run the command and passing through the newly created device to the container [here](https://github.com/NVIDIA/libnvidia-container/issues/241#issuecomment-1964323788) it doesn't work.

Error message from Jellyfin:
```
...
frame=    1 fps=0.0 q=0.0 size=N/A time=00:00:00.00 bitrate=N/A speed=   0x    
[h264_nvenc @ 0x561522fa7240] Failed locking bitstream buffer: invalid param (8): 
Error submitting video frame to the encoder
[libfdk_aac @ 0x561522fa4400] 2 frames left in the queue on closing
Conversion failed!
```

Error message from Plex:
```
...
Feb 26, 2024 15:29:24.384 [23302936435512] Fehlersuche — Jobs: '/usr/lib/plexmediaserver/Plex Transcoder' exit code for process 1386 is -9 (signal: Killed)
...
```

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/14416781/nvidia-bug-report.log.gz)

### More Info

I already reported that [here](https://github.com/NVIDIA/libnvidia-container/issues/241#issuecomment-1964284257) to rule out libnvidia-container and nvidia-container-toolkit and [here](https://forums.developer.nvidia.com/t/nvidia-linux-driver-v550-54-14-docker-broken/284057) for the proprietary driver on the Nvidia Developer Forums (the same also happens also on the proprietary driver).

Transcoding in Docker is working fine with Nvidia driver version 550.40.07 and below (open source and proprietary).

shperrung, 2024-02-28:
Confirm that issue.
HW transcoding doesn't work in go-vod container (stream provider for Nextcloud - Memories) after nvidia driver/applications upgraded to v. 550.54.14. Error output is the same as wrote @ich777 

Description: Debian GNU/Linux 11 (bullseye)
Release: 11
Codename: bullseye

openmediavault information

Release: 6.9.14-1
Codename: Shaitan

System information

Linux omv6 5.19.17-2-pve #1 SMP PREEMPT_DYNAMIC PVE 5.19.17-2 (Sat, 28 Jan 2023 16:40:25  x86_64 GNU/Linux

Device:
Nvidia T400 Quadro.

ich777, 2024-02-28:
@shperrung did you yet try what @klueska recommended [here](https://github.com/NVIDIA/libnvidia-container/issues/241#issuecomment-1964323788).

shperrung, 2024-02-28:
@ich777 No I don't, because I read it just now.
In my case I checked a GPU availability inside go-vod container with 'nvidia-smi' and got standard positive output. Therefore, I don't see reasons that creation of ".../channel0" may change something.
Applications in container sees GPU but ffmpeg can't utilise HW codecs. I'm waiting solution from developers either Nvidia or ffmpeg staying on previous Nvidia drivers.

ich777, 2024-02-28:
@shperrung reason why I was asking because some new component was introduced like mentioned [here](https://github.com/NVIDIA/libnvidia-container/issues/241#issuecomment-1964187713) and just to rule out `libnvidia-container` and `nvidia-container-toolkit` as the issue but no worries if you can't try that since you already rolled back the driver, most likely it will be the same as on my system.

It's the same here too, `nvidia-smi` is working just fine from a container terminal.

davidtavarez, 2024-03-03:
I can confirm this. Rolling back to **535.161.05** fixed the issue for e.

ich777, 2024-03-03:
@davidtavarez FYI you can use all the way up to **550.40.07** (this is a beta driver but it is working just fine).

ich777, 2024-03-19:
Just to let you guys know, the driver version that was released today (v**550.67**) is working fine again!

I'll let the issue open a few more days and close it.

closed 2024-03-20 by ich777