NVIDIA/open-gpu-kernel-modules #604: Provide PM firmware for the nouveau

PavloIT115, 2024-02-29:
### NVIDIA Open GPU Kernel Modules Version

0

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [X] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Description:    Arch Linux

### Kernel Release

Linux archlinux 6.7.6-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Fri, 23 Feb 2024 16:32:48 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

GP107

### Describe the bug

Hello. Why not provide the necessary firmware (for power management) for a new project? I'm talking about the GM20x, GP10x and GV100 architectures. It is the GP10x and GV100 that suffer from the lack of power management. This will increase your karma and more and more people will want to buy nvidia products. Please provide the firmware for the nouveau project. You have already provided the GSP firmware, for which we are very grateful, it gave a new lease of life. Why not do this for older architectures?

### To Reproduce

Installing nouveau

### Bug Incidence

Always

### nvidia-bug-report.log.gz

.

ttabi, 2024-05-01:
Please read this: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/19

PavloIT115, 2024-05-01:
> Please read this: #19
Hello! I've read this discussion before. What else should I see there?

> I don't know. But, based on past history, we haven't dropped support for that many GPU architectures in one shot. E.g., Volta is much newer than Maxwell, so I doubt we would drop support for all of those together.

The fact that nvidia will not provide firmware, because they are still supported, I understood. I have little hope for nvidia in this regard, that they will change their minds

mtijanic, 2024-05-02:
You have already brought this up previously in #603 and have gotten an answer. There's not much more to say. If things change, I'm sure you'll hear about it.

closed 2024-05-02 by mtijanic

PavloIT115, 2024-05-03:
> You have already brought this up previously in #603 and have gotten an answer. There's not much more to say. If things change, I'm sure you'll hear about it.

I have a gotten an answer? I can only see how they skillfully evade this question in topic after topic. Cool Story, Bob

Ramen-LadyHKG, 2024-09-25:
I'm using `GTX 1060 Mobile 6GB` on `Microsoft Surface Book 2`

The GPU fan does not turn on until 70C degree, is nvidia driver causing this issue?

[https://forums.developer.nvidia.com/t/changing-power-management-limit-is-not-supported-for-gpu-pascal-gtx1060-laptop-mobile-linux-555-nvidia-smi-powerlimit/298873](url)

Why removed power management support in 535 version when it was supported in 525 version?