NVIDIA/open-gpu-kernel-modules #608: build failed on ub22.04 with 550.40.07

SeekPoint, 2024-03-06:
### NVIDIA Open GPU Kernel Modules Version

HEAD detached at 550.40.07

### Operating System and Version

22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb  9 13:32:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### Kernel Release

6.5.0-21-generic 

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Build Command

make modules -j`nproc` NV_VERBOSE=1 DEBUG=1

### Terminal output/Build Log

In file included from generated/g_object_nvoc.h:44,
                 from inc/libraries/nvoc/object.h:28,
                 from inc/libraries/nvoc/runtime.h:36,
                 from inc/kernel/core/prelude.h:42,
                 from inc/kernel/core/core.h:26,
                 from src/kernel/rmapi/resource_desc.c:23:
inc/libraries/nvoc/prelude.h:68:20: error: ‘__nvoc_class_id_CeUtilsApi’ undeclared here (not in a function); did you mean ‘__nvoc_class_def_CeUtilsApi’?
   68 | #define classId(x) __nvoc_class_id_##x
      |                    ^~~~~~~~~~~~~~~~
src/kernel/rmapi/resource_desc.c:81:5: note: in expansion of macro ‘classId’
   81 |     classId(internalClass), \
      |     ^~~~~~~
src/kernel/rmapi/resource_list.h:1583:1: note: in expansion of macro ‘RS_ENTRY’
 1583 | RS_ENTRY(
      | ^~~~~~~~
make[1]: *** [Makefile:202: _out/Linux_x86_64/resource_desc.o] Error 1

mtijanic, 2024-03-06:
(duplicate/repeat of #545 )

Hey, thanks for the report. `CeUtilsApi` was never meant to be exposed, it's an internal debugging tool that was disabled but apparently not fully, and it broke the DEBUG=1 build. We'll fix this in the next release, but as a workaround you can use:
```diff
diff --git a/src/nvidia/src/kernel/rmapi/resource_list.h b/src/nvidia/src/kernel/rmapi/resource_list.h
index f0600a0..71cecc6 100644
--- a/src/nvidia/src/kernel/rmapi/resource_list.h
+++ b/src/nvidia/src/kernel/rmapi/resource_list.h
@@ -1579,7 +1579,7 @@ RS_ENTRY(
     /* Flags                  */ RS_FLAGS_ALLOC_NON_PRIVILEGED | RS_FLAGS_ACQUIRE_GPUS_LOCK,
     /* Required Access Rights */ RS_ACCESS_NONE
 )
-#if RMCFG_CLASS_NV_CE_UTILS && (defined(DEBUG) || defined(DEVELOP))
+#if 0
 RS_ENTRY(
     /* External Class         */ NV_CE_UTILS,
     /* Internal Class         */ CeUtilsApi,
```

SeekPoint, 2024-03-06:
great!

closed 2024-03-06 by SeekPoint

eezstreet, 2024-07-10:
I don't know why this was closed. This is still an issue if you want to compile with `DEBUG=1`.

Can someone reopen this?

norohind, 2024-07-14:
The problem still exists with the latest release. Though suggested above patch allows to build both DEVELOP and DEBUG builds.

Given line in kernel-open/nvidia-uvm/uvm_common.h:34:`TODO Bug 1773100: Figure out the right distinction between develop and debug`, I'm not sure NVIDIA knows distinction between builds, lol.

mtijanic, 2024-07-15:
Fix will be present in 560.xx.

In general, the big issue is with our packaging of the code for distribution. Internally, we have RELEASE, DEBUG and DEVLEOP builds that we use; but when packaging it up for public, we only provide the RELEASE package, which you can then compile as DEBUG if you want. Problem is that we don't have testing coverage of RELEASE-package-built-as-DEBUG, so sometimes these issues creep in. Hopefully that will get resolved soon.