NVIDIA/open-gpu-kernel-modules #617: Dynamic boost can't be enabled

500InternalError, 2024-04-05:
### NVIDIA Open GPU Kernel Modules Version

550.67

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

Linux Zephyrus 6.8.2-arch2-1 #1 SMP PREEMPT_DYNAMIC Thu, 28 Mar 2024 17:06:35 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 4060 Laptop GPU (UUID: GPU-15764050-a2ea-30c8-e844-a5f96fbe0389)

### Describe the bug

When enable dynamic boost by ```sudo systemctl enable nvidia-powerd``` and reboot, nvidia-powerd complains
```
4月 05 10:46:34 Zephyrus /usr/bin/nvidia-powerd[605]: Error in getting the NvPCF static configuration
4月 05 10:46:34 Zephyrus /usr/bin/nvidia-powerd[605]: Failed to initialize Dynamic Boost
4月 05 10:46:34 Zephyrus /usr/bin/nvidia-powerd[605]: Failed to detach GPU id 256
4月 05 10:46:34 Zephyrus /usr/bin/nvidia-powerd[605]: Failed to initialize Dynamic Boost
4月 05 10:46:34 Zephyrus /usr/bin/nvidia-powerd[605]: Failed to detach GPU id 256
```

### To Reproduce

Just ```sudo systemctl enable nvidia-powerd``` and reboot

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/14879290/nvidia-bug-report.log.gz)

timur-tabi, 2024-04-05:
Did you test this with the proprietary driver?  You must do that before opening a bug report here.

500InternalError, 2024-04-05:
Yes I just confirmed that this does happen on proprietary driver

timur-tabi, 2024-04-05:
Then please close this bug and report the problem here instead: https://forums.developer.nvidia.com/c/gpu-graphics/linux/148

As it says on the top of this bug report form, "This issue tracker is only for bugs specific to the open kernel driver."  That is why we ask, "I confirm that this does not happen with the proprietary driver package."

500InternalError, 2024-04-05:
I'm sorry I didn't read that carefully.

closed 2024-04-05 by 500InternalError

500InternalError, 2024-04-12:
Hey it has been 1 week since I asked there but no one respond. What should I do?