NVIDIA/open-gpu-kernel-modules #622: Atomic commit fails if IN_FENCE_FD is set

mahkoh, 2024-04-10:
### NVIDIA Open GPU Kernel Modules Version

550.40.59

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

n/a

### Kernel Release

n/a

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

n/a

### Describe the bug

When an atomic commit contains an IN_FENCE_FD, the nvidia driver rejects the commit with EPERM:

```c
    if (plane_state->fence != NULL || nv_drm_plane_state->fd_user_ptr) {
        if (!nv_dev->supportsSyncpts) {
            return -1;
        }
```

-1 is EPERM.

First of all this is not a correct error code for this situation.

But more importantly, this makes it impossible to synchronize presentation with rendering from the compositor side. The [workaround](https://invent.kde.org/plasma/kwin/-/merge_requests/4770) is to just not use IN_FENCE_FD if the nvidia driver is detected. Kinda absurd since nvidia was the vendor pushing for explicit sync in the first place.

If that is the expected workaround, then maybe the driver should just accept such commits instead of returning an error.

----

For whatever reason, one of my users reports that the commits are nevertheless applied by the nvidia driver even though the atomic commit fails. This might be a separate bug.

### To Reproduce

n/a

### Bug Incidence

Always

### nvidia-bug-report.log.gz

n/a

mahkoh, 2024-04-12:
Thanks for the update.

PS: The issue I mentioned re "commit being applied anyway" turned out to be a bug on my side where I would render directly to the front buffer if too many commits failed in a row.

ids1024, 2024-07-23:
It's nice to see this added in the 560.28.03 beta. Though testing this with cosmic-comp/smithay (https://github.com/Smithay/smithay/pull/1489) on a Intel+NVIDIA graphics laptop it seems to be running into an issue and logging "Failed to initialize semaphore for plane fence" to dmesg. I don't *think* the compositor is doing anything wrong here.

Fxzzi, 2024-08-18:
I can confirm that whilst this was added in the 560 drivers, it does not seem to be implemented fully correctly as ids1024 mentioned. I seem to be getting constant freezes and display lockups which is caused by an issue with the implementation.

cubanismo, 2024-08-20:
@ids1024, @Fxzzi, were you using more than one display? We identified a bug in the implementation in some situations on Wayland desktops with multiple displays. It should be resolved in subsequent 560-series drivers.

Binary-Eater, 2024-08-20:
Here is the relevant forum post for the fix refered to by @cubanismo.

>4769163 - Screen freeze observed with directly scanned out windows on a multi-monitor configuration

https://forums.developer.nvidia.com/t/560-release-feedback-discussion/300830/210

Fxzzi, 2024-08-20:
> @ids1024, @Fxzzi, were you using more than one display? We identified a bug in the implementation in some situations on Wayland desktops with multiple displays. It should be resolved in subsequent 560-series drivers.

I am personally using 2 displays. Hopefully the issue you've resolved is the same as what I was experiencing.

Binary-Eater, 2024-08-22:
We released driver 560.35.03 yesterday, which contains the fix we are referring to for multiple displays.

https://www.nvidia.com/en-us/drivers/details/230918/

gauravjuvekar, 2024-08-24:
Fix released as @Binary-Eater mentions above.

https://forums.developer.nvidia.com/t/560-release-feedback-discussion/300830/309
Internal bug 4769163