NVIDIA/open-gpu-kernel-modules #626: Failed to get power and usage information in nvidia-smi

Reverier-Xu, 2024-04-20:
### NVIDIA Open GPU Kernel Modules Version

550.76

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

Linux 6.8.7-arch1-1.1-g14 #1 SMP PREEMPT_DYNAMIC Fri, 19 Apr 2024 05:48:33 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 4070 Laptop GPU (UUID: GPU-e8108ab1-bcb6-22ff-7cab-c21072716616)

### Describe the bug

`nvidia-smi` shows `ERR!` on the power section and shows always `0%` in usage section.

```
Sat Apr 20 20:20:49 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.76                 Driver Version: 550.76         CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4070 ...    Off |   00000000:01:00.0 Off |                  N/A |
| N/A   55C    P0            ERR! /   80W |       2MiB /   8188MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
```

### To Reproduce

None.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/15048170/nvidia-bug-report.log.gz)

ttabi, 2024-04-20:
Since you said that the problem does not occur with the proprietary driver, please post an nvidia-bug-report.log.gz with the proprietary driver installed, so that we can compare the two.

Reverier-Xu, 2024-04-20:
Using the primary driver:

```
Sun Apr 21 01:33:56 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.76                 Driver Version: 550.76         CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4070 ...    Off |   00000000:01:00.0 Off |                  N/A |
| N/A   55C    P0             13W /   80W |       2MiB /   8188MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
```

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/15049208/nvidia-bug-report.log.gz)

RashadGasimli, 2024-04-26:
same, i can't get any information about power usage with latest drivers (both of proprieatry and open gpu kernel modules)

mtijanic, 2024-05-03:
HI there! The nvidia-bug-report.log from the original post shows:
```
4月 19 22:36:49 Reverier-Arch kernel: NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64  550.67  Release Build  (archlinux-builder@)  
[...snip...]
4月 20 19:55:29 Reverier-Arch kernel: NVRM: API mismatch: the client has the version 550.76, but
                                       NVRM: this kernel module has the version 550.67.  Please
                                       NVRM: make sure that this kernel module and all NVIDIA driver
                                       NVRM: components have the same version.
```
but your nvidia-smi output shows `Driver Version: 550.76`. Are you sure these are from the same run? The mismatched kernelmode/usermode version could easily cause this error. It can happen if you built the wrong version of the open driver from source. dkms shows you built 550.76, but you're still loading 550.67 somehow.

Reverier-Xu, 2024-05-03:
> HI there! The nvidia-bug-report.log from the original post shows:
>
> ```
> 4月 19 22:36:49 Reverier-Arch kernel: NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64  550.67  Release Build [...]

Previously the driver versions were inconsistent probably due to distro packaging issues, sorry for that. so far I can confirm that the driver versions are consistent, but the issue remains. I re-generated a log file using `nvidia-bug-report.sh`:

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/15198004/nvidia-bug-report.log.gz)

I could confirm the dkms driver has the right version `550.76` in package manager.

```
$ pacman -Qi nvidia-open-dkms 
Name            : nvidia-open-dkms
Version         : 550.76-3
Description     : NVIDIA open kernel modules
Architecture    : x86_64
URL             : https://github.com/NVIDIA/open-gpu-kernel-modules
Licenses        : GPL
Groups          : None
Provides        : nvidia-open  NVIDIA-MODULE
Depends On      : nvidia-utils=550.76  libglvnd  dkms
Optional Deps   : None
Required By     : None
Optional For    : None
Conflicts With  : nvidia-open  NVIDIA-MODULE
Replaces        : None
Installed Size  : 77.26 MiB
Packager        : Jan Alexander Steffens (heftig) <heftig@archlinux.org>
Build Date      : Mon Apr 29 00:27:24 2024
Install Date    : Tue Apr 30 22:05:55 2024
Install Reason  : Explicitly installed
Install Script  : No
Validated By    : Signature
```

mtijanic, 2024-05-03:
Thanks for the update! Looking at the new log, this really stands out:
```
[    6.999219] NVRM: GPU at PCI:0000:01:00: GPU-e8108ab1-bcb6-22ff-7cab-c21072716616
[    6.999223] NVRM: Xid (PCI:0000:01:00): 62, pid='<unknown>', name=<unknown>, 20262044 2027f08e 2027df5c 2022ec3e 20281296 2022adbe 00000000 00000000
```
Xid 62 is [PMU_HALT_ERROR](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/550.76/src/common/sdk/nvidia/inc/nverror.h#L68) which would make a lot of the power readings unavailable, but could also lead to bigger system instability. GSP logs confirm as much. I've filed bug 4630466 so that our PMU experts can look into it.

In the meantime, could you please load nvidia.ko with [NVreg_RmMsg=":"](https://github.com/NVIDIA/open-gpu-kernel-modules/discussions/197) and try one more time? That should flood your dmesg with a lot of debug info, and hopefully some of it can help us narrow it down.

Reverier-Xu, 2024-05-07:
Archlinux has pushed nvidia-open 550.78 into the repository, seems that this issue is solved, thanks!