NVIDIA/open-gpu-kernel-modules #634: using Clang to build and go error

dockerg, 2024-04-30:
### NVIDIA Open GPU Kernel Modules Version

550.78

### Operating System and Version

ubuntu 20.04

### Kernel Release

51.1

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Build Command

make CC=clang

### Terminal output/Build Log

[image]

dockerg, 2024-04-30:
Here is a  wrong kernel verson, the match is 5.15.0-88

mtijanic, 2024-05-02:
Hi there! None of those are set anywhere in this project's makefiles. Could you be inheriting them from elsewhere on your system?

Could you please add `NV_VERBOSE=1` to your `make` command and post the full build log?

dockerg, 2024-05-06:
Could you be inheriting them from elsewhere on your system?
------I dont find anywhere setup it, and there are some other project could build with clang
Could you please add NV_VERBOSE=1 to your make command and post the full build log?
------ok,i update full log of build
[make.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/15215177/make.log)

dockerg, 2024-05-06:
And i also try to build it within kernel, it is ok to build when i re-write kbuild and with clang

closed 2024-05-09 by dockerg