NVIDIA/open-gpu-kernel-modules #635: Can't build on fedora 40

ratuka, 2024-05-03:
### NVIDIA Open GPU Kernel Modules Version

545.29.02

### Operating System and Version

Fedora Linux 40 (Workstation Edition)

### Kernel Release

6.8.7-300.fc40.x86_64

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Build Command

dkms autoinstall

### Terminal output/Build Log

DKMS make.log for nvidia-545.29.02 for kernel 6.8.7-300.fc40.x86_64 (x86_64)
2024. máj. 3., péntek, 10:07:23 CEST
make[1]: Entering directory '/usr/src/kernels/6.8.7-300.fc40.x86_64'
  SYMLINK /var/lib/dkms/nvidia/545.29.02/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/545.29.02/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: ioremap_driver_hardened
 CONFTEST: ioremap_driver_hardened_wc
 CONFTEST: ioremap_cache_shared
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_rebar_get_possible_sizes
 CONFTEST: wait_for_random_bytes
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write_has_pointer_pos_arg
 CONFTEST: dma_direct_map_resource
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
 CONFTEST: flush_cache_all
 CONFTEST: vmf_insert_pfn
 CONFTEST: jiffies_to_timespec
 CONFTEST: ktime_get_raw_ts64
[... 600 lines cut ...]
  CC [M]  /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-linux.o
  CC [M]  /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-helper.o
  CC [M]  /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nv-kthread-q.o
In file included from /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-drv.c:64:
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-drv.c:1489:40: error: ‘DRM_UNLOCKED’ undeclared here (not in a function); did you mean ‘VM_LOCKED’?
 1489 |                       DRM_RENDER_ALLOW|DRM_UNLOCKED),
      |                                        ^~~~~~~~~~~~
./include/drm/drm_ioctl.h:155:26: note: in definition of macro ‘DRM_IOCTL_DEF_DRV’
  155 |                 .flags = _flags,                                        \
      |                          ^~~~~~
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-drv.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:115:16: error: initialization of ‘int (*)(struct drm_gem_object *, struct iosys_map *)’ from incompatible pointer type ‘void * (*)(struct drm_gem_object *)’ [-Wincompatible-pointer-types]
  115 |     .vmap    = nv_drm_gem_prime_vmap,
      |                ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:115:16: note: (near initialization for ‘nv_drm_gem_funcs.vmap’)
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:116:16: error: initialization of ‘void (*)(struct drm_gem_object *, struct iosys_map *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *, void *)’ [-Wincompatible-pointer-types]
  116 |     .vunmap  = nv_drm_gem_prime_vunmap,
      |                ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:116:16: note: (near initialization for ‘nv_drm_gem_funcs.vunmap’)
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.o] Error 1
make[2]: *** [/usr/src/kernels/6.8.7-300.fc40.x86_64/Makefile:1939: /var/lib/dkms/nvidia/545.29.02/build] Error 2
make[1]: *** [Makefile:252: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.8.7-300.fc40.x86_64'
make: *** [Makefile:82: modules] Error 2

d3rpp, 2024-05-03:
Ok at least I'm not the only one with this exact error on this exact Distro (I've been pulling my hair out for ages over this lmao)

QiuMike, 2024-05-11:
commit 2798ffcc1d6a788b5769b1fbcf0750dfc06ae98a
Author: Thomas Zimmermann <tzimmermann@suse.de>
Date:   Wed Nov 22 13:09:40 2023 +0100

    drm: Remove locking for legacy ioctls and DRM_UNLOCKED

    Modern DRM drivers acquire ioctl locks by themselves. Legacy ioctls
    for user-space mode setting used to acquire drm_global_mutex. After
    removing the ioctl entry points, also remove the locking code. The only
    legacy ioctl without global locking was VBLANK_WAIT, which has been
    removed as well. Hence remove the related DRM_UNLOCKED flag.

    Signed-off-by: Thomas Zimmermann <tzimmermann@suse.de>
    Reviewed-by: David Airlie <airlied@gmail.com>
    Reviewed-by: Daniel Vetter <daniel@ffwll.ch>
    Acked-by: Alex Deucher <alexander.deucher@amd.com>
    Link: https://patchwork.freedesktop.org/patch/msgid/20231122122449.11588-12-tzimmermann@suse.de

The new kernel has removed the flag DRM_UNLOCKED, so lead to this issue.
So you could simply remove this flag in code and re-compile

zcobol, 2024-05-11:
Release `550.78` builds okay on Fedora 40 with `kernel-6.8.9-300`:

```
zcobol@april:~/open-gpu-kernel-modules-550.78$ modinfo nvidia
filename:       /lib/modules/6.8.9-300.fc40.x86_64/kernel/drivers/video/nvidia.ko
import_ns:      DMA_BUF
alias:          char-major-195-*
version:        550.78
supported:      external
license:        Dual MIT/GPL
firmware:       nvidia/550.78/gsp_tu10x.bin
firmware:       nvidia/550.78/gsp_ga10x.bin
softdep:        pre: ecdh_generic,ecdsa_generic
rhelversion:    9.99
srcversion:     559B949A4E28E211ED21D4D
alias:          pci:v000010DEd*sv*sd*bc06sc80i00*
alias:          pci:v000010DEd*sv*sd*bc03sc02i00*
alias:          pci:v000010DEd*sv*sd*bc03sc00i00*
depends:        
retpoline:      Y
name:           nvidia
vermagic:       6.8.9-300.fc40.x86_64 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Nvidia Driver Kmod Signing MOK
sig_key:        66:35:85:E7:27:E0:37:7A:5C:D9:65:06:B1:B3:5D:81:FC:FE:76:14
sig_hashalgo:   sha256
--- cut ---
```

```
zcobol@april:~/open-gpu-kernel-modules-550.78$ cat /etc/os-release 
NAME="Fedora Linux"
VERSION="40 (Workstation Edition)"
ID=fedora
VERSION_ID=40
--- cut ---
```

ratuka, 2024-05-23:
'make' -j12 NV_EXCLUDE_BUILD_MODULES='' KERNEL_UNAME=6.8.10-300.fc40.x86_64 modules
Error! Bad return status for module build on kernel: 6.8.10-300.fc40.x86_64 (x86_64)
Consult /var/lib/dkms/nvidia/545.29.02/build/make.log for more information.
dkms autoinstall on 6.8.10-300.fc40.x86_64/x86_64 failed for nvidia(10)

DKMS make.log for nvidia-545.29.02 for kernel 6.8.10-300.fc40.x86_64 (x86_64)
2024. máj. 23., csütörtök, 10:12:19 CEST
make[1]: Entering directory '/usr/src/kernels/6.8.10-300.fc40.x86_64'
  SYMLINK /var/lib/dkms/nvidia/545.29.02/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/545.29.02/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: ioremap_driver_hardened
 CONFTEST: ioremap_driver_hardened_wc
 CONFTEST: ioremap_cache_shared
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_rebar_get_possible_sizes
 CONFTEST: wait_for_random_bytes
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write_has_pointer_pos_arg
 CONFTEST: dma_direct_map_resource
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
 CONFTEST: flush_cache_all
 CONFTEST: vmf_insert_pfn
 CONFTEST: jiffies_to_timespec
 CONFTEST: ktime_get_raw_ts64
 CONFTEST: ktime_get_real_ts64
 CONFTEST: full_name_hash
 CONFTEST: pci_enable_atomic_ops_to_root
 CONFTEST: vga_tryget
 CONFTEST: cc_platform_has
 CONFTEST: seq_read_iter
 CONFTEST: unsafe_follow_pfn
 CONFTEST: drm_gem_object_get
 CONFTEST: drm_gem_object_put_unlocked
 CONFTEST: add_memory_driver_managed
 CONFTEST: device_property_read_u64
 CONFTEST: devm_of_platform_populate
 CONFTEST: of_dma_configure
 CONFTEST: of_property_count_elems_of_size
 CONFTEST: of_property_read_variable_u8_array
 CONFTEST: of_property_read_variable_u32_array
 CONFTEST: i2c_new_client_device
[... 582 lines cut ...]
  CC [M]  /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-fence.o
  CC [M]  /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-linux.o
  CC [M]  /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-helper.o
In file included from /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-drv.c:64:
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-drv.c:1489:40: error: ‘DRM_UNLOCKED’ undeclared here (not in a function); did you mean ‘VM_LOCKED’?
 1489 |                       DRM_RENDER_ALLOW|DRM_UNLOCKED),
      |                                        ^~~~~~~~~~~~
./include/drm/drm_ioctl.h:155:26: note: in definition of macro ‘DRM_IOCTL_DEF_DRV’
  155 |                 .flags = _flags,                                        \
      |                          ^~~~~~
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-drv.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:115:16: error: initialization of ‘int (*)(struct drm_gem_object *, struct iosys_map *)’ from incompatible pointer type ‘void * (*)(struct drm_gem_object *)’ [-Wincompatible-pointer-types]
  115 |     .vmap    = nv_drm_gem_prime_vmap,
      |                ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:115:16: note: (near initialization for ‘nv_drm_gem_funcs.vmap’)
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:116:16: error: initialization of ‘void (*)(struct drm_gem_object *, struct iosys_map *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *, void *)’ [-Wincompatible-pointer-types]
  116 |     .vunmap  = nv_drm_gem_prime_vunmap,
      |                ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:116:16: note: (near initialization for ‘nv_drm_gem_funcs.vunmap’)
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.o] Error 1
make[2]: *** [/usr/src/kernels/6.8.10-300.fc40.x86_64/Makefile:1939: /var/lib/dkms/nvidia/545.29.02/build] Error 2
make[1]: *** [Makefile:252: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.8.10-300.fc40.x86_64'
make: *** [Makefile:82: modules] Error 2

ratuka, 2024-06-02:
```
cat /etc/os-release
NAME="Fedora Linux"
VERSION="40 (Workstation Edition)"
ID=fedora
VERSION_ID=40
VERSION_CODENAME=""
PLATFORM_ID="platform:f40"
PRETTY_NAME="Fedora Linux 40 (Workstation Edition)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:40"
DEFAULT_HOSTNAME="fedora"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f40/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=40
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=40
SUPPORT_END=2025-05-13
VARIANT="Workstation Edition"
VARIANT_ID=workstation
```

DKMS make.log for nvidia-545.29.02 for kernel 6.8.11-300.fc40.x86_64 (x86_64)
2024. jún. 2., vasárnap, 12:13:35 CEST
make[1]: Entering directory '/usr/src/kernels/6.8.11-300.fc40.x86_64'
  SYMLINK /var/lib/dkms/nvidia/545.29.02/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/545.29.02/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: ioremap_driver_hardened
 CONFTEST: ioremap_driver_hardened_wc
 CONFTEST: ioremap_cache_shared
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_rebar_get_possible_sizes
 CONFTEST: wait_for_random_bytes
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write_has_pointer_pos_arg
 CONFTEST: dma_direct_map_resource
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
[... 602 lines cut ...]
  CC [M]  /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-modeset.o
  CC [M]  /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-fence.o
In file included from /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-drv.c:64:
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-drv.c:1489:40: error: ‘DRM_UNLOCKED’ undeclared here (not in a function); did you mean ‘VM_LOCKED’?
 1489 |                       DRM_RENDER_ALLOW|DRM_UNLOCKED),
      |                                        ^~~~~~~~~~~~
./include/drm/drm_ioctl.h:155:26: note: in definition of macro ‘DRM_IOCTL_DEF_DRV’
  155 |                 .flags = _flags,                                        \
      |                          ^~~~~~
  CC [M]  /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-linux.o
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-drv.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:115:16: error: initialization of ‘int (*)(struct drm_gem_object *, struct iosys_map *)’ from incompatible pointer type ‘void * (*)(struct drm_gem_object *)’ [-Wincompatible-pointer-types]
  115 |     .vmap    = nv_drm_gem_prime_vmap,
      |                ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:115:16: note: (near initialization for ‘nv_drm_gem_funcs.vmap’)
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:116:16: error: initialization of ‘void (*)(struct drm_gem_object *, struct iosys_map *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *, void *)’ [-Wincompatible-pointer-types]
  116 |     .vunmap  = nv_drm_gem_prime_vunmap,
      |                ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.c:116:16: note: (near initialization for ‘nv_drm_gem_funcs.vunmap’)
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/nvidia/545.29.02/build/nvidia-drm/nvidia-drm-gem.o] Error 1
make[2]: *** [/usr/src/kernels/6.8.11-300.fc40.x86_64/Makefile:1939: /var/lib/dkms/nvidia/545.29.02/build] Error 2
make[1]: *** [Makefile:252: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.8.11-300.fc40.x86_64'
make: *** [Makefile:82: modules] Error 2

zcobol, 2024-06-03:
@ratuka could you try a newer driver, i.e. `555.42.02`? And instead of using `dkms autoinstall` build the driver for the current kernel only. Also, try building the driver outside `dkms`. It looks like a `dkms` issue.

The driver builds okay on my Fedora 40 setup:

```
zcoboll@april:~$ dkms status
nvidia/555.42.02, 6.8.11-300.fc40.x86_64, x86_64: installed
```

ratuka, 2024-07-19:
> @ratuka could you try a newer driver, i.e. `555.42.02`? And instead of using `dkms autoinstall` build the driver for the current kernel only. Also, try building the driver outside `dkms`. It looks [...]

nice, but i'm only dnf update my machine

ptr1337, 2024-07-31:
545 does not support 6.9.11. You need to either patch it or update your NVIDIA Version. 535, 550 and 555 branches do work. 545 is deprecated.