NVIDIA/open-gpu-kernel-modules #642: `follow_pfn()` is removed from kernel

flukejones, 2024-05-21:
### NVIDIA Open GPU Kernel Modules Version

550, 555

### Operating System and Version

Fedora 40

### Kernel Release

6.9+, 6.10 (merge window)

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Build Command

`sudo ./NVIDIA.run`

### Terminal output/Build Log

```
/tmp/selfgz255374/NVIDIA-Linux-x86_64-555.42.02/kernel-open/nvidia/os-mlock.c: In function 'nv_follow_pfn':
/tmp/selfgz255374/NVIDIA-Linux-x86_64-555.42.02/kernel-open/nvidia/os-mlock.c:42:12: error: implicit declaration of function 'follow_pfn'; did you mean 'follow_pte'? [-Wimplicit-function-declaration]
   42 |     return follow_pfn(vma, address, pfn);
      |            ^~~~~~~~~~
      |            follow_pte
make[3]: *** [scripts/Makefile.build:244: /tmp/selfgz255374/NVIDIA-Linux-x86_64-555.42.02/kernel-open/nvidia/os-mlock.o] Error 1
```

### More Info

Easier to link to the commit on github https://github.com/torvalds/linux/commit/cb10c28ac82c9b7a5e9b3b1dc7157036c20c36dd

Yeah I know not a proper stable kernel, but I develop with the kernel regularly and I'm trying to get some patches prepped for 6.10 and 6.11. So I have to reboot to a previous kernel if I want to use the dGPU on my laptop, or use nouveau + NVK.

ttabi, 2024-05-21:
[I wonder why you thought it was okay to file a bug report with ](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/597#issuecomment-2121005245)

mtijanic, 2024-05-22:
> Yeah I know not a proper stable kernel

We do follow the kernel development internally and will catch most such issues early. This is why the bug report form has the stable kernel checkbox - there's no need to report breakages with RCs (unless they're specific to some non standard config). _When_ the fixes will actually make it to a driver release (and which ones) is a different story. But, on the plus side, since you're already building the driver from source, you can just fix it yourself for now. Try this patch:

```diff
diff --git a/kernel-open/nvidia/os-mlock.c b/kernel-open/nvidia/os-mlock.c
index 46f99a1..b8f4100 100644
--- a/kernel-open/nvidia/os-mlock.c
+++ b/kernel-open/nvidia/os-mlock.c
@@ -30,11 +30,21 @@ static inline int nv_follow_pfn(struct vm_area_struct *vma,
                                 unsigned long address,
                                 unsigned long *pfn)
 {
-#if defined(NV_UNSAFE_FOLLOW_PFN_PRESENT)
-    return unsafe_follow_pfn(vma, address, pfn);
-#else
-    return follow_pfn(vma, address, pfn);
-#endif
+    int status = 0;
+    spinlock_t *ptl;
+    pte_t *ptep;
+
+    if (!(vma->vm_flags & (VM_IO | VM_PFNMAP)))
+        return status;
+
+    status = follow_pte(vma, address, &ptep, &ptl);
+    if (status)
+        return status;
+    *pfn = pte_pfn(ptep_get(ptep));
+
+    // The lock is acquired inside follow_pte()
+    pte_unmap_unlock(ptep, ptl);
+    return 0;
 }
 
 /*!
```

I am going to close this issue since it's not a supported configuration, yet.

closed 2024-05-22 by mtijanic

flukejones, 2024-05-23:
Thanks mate.. Still some errors however:
```
6855   │      The kernel was built by: gcc (GCC) 14.1.1 20240507 (Red Hat 14.1.1-1)
6856   │      You are using:           cc (GCC) 14.1.1 20240507 (Red Hat 14.1.1-1)
6857   │      CC [M]  /home/luke/Downloads/NVIDIA-Linux-x86_64-555.42.02/kernel-open/nvidia/os-mlock.o
6858   │    /home/luke/Downloads/NVIDIA-Linux-x86_64-555.42.02/kernel-open/nvidia/os-mlock.c: In function 'nv_follow_pfn':
6859   │    /home/luke/Downloads/NVIDIA-Linux-x86_64-555.42.02/kernel-open/nvidia/os-mlock.c:46:25: error: passing argument 1 of 'follow_pte' from incompatible pointer type [-Wincompatible-pointer-types]
6860   │       46 |     status = follow_pte(vma, address, &ptep, &ptl);
6861   │          |                         ^~~
6862   │          |                         |
6863   │          |                         struct vm_area_struct *
6864   │    In file included from /home/luke/Downloads/NVIDIA-Linux-x86_64-555.42.02/kernel-open/common/inc/nv-pgprot.h:30,

```

mtijanic, 2024-05-23:
You're right, sorry, it should be `vma->vm_mm` as the first argument.

mtijanic, 2024-05-23:
Or.. also get your kernel tree to include https://github.com/torvalds/linux/commit/29ae7d96d166fa08c7232daf8a314ef5ba1efd20

flukejones, 2024-05-23:
@mtijanic really appreciate your help here :)

amie1972, 2024-05-24:
> We do follow the kernel development internally and will catch most such issues early. This is why the bug report form has the stable kernel checkbox - there's no need to report breakages with RCs [...]

Tried in fedora 41 kernel-6.10.rc0 after 20240520,Still can't build success.

bilias, 2024-08-09:
Same problem with 470xx on 6.10 kernel

clmpo, 2024-08-16:
same problem with kernel 6.10.3-200.fc40.x86_64 on Fedora 40

```
2024/08/16 06:58:38 akmodsbuild: /tmp/akmodsbuild.fJiRMLta/BUILD/nvidia-470xx-kmod-470.256.02/_kmod_build_6.10.3-200.fc40.x86_64/nvidia/os-mlock.c:23:12: error: implicit declaration of function ‘follow_pfn’; did you mean ‘follow_pte’? [-Wimplicit-function-declaration]
            2024/08/16 06:58:38 akmodsbuild:    23 |     return follow_pfn(vma, address, pfn);
            2024/08/16 06:58:38 akmodsbuild:       |            ^~~~~~~~~~
            2024/08/16 06:58:38 akmodsbuild:       |            follow_pte
```