NVIDIA/open-gpu-kernel-modules #644: Chromium GPU Process Cannot Start

pravinxor, 2024-05-22:
### NVIDIA Open GPU Kernel Modules Version

555.42.02

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [X] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

Linux 6.9.1-hardened1-1-hardened #1 SMP PREEMPT_DYNAMIC Mon, 20 May 2024 12:54:08 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 4060 Laptop GPU (UUID: GPU-57e1b957-4845-a325-50fb-12cb069295cd)

### Describe the bug

When starting Chromium (or any chromium based program) using the `--ozone-platform=wayland` flag, the GPU process for Chromium cannot start, thus causing hardware acceleration to be completely unavailable- even if the browser is not tasked with performing the hardware acceleration on the Nvidia GPU.

Relevant parts of the Chromium event log:
```
[1182:1182:0521/234950.186727:ERROR:gl_display.cc(520)] : EGL Driver message (Critical) : eglCreateImage failed with 0x00003003
[1182:1182:0521/234950.186814:ERROR:scoped_egl_image.cc(23)] : Failed to create EGLImage: EGL_BAD_ALLOC
[1182:1182:0521/234950.187022:ERROR:native_pixmap_egl_binding.cc(113)] : Unable to initialize binding from pixmap
[1182:1182:0521/234950.187082:ERROR:ozone_image_backing.cc(365)] : OzoneImageBacking::ProduceSkiaGanesh failed to create GL representation
[1182:1182:0521/234950.187126:ERROR:shared_image_manager.cc(232)] : SharedImageManager::ProduceSkia: Trying to produce a Skia representation from an incompatible backing: OzoneImageBacking
GpuProcessHost: The GPU process exited with code 8704.
```

### To Reproduce

- Launch chromium (or equivalent) with `--ozone-platform=wayland` flag, so that chromium is running as a native wayland app (and not via Xwayland)

Note: hardware acceleration is active and performs correctly when Chromium is running via XWayland

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/15398162/nvidia-bug-report.log.gz)

mtijanic, 2024-05-22:
Hi there. Are you _certain_ about this bit:

> Please confirm this issue does not happen with the **proprietary driver** ***(of the same version)***. This issue tracker is only for bugs specific to the open kernel driver.
> [x] I confirm that this does not happen with the proprietary driver package.

I know it's easier to just tick that box than to report the bug to linux-bugs@nvidia.com or on [the forums](https://forums.developer.nvidia.com/c/gpu-graphics/linux/148), but what you are effectively saying is that the bug is in the kernel modules (plausible) _and that it is in the delta between Open and Proprietary_. That delta in 555.xx is very very tiny, so I find it extremely unlikely. Please double-check, otherwise kernel engineers who monitor this tracker (which is for kernel module issues only) could waste time looking in the wrong place.

PS, it seems like in your testing you installed the old kernel module, but still kept the new userspace. This can cause all sorts of issues, so best get that fixed:
```
May 21 23:39:19 zephyrus kernel: NVRM: API mismatch: the client has the version 555.42.02, but
                                 NVRM: this kernel module has the version 550.78.  Please
                                 NVRM: make sure that this kernel module and all NVIDIA driver
                                 NVRM: components have the same version.
```

pravinxor, 2024-05-22:
Thanks for getting back, sorry about the mismatch between the userspace and kernel drivers- I've sorted that out, so that they're both on the save version, however the error still occurs. As for whether this is specific to the open kernel modules, I can confirm that the proprietary does work correctly. I've attached 2 sets of log files (open and proprietary kernel modules). Each set includes an nvidia bug report log, as well as a report from chromium. I'm happy to provide other information or perform debugging as well, if you believe it could help.
[about-gpu-open.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/15405672/about-gpu-open.txt)
[about-gpu-proprietary.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/15405673/about-gpu-proprietary.txt)
[nvidia-bug-report-open.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/15405674/nvidia-bug-report-open.log.gz)
[nvidia-bug-report-proprietary.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/15405675/nvidia-bug-report-proprietary.log.gz)

mtijanic, 2024-05-22:
Thanks for double-checking. That is _very_ surprising to me, I don't see anything in the logs suggesting any meaningful difference (except maybe some external monitor unplugging - was the test for both with the same monitors attached).

We'll try to repro this internally. It's very concerning that there's a functional difference here. Thanks!

pravinxor, 2024-05-22:
Between the two tests I most recently posted, the display configuration was exactly the same. However between the recent two tests and the first test I posted, one of the attached displays was different. Though, I don't believe this is a significant factor, since the issue occurs regardless of the displays configuration.

pravinxor, 2024-06-26:
I just wanted to update this thread with a small change that has happened between then and now. The log messages from EGL appear a little different.

[about-gpu-2024-06-26T18-24-54-455Z.txt](https://github.com/user-attachments/files/15993761/about-gpu-2024-06-26T18-24-54-455Z.txt)
[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/15993763/nvidia-bug-report.log.gz)

Hanssen0, 2024-10-27:
Reproduced with proprietary driver

nvidia-dkms 560.35.03-18

Arch Linux

Linux Hanssen-Linux 6.11.5-arch1-1-g14 #1 SMP PREEMPT_DYNAMIC Sun, 27 Oct 2024 17:01:27 +0000 x86_64 GNU/Linux

GPU 0: NVIDIA GeForce RTX 4060 Laptop GPU (UUID: GPU-5737ef92-c92d-bbb1-c337-b01b9b5e7640)

```
[12784:12784:1028/033113.573410:ERROR:angle_platform_impl.cc(44)] ImageEGL.cpp:112 (operator()): eglCreateImage failed with 0x00003003
ERR: ImageEGL.cpp:112 (operator()): eglCreateImage failed with 0x00003003
[12784:12784:1028/033113.573596:ERROR:scoped_egl_image.cc(23)] Failed to create EGLImage: EGL_SUCCESS
[12784:12784:1028/033113.573804:ERROR:native_pixmap_egl_binding.cc(118)] Unable to initialize binding from pixmap
[12784:12784:1028/033113.573941:ERROR:ozone_image_backing.cc(309)] OzoneImageBacking::ProduceSkiaGanesh failed to create GL representation
[12784:12784:1028/033113.574008:ERROR:shared_image_manager.cc(255)] SharedImageManager::ProduceSkia: Trying to produce a Skia representation from an incompatible backing: OzoneImageBacking
[12784:12784:1028/033113.574139:ERROR:gpu_service_impl.cc(1161)] Exiting GPU process because some drivers can't recover from errors. GPU process will restart shortly.
[12735:12782:1028/033113.584699:ERROR:shared_image_interface_proxy.cc(134)] Buffer handle is null. Not creating a mailbox from it.
```

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/17535205/nvidia-bug-report.log.gz)

Hanssen0, 2024-10-27:
> Reproduced with proprietary driver

Same on the open driver.

nvidia-open-dkms 560.35.03-18

Hanssen0, 2024-10-27:
For anyone suffering from this problem and reaching here, I bypassed this problem by adding `--disable-gpu-compositing` flag to Chrome as a temporary solution.

Check [Hardware acceleration in electron apps on nvidia doesn't work](https://www.reddit.com/r/archlinux/comments/1due6ni/hardware_acceleration_in_electron_apps_on_nvidia/) for more information.

ghost91-, 2024-12-09:
I'm seeing a similar issue on my system:

nvidia-open-dkms 565.77
Arch Linux
Linux pollux 6.12.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 06 Dec 2024 11:15:43 +0000 x86_64 GNU/Linux
GPU 0: NVIDIA GeForce RTX 4070 Laptop GPU (UUID: GPU-96f395c0-6e49-a3ea-23da-e8da84026b54)

```
[123573:123573:1209/114113.965607:ERROR:angle_platform_impl.cc(44)] ImageEGL.cpp:112 (operator()): eglCreateImage failed with 0x00003003
ERR: ImageEGL.cpp:112 (operator()): eglCreateImage failed with 0x00003003
[123573:123573:1209/114113.965717:ERROR:scoped_egl_image.cc(23)] Failed to create EGLImage: EGL_SUCCESS
[123573:123573:1209/114113.965774:ERROR:native_pixmap_egl_binding.cc(118)] Unable to initialize binding from pixmap
[123573:123573:1209/114113.965832:ERROR:ozone_image_backing.cc(309)] OzoneImageBacking::ProduceSkiaGanesh failed to create GL representation
[123573:123573:1209/114113.965882:ERROR:shared_image_manager.cc(257)] SharedImageManager::ProduceSkia: Trying to produce a Skia representation from an incompatible backing: OzoneImageBacking
[123573:123573:1209/114113.965965:ERROR:gpu_service_impl.cc(1154)] Exiting GPU process because some drivers can't recover from errors. GPU process will restart shortly.
[123534:123534:1209/114113.993989:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=8704
[123683:10:1209/114114.062669:ERROR:command_buffer_proxy_impl.cc(131)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
[123743:123743:1209/114114.250986:ERROR:angle_platform_impl.cc(44)] ImageEGL.cpp:112 (operator()): eglCreateImage failed with 0x00003003
ERR: ImageEGL.cpp:112 (operator()): eglCreateImage failed with 0x00003003
[123743:123743:1209/114114.251071:ERROR:scoped_egl_image.cc(23)] Failed to create EGLImage: EGL_SUCCESS
[123743:123743:1209/114114.251120:ERROR:native_pixmap_egl_binding.cc(118)] Unable to initialize binding from pixmap
[123743:123743:1209/114114.251160:ERROR:ozone_image_backing.cc(309)] OzoneImageBacking::ProduceSkiaGanesh failed to create GL representation
[123743:123743:1209/114114.251189:ERROR:shared_image_manager.cc(257)] SharedImageManager::ProduceSkia: Trying to produce a Skia representation from an incompatible backing: OzoneImageBacking
[123743:123743:1209/114114.251247:ERROR:gpu_service_impl.cc(1154)] Exiting GPU process because some drivers can't recover from errors. GPU process will restart shortly.
[123534:123534:1209/114114.270095:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=8704
[123782:123782:1209/114114.498344:ERROR:angle_platform_impl.cc(44)] ImageEGL.cpp:112 (operator()): eglCreateImage failed with 0x00003003
ERR: ImageEGL.cpp:112 (operator()): eglCreateImage failed with 0x00003003
[123782:123782:1209/114114.498409:ERROR:scoped_egl_image.cc(23)] Failed to create EGLImage: EGL_SUCCESS
[123782:123782:1209/114114.498433:ERROR:native_pixmap_egl_binding.cc(118)] Unable to initialize binding from pixmap
[123782:123782:1209/114114.498465:ERROR:ozone_image_backing.cc(309)] OzoneImageBacking::ProduceSkiaGanesh failed to create GL representation
[123782:123782:1209/114114.498486:ERROR:shared_image_manager.cc(257)] SharedImageManager::ProduceSkia: Trying to produce a Skia representation from an incompatible backing: OzoneImageBacking
[123782:123782:1209/114114.498527:ERROR:gpu_service_impl.cc(1154)] Exiting GPU process because some drivers can't recover from errors. GPU process will restart shortly.
[123534:123534:1209/114114.513856:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=8704
[123648:7:1209/114114.652418:ERROR:command_buffer_proxy_impl.cc(131)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
```

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/18060132/nvidia-bug-report.log.gz)

kierun, 2025-07-17:
I am seeing the same thing on Fedora 42. 

@Hanssen0 quoth:
> For anyone suffering from this problem and reaching here, I bypassed this problem by adding `--disable-gpu-compositing` flag to Chrome as a temporary solution.

This worked for me. Thank you so much! 🤩