NVIDIA/open-gpu-kernel-modules #645: Black screen on 555.x.x, Gnome 46, Fedora 40

flukejones, 2024-05-23:
### NVIDIA Open GPU Kernel Modules Version

555.42.02 installer

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Fedora Linux 40 (Workstation Edition)

### Kernel Release

Linux fedora 6.9.0+

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 4060 Laptop GPU (UUID: GPU-d923d028-392d-8c7d-fb6d-7bcf1d5c8a58)

### Describe the bug

External dGPU output is blank or undetected on the 555.x.x drivers

This occurs in both Gnome and Cosmic. Although Gnome is showing no screen at all while I get https://github.com/pop-os/cosmic-comp/issues/497 in cosmic

NOTE: both sessions are wayland only. Gnome + x11 does not have this issue.

### To Reproduce

Best as I can tell it is universal across the 555.x.x release for Wayland sessions

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/15414957/nvidia-bug-report.log.gz)

stefanos-kalantzis, 2024-06-24:
I had a working system for a few days, and then this started happening consistently. I'm on the KDE spin of fedora 40 (Wayland).

Hyphaed, 2024-07-13:
same in Ubuntu 24.04 on Xorg Session

NVIDIA GeForce RTX 3070
NVIDIA-SMI 555.52.04              
Driver Version: 555.52.04      
CUDA Version: 12.5
Gnome 46.2 
xserver-xorg 1:7.7+23ubuntu3
Mutter 46.0