NVIDIA/open-gpu-kernel-modules #649: 6.10 RC kernel fails to build DKMS

ryzendew, 2024-05-30:
### NVIDIA Open GPU Kernel Modules Version

555.42.02

### Operating System and Version

Fedora 40

### Kernel Release

6.10 RC1

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Build Command

./install.sh install to build the tkg kernel 

### Terminal output/Build Log

[image]

### More Info

This issue matters and i hope you are able to fix it because 6.10 holds alot for amd cpu users

ttabi, 2024-05-30:
`□ I am running on a stable kernel release.`

Why did you ignore this?  There is a reason we ask this question.

mtijanic, 2024-05-30:
Closing as not supported. See https://github.com/NVIDIA/open-gpu-kernel-modules/issues/642#issuecomment-2124213782

closed 2024-05-30 by mtijanic

ryzendew, 2024-05-30:
> `□ I am running on a stable kernel release.`
> 
> Why did you ignore this? There is a reason we ask this question.

because it's silly rule as you were told long ago .. you guys need to have better community support here cause this a joke like it was with the 6.8 kernel. remember it takes less effort to improve and be better than to just say no to reports.

veldenb, 2024-06-17:
For what it's worth, on the Ubuntu mainline 6.10-rc4 kernel this patch worked for me:
```
diff -up a/nvidia/os-mlock.c b/nvidia/os-mlock.c
--- a/nvidia/os-mlock.c	2024-05-31 04:07:05.148392927 +0200
+++ b/nvidia/os-mlock.c	2024-05-31 04:09:46.722091133 +0200
@@ -36,11 +36,29 @@ static inline int nv_follow_pfn(struct v
                                 unsigned long address,
                                 unsigned long *pfn)
 {
+#if LINUX_VERSION_CODE < KERNEL_VERSION(6, 10, 0)
 #if defined(NV_UNSAFE_FOLLOW_PFN_PRESENT)
     return unsafe_follow_pfn(vma, address, pfn);
 #else
     return follow_pfn(vma, address, pfn);
 #endif
+#else
+    int status = 0;
+    spinlock_t *ptl;
+    pte_t *ptep;
+
+    if (!(vma->vm_flags & (VM_IO | VM_PFNMAP)))
+        return status;
+
+    status = follow_pte(vma, address, &ptep, &ptl);
+    if (status)
+        return status;
+    *pfn = pte_pfn(ptep_get(ptep));
+
+    // The lock is acquired inside follow_pte()
+    pte_unmap_unlock(ptep, ptl);
+    return 0;
+#endif
 }
 
 /*!
```

Based on this patch from OpenMandriva: https://github.com/OpenMandrivaAssociation/nvidia/blob/master/nvidia-kernel-6.10.patch