NVIDIA/open-gpu-kernel-modules #650: Low fps on external monitor connected to nvidia hdmi port

tm4ig, 2024-06-02:
### NVIDIA Open GPU Kernel Modules Version

555.42.02

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.9.3

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

AMD Radeon 780M IGPU + NVIDIA GeForce RTX 4060 Laptop DGPU (UUID: GPU-36f796c8-ee38-5be5-08a5-c7d8635be2d6)

### Describe the bug

I have asus laptop with AMD Radeon 780M IGPU + NVIDIA GeForce RTX 4060 Mobile MAX-Q DGPU and KDE6 and Wayland session.
Laptop monitor connected to AMD GPU, external monitor connected to nvidia GPU (HDMI port).
When I run glxgears benchmark test in kde 6 wayland session on 555.42.02 nvidia-open driver or 555.42.02 nvidia proprietary driver without nvidia.NVreg_EnableGpuFirmware=0 kernel option on my external monitor connected to nvidia hdmi port I have low fps framerate equal to half the screen refresh rate (in my case I have only ~37-38 fps when external screen refresh rate 75).
This looks like a  bug https://bugs.kde.org/show_bug.cgi?id=452219 but it nvidia diriver regression because on nvidia-open 550.xx driver or nvidia proprietary driver 555.42 drver wih  nvidia.NVreg_EnableGpuFirmware=0 kernel option I have normal framerate on extenal monitor.
I can not use nvidia proprietary driver 550.xx or  555.42 because it causes the kernel to panic https://forums.developer.nvidia.com/t/series-550-freezes-laptop/284772/135 and nvidia can not fix this problem more than 3 monthes.
I do not want use nvidia open driver 550.xx because with this driver and external monitor I have very large cpu utilization for kwin_wayland  proccess

### To Reproduce

1) Connect external monitor to nvidia hdmi port, wayland, kwin 6 and  nvidia-open driver 555.42.02 
2) run glxgears on external monitor

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/15525756/nvidia-bug-report.log.gz)

tm4ig, 2024-06-06:
similar problem https://forums.developer.nvidia.com/t/wayland-external-monitor-refresh-rate-issue/290752
But in my case problem with nvidia-open 555 driver or (nvidia 555 closed driver with GSP Firmware).
With nvidia-open 550 driver (or nvidia proprietary driver 555 with NVreg_EnableGpuFirmware=0 )I have normal framerate on external monitor

tm4ig, 2024-07-10:
With nvidia proprietary linux driver 555, nvidia.NVreg_EnableGpuFirmware=0 kernel option, Wayland and KDE Plasma 6.1 I have low CPU usage (around 5-20% for one core activity for kwin_wayland) and hight frame rate (around 70-75 fps with monitor refresh rate 75 Hz) on external monitor connected to nvidia HDMI port but I have kernels panics as in case https://forums.developer.nvidia.com/t/series-550-freezes-laptop/284772/210

With nvidia open linux driver 555, Wayland and KDE Plasma 6.1 I have hight CPU usage (around 20-80% for one core with activity for kwin_wayland) and low frame rate (around 65-70 fps with monitor refresh rate 75 Hz) on external monitor connected to nvidia HDMI port but I have not kernels panics as in case https://forums.developer.nvidia.com/t/series-550-freezes-laptop/284772/210

With nvidia proprietary linux driver 555, nvidia.NVreg_EnableGpuFirmware=1 kernel option, Wayland and KDE Plasma 6.1 I have hight CPU usage (around 20-80% for one core with activity for kwin_wayland) and low frame rate (around 65-70 fps with monitor refresh rate 75 Hz) on external monitor connected to nvidia HDMI port and I have kernels panics as in case https://forums.developer.nvidia.com/t/series-550-freezes-laptop/284772/210

So on nvidia proprietary driver (full closed mode) I have best performance, but I also have kernel panics
On nvidia open driver I have low performance, but have not kernel panics
On nvidia proprietary driver with Gpu Firmware enabled (default for 555) I have low performance and I have kernels panics.

Nvidia can not fix kernel panics with hybrid graphics five months

Kimiblock, 2024-07-31:
Reverse Prime do get tricky on my machine. It’s a known issue for a long time.

Enverbalalic, 2024-08-07:
Are there any updates to this issue ? I'm having the same problem with an external monitor connected via DisplayPort (USB-C). Specific configuration is a Ryzen 7000 laptop with a RTX 4080, the external display is basically unusable

Kimiblock, 2024-09-09:
Had to use only the dedicated GPU for now. Power consumption is insane.

NGStaph, 2024-10-01:
Relevant discussions on:

1. https://gitlab.gnome.org/GNOME/mutter/-/issues/3461
2. https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/4027#note_2231341
3. https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/1441#note_2236368

lucasslima, 2024-10-08:
Hi all, we also have a thread in the nvidia forums related to the same issue: https://forums.developer.nvidia.com/t/nvidia-please-get-it-together-with-external-monitors-on-wayland/301684/30

mtijanic, 2024-10-09:
This is being tracked as NV bug 4830125

NGStaph, 2024-10-28:
i use usb4 displayport/thunderbolt and funnily enough, hit the advertised target refresh rates on both monitors in KDE plasma, but not on Gnome.

mtijanic, 2024-11-25:
Hey there! Sorry, the NV bug is private, but we can provide public updates here. We have a machine with local repro and are actively working on it, but we don't have a root cause yet. The issue seems to be related to the power savings feature of the GSP or one of the display-specific components. Going to a lower (more power) pstate makes the issue go away.

That's not a solution though, and we're working on understanding the exact cause and how to fix it without consuming excess power. Will update here when we have more to share. And if the fix involves only kernel-side changes, we can post the patches as well.

Thanks for the patience!

lucasslima, 2024-11-26:
> Hey there! Sorry, the NV bug is private, but we can provide public updates here. We have a machine with local repro and are actively working on it, but we don't have a root cause yet. The issue [...]

I find this very unlikely. I'm using the closed source drivers with the GSP disabled so I don't think it's related to that, while using the open module does makes the framerate on the external monitor more jittery. Setting the clocks to max also makes no difference, and I can't find how to set the power limit using nvidia-settings in Wayland/nvidia-smi.

mtijanic, 2024-11-26:
> I find this very unlikely. I'm using the closed source drivers with the GSP disabled so I don't think it's related to that, while using the open module does makes the framerate on the external monitor more jittery.

Hi there. Am I understanding correctly that you are also seeing the "half-FPS on external monitor" issue with GSP disabled too? That doesn't match our experiments.

While debugging this we did find some causes of jitter where individual frames would take longer, but that wasn't the core issue. Eventually we got to the point where, on GSP only, running with <=P4 pstate a monitor runs at 60.0fps, and with >=P5 at 30.000fps. This is the issue we are debugging and that is tracked here and in NV bug 4830125. 

>  Setting the clocks to max also makes no difference, and I can't find how to set the power limit using nvidia-settings in Wayland/nvidia-smi.

You can poll your pstate with
```
nvidia-smi --query-gpu="pstate" --format=csv --loop-ms=1000
```

and any of the following should cause it to change:
- Running a graphics intensive app, such as ` __GL_SYNC_TO_VBLANK=0 glxgears` (disabling vsync makes it render at max fps and warms up the GPU) should set it to P0
- Running any CUDA app, such as `mpv --hwdec=nvdec-copy video.mp4` will set it to P2
- This little app should set it to P0: https://gist.github.com/mtijanic/9c129900bfba774b39914ad11b0041f6

ngoquang2708, 2024-11-27:
@lucasslima What happen when you move the Firefox window to the same monitor as glxgears?

mtijanic, 2024-11-27:
@lucasslima thanks for that video. I don't think this is the same issue that we're talking about here. From first glance, it looks like a weird interaction between firefox, kwin and the NVIDIA usermode drivers (related to explicit sync?).

Could you please send this video and the output of your `nvidia-bug-report.sh` to `linux-bugs@nvidia.com` and then that will get routed internally to where it needs to be, since this repo doesn't seem to be right place for it. Oh, and please mention the make&model of the external monitor, I don't know if it's caught in the bug report log.

Thanks!

lucasslima, 2024-11-27:
Glad for be helpful. 

> @lucasslima What happen when you move the Firefox window to the same monitor as glxgears?

Both are running on the same screen, it just happens to be ultra-wide.

Simpuis, 2024-11-28:
Using the program above to force P0 fixes the problem for me, although just like @lucasslima on a KDE wayland session with firefox I also get said UFO problem, but without firefox both monitors run smoothly to my eyes.

busybox11, 2024-12-01:
I have also experienced similar issues on both proprietary and open NVIDIA modules, but only on high refresh rate monitors - I do experience subtle performance issues on my 1080p60 monitor very occasionally, but ALWAYS on high refresh rate ones - 1440p ultrawide 180hz and 1440p 16/9 144hz.

Both GNOME and KDE have this issue, couldn't test on Hyprland as it straight up doesn't work with NVIDIA.

clone-888, 2025-01-06:
Just want to mention for the devs that version 550.76 of the driver does not have this issue. I found it to be the only driver version that doesn't.

virusapex, 2025-01-20:
@mtijanic Good day! Sorry to bother again, but was there any progress on this?

virusapex, 2025-01-27:
> Hi @virusapex , sorry, I was out of office over the holidays, for an unexpectedly long period of time. Other people have since taken over this issue; I see progress but it still hasn't been fully closed down. I'll follow up and get back here a bit later.

Thank you for notifying us!

NGStaph, 2025-02-03:
Wanted to check if anyone has tested version 570.86.16 of the nvidia beta driver, is the issue fixed/ can i use  `nvidia.NVreg_EnableGpuFirmware=0` with this version of the 'open' driver.

virusapex, 2025-02-05:
> 570.86.16 seems to mostly fix the issue for me, which is to say that performance (with the beta drivers)  is almost as good as 565 with `nvidia.NVreg_EnableGpuFirmware=0`.  
> 
> others still seem to be facing difficulties (https://forums.developer.nvidia.com/t/570-release-feedback/321956/13) but this has not been my case 

As far as I know, you can't disable GSP firmware on Open kernel module, so `nvidia.NVreg_EnableGpuFirmware=0` can't be used in the current situation. As for 570.86.16, I tried but still got a sufficient lag when moving windows around the screen.

uentity, 2025-02-09:
Agree with @virusapex, low rate lag is not fixed in 570.86

x0wllaar, 2025-02-19:
> Using the program above to force P0 fixes the problem for me, although just like [@lucasslima](https://github.com/lucasslima) on a KDE wayland session with firefox I also get said UFO problem, but without firefox both monitors run smoothly to my eyes.

It's probably because #743 also plays a role here (and it's more of a userspace issue maybe?)

mtijanic, 2025-02-19:
Quick update here: We now think we fully understand the issue. The short version is that when the GPU is in a low power state (e.g. idle desktop) then there's an extra bit of latency between the various components when a per-frame notification is delivered from the display hardware all the way up to the compositor. This causes us to just miss the vblank interval and drop a frame, which is why the framerate gets cut exactly in half. These delays are bigger with GSP in the picture than when it is disabled, so the issue does not reproduce on the legacy driver mode\*.

We're still looking into what a proper fix will be, but the two short term workarounds we found are to either run the GPU at a slightly higher pstate (P4-P5 seems sufficient, depending on the HW) or to deliver the events to kwin on a fixed timer instead.

Will keep you posted.

---
\* It _could_ with a high refresh rate monitor, unless driving such a high refresh rate didn't bump up the power draw anyway

x0wllaar, 2025-03-01:
From https://www.nvidia.com/en-us/drivers/details/241089/:

> Highlights since R570 Beta Release, 570.86.16
> ...
> Fixed an issue which caused stuttering and performance issues when scrolling windows in Wayland with GSP firmware enabled. 

Is this the fix for this issue? Thank you!

MatMercer, 2025-03-01:
> From https://www.nvidia.com/en-us/drivers/details/241089/:
> 
> > Highlights since R570 Beta Release, 570.86.16
> > ...
> > Fixed an issue which caused stuttering and performance issues when scrolling windows in Wayland with GSP firmware enabled.
> 
> Is this the fix for this issue? Thank you!

Doesn't seem like. Just updated to this version now and nothing changed

MatMercer, 2025-03-01:
Update:

At Plasma enabling adaptative sync makes the mouse cursor laggy but windows snappy!

lucasslima, 2025-03-02:
> From https://www.nvidia.com/en-us/drivers/details/241089/:
> 
> > Highlights since R570 Beta Release, 570.86.16
> > ...
> > Fixed an issue which caused stuttering and performance issues when scrolling windows in Wayland with GSP firmware enabled.
> 
> Is this the fix for this issue? Thank you!

I believe it should be a fix for when you are using the nvidia GPU as primar.

urbenlegend, 2025-03-03:
Hello all, I am coming from bug https://github.com/NVIDIA/open-gpu-kernel-modules/issues/777 because I am still seeing stutter issues even though it was closed today. Just wanted to add a bit more data.

Doing the glxgears test that OP mentioned on my 120Hz display, I noticed that with GSP on, the framerate wouldn't exactly half, but there was a pronounced decrease. FPS would drop from 119 to 112:

```
[arch-desktop ~]$ glxgears
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
600 frames in 5.0 seconds = 119.804 FPS
592 frames in 5.0 seconds = 118.253 FPS
562 frames in 5.0 seconds = 112.125 FPS # Starts to drop here, causing frame pacing issues
562 frames in 5.0 seconds = 112.297 FPS
562 frames in 5.0 seconds = 112.266 FPS
563 frames in 5.0 seconds = 112.457 FPS
559 frames in 5.0 seconds = 111.746 FPS
590 frames in 5.0 seconds = 117.883 FPS
594 frames in 5.0 seconds = 118.682 FPS # Sometimes it recovers slightly like shown here, but seems to dip below 119
594 frames in 5.0 seconds = 118.681 FPS
595 frames in 5.0 seconds = 118.881 FPS
594 frames in 5.0 seconds = 118.680 FPS
596 frames in 5.0 seconds = 119.058 FPS
595 frames in 5.0 seconds = 118.900 FPS
594 frames in 5.0 seconds = 118.655 FPS
594 frames in 5.0 seconds = 118.702 FPS
594 frames in 5.0 seconds = 118.678 FPS
593 frames in 5.0 seconds = 118.478 FPS
```

Compare this to GSP off:
```
[arch-desktop ~]$ glxgears
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
608 frames in 5.0 seconds = 121.519 FPS
606 frames in 5.0 seconds = 121.127 FPS
603 frames in 5.0 seconds = 120.516 FPS
601 frames in 5.0 seconds = 120.179 FPS
601 frames in 5.0 seconds = 120.022 FPS
600 frames in 5.0 seconds = 119.965 FPS # Stabilizes at 119 and is usually pretty consistent
600 frames in 5.0 seconds = 119.831 FPS
599 frames in 5.0 seconds = 119.691 FPS
599 frames in 5.0 seconds = 119.713 FPS
599 frames in 5.0 seconds = 119.767 FPS
599 frames in 5.0 seconds = 119.793 FPS
600 frames in 5.0 seconds = 119.809 FPS
600 frames in 5.0 seconds = 119.844 FPS
600 frames in 5.0 seconds = 119.805 FPS
```

This seems related to the issue I posted about in https://github.com/NVIDIA/open-gpu-kernel-modules/issues/777, where simply dragging windows around felt stuttery and badly frame paced. Indeed, when glxgears reports 112, the windows are stuttery. In the rare times it recovers back to 118-119, dragging windows around is smoother. Sometimes it never recovers and just gets stuck at 112.

yvesb04, 2025-03-05:
With GSP enabled (default settings) and a fresh install of 570.86.16, in an external monitor connected by hdmi, running at 144hz, with the laptop cable plugged in (Legion 5 Pro 2022, Ryzen 5600h, RTX 3060), I am getting these values in glxgears:
```sh
yves@Legion5Pro:~$ __GL_SYNC_TO_VBLANK=0 glxgears
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
432 frames in 5.0 seconds = 86.145 FPS
415 frames in 5.0 seconds = 82.848 FPS
473 frames in 5.0 seconds = 94.588 FPS
481 frames in 5.0 seconds = 96.135 FPS
496 frames in 5.0 seconds = 99.031 FPS
376 frames in 5.0 seconds = 75.077 FPS
488 frames in 5.0 seconds = 97.471 FPS
```
Using that boost c script that was commented somewhere around here brings power to P0 and dragging windows is much better, glxgears still has bad performance. Performance is better with gsp off (still not perfect and under 140fps most of the time. Moving the glxgears window to the main laptop display instantly locks in at 144fps.
I've been having this issue since I got this laptop on all ports connected to the Nvidia Card (The usb 4.0 connected to the igpu works perfectly). Another fix for me is to disable hybrid graphics in UEFI and run on Dedicated Mode only, but this results in half of my programs crashing on launch.

yvesb04, 2025-03-14:
Hello, I'm sorry to bother @mtijanic, is there any update on this issue?

mtijanic, 2025-03-14:
We have a fix in testing, but I can't really say which release it will show up in. Probably not 570.xx but one of the later major ones, so it gets sufficient QA coverage.

yvesb04, 2025-03-14:
I seem to have completely fixed this issue on my end, without the need of running the ./perf_boost that was made available a while back. I can't exactly pinpoint what fixed it, but a combination of all of these made glxgears run almost always locked at 144hz (and no more stuttering while draggins windows). This with the latest PPA proprietary drivers, 570.86.16.
I should mention too that the laptop is a Legion 5 Pro, running in Hybrid Graphics mode.
1. `sudo prime-select nvidia` 
2. Going into windows (I have a dualboot installation), and pressing Fn + Q to change into performance mode (can tell by the littled LED indicator in the power button becoming red).
3. Reboot to linux
4. Change Power mode to performance (using ubuntu 24.10)
5. no more sutter.
For some reason the performance mode in ubuntu never seemed to do much to improve performance but I think it in Windows was what made the issue go away for me after almost 2 years!

ps:  I noticed because in linux i always have the red indicator (perf mode), but I booted into windows for some security updates and noticed it changed to blue (power saver), then I changed it to red, and when booting back to linux everything was buttery smooth.

vanvugt, 2025-04-04:
I think the bug title here needs to make it more obvious this is about KDE.

For Ubuntu/GNOME please see https://bugs.launchpad.net/bugs/2105500 which was declared fixed in Mutter 47.4 onward. The fixes were:
* https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/4015
* https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/4027

NGStaph, 2025-04-04:
@vanvugt is that the case? I still experience low frame rates on my external monitor (on gnome 48) unless I switch to the proprietary drivers and `NVreg_EnableGpuFirmware=0`

vanvugt, 2025-04-04:
There are many factors here. The links I gave are just the biggest ones in GNOME. I agree there has also been some noticeable performance variation between different Nvidia drivers in the past year.

I want to say that we should be careful to avoid conflating other peoples' bugs with our own experiences, but now I might be guilty myself.

Kimiblock, 2025-04-04:
@vanvugt GNOME is still at very low framerate on both the internal and external monitors. Applications on the internal screen is somewhat okay, but those on external screen are not and stays in low framerate.

vanvugt, 2025-04-04:
Since copying between GPUs is different for each desktop environment, please be sure to report your issues to the desktop environment first. If you are using GNOME then please go to https://gitlab.gnome.org/GNOME/mutter/-/issues

Kimiblock, 2025-04-04:
@vanvugt I'm using GNOME 48.1. And [this issue](https://gitlab.gnome.org/GNOME/mutter/-/issues/3461) seems to be exactly the case.

Regarding differences between DEs, from my testing KDE does much better on the internal screen but suffers on framerate of the external screen too.

JelteF, 2025-04-16:
> We have a fix in testing, but I can't really say which release it will show up in. Probably not 570.xx but one of the later major ones, so it gets sufficient QA coverage.

@mtijanic any more updates on when the fix will be released?

NGStaph, 2025-04-26:
beta driver (575.51) doesn't seem to fix the issue for me. would it be possible for someone to test and confirm?

1vanK, 2025-04-30:
I have this problem on Desktop

queenbiscuit311, 2025-05-04:
> beta driver (575.51) doesn't seem to fix the issue for me. would it be possible for someone to test and confirm?

In my case the beta driver made the issue come up again after id managed to work my way around it on kde wayland. Trying to downgrade now to see if I can un-break it. Starting to feel that this issue's just never getting fixed ngl, or if it is its going to be a lot longer

NGStaph, 2025-06-06:
@mtijanic sorry to bother you again but when could we expect a fix for this? Or has this already been fixed in 575.57?

queenbiscuit311, 2025-06-06:
### 

> [@mtijanic](https://github.com/mtijanic) sorry to bother you again but when could we expect a fix for this? Or has this already been fixed in 575.57?

Still broken, I just gave up and disabled my iGPU. I'm hoping it's just going through development and testing rn.

stdereka, 2025-06-12:
Okay, until a proper fix arrives, the two workarounds are switching P-states and disabling the iGPU. I managed to automate the process and wrote a script that switches P-states when the external monitor is connected. When the monitor is disconnected, the clocks are reset.

`/etc/udev/rules.d/80-nvidia-hotplug.rules`
```
ACTION=="change", SUBSYSTEM=="drm", ENV{HOTPLUG}=="1", RUN+="/usr/bin/nvidia-hotplug.sh"
```

`/usr/bin/nvidia-hotplug.sh`
```bash
#!/bin/bash

PATH="/usr/bin:/bin"
LOG="/var/log/nvidia-hotplug.log"

num_connected=0
for st in /sys/class/drm/card*-*/status; do
    [[ $(cat "$st") == "connected" ]] || continue
    num_connected=$((num_connected+1))
done

if [ "$num_connected" -eq 1 ]; then
    EXT_CONNECTED=false
else
    EXT_CONNECTED=true
fi

if $EXT_CONNECTED; then
    GCLOCK=$(nvidia-smi --query-supported-clocks=gr  --format=csv,noheader \
             | awk 'NR==1 {print $1}')
    MCLOCK=$(nvidia-smi --query-supported-clocks=mem --format=csv,noheader \
             | awk 'NR==1 {print $1}')
    nvidia-smi -pm 1
    nvidia-smi -lgc "$GCLOCK","$GCLOCK"
    nvidia-smi -lmc "$MCLOCK","$MCLOCK"
    echo "$(date '+%F %T') $num_connected displays → lock ${GCLOCK}/${MCLOCK}" >> "$LOG"
else
    nvidia-smi --reset-gpu-clocks
    nvidia-smi --reset-memory-clocks
    echo "$(date '+%F %T') $num_connected display  → clocks reset" >> "$LOG"
fi
```

I also noticed that GPU clocks are not related to the issue (only memory ones are), so if I omit `nvidia-smi -lgc "$GCLOCK","$GCLOCK"`, the windows are still snappy.

Disabling the iGPU completely in BIOS settings is also not necessary. In case of KWin WM, specifying the GPU priority seems to be enough:
`~/.config/plasma-workspace/env/kwin.sh`
```
export KWIN_DRM_DEVICES=/dev/dri/card1:/dev/dri/card2
```

I am running Arch on 4080 GPU and i9-13900HX CPU.

RileyMathews, 2025-06-12:
Thanks for that script @stdereka! Unfortunately system76 hasn't gotten around to exposing the switch to completely disable the iGPU in their BIOS even though their top end hardware supports it so my only hope is trying the p-state trick. I'll try out that script next time I'm using an external monitor. Came here after doing some digging on similar issues to say I really hope this gets fixed at the nvidia level so that the render offload performance improves because for a some people like me. Just disabling the iGPU isn't an option so render offload is the only way we can use both internal and external monitors.

NGStaph, 2025-06-26:
right. I caved and moved to the mesa nvk driver. I have no idea how much worse power management and other features are but the experience of using my laptop docked is infinitely more tolerable now.

the only downside is that my monitor is not recognized as an audio output.

S0und, 2025-07-17:
I've tried @stdereka suggestion, the constantly boosting both clocks, which somewhat helps. 

Also tried to set `KWIN_DRM_DEVICES`, which helped the most, but it's buggy. Right after a fresh boot, everything (including animations) are super smooth, but as time goes on, even after a few hours, the driver starts to glitch. 

I've set my laptop's 3060 as primary and left the 5900HX iGPU as secondary. 

Exp.:

* KDE's Overview takes seconds to open, usually, some random window clipping the animation

* Minimized window gets activated, and the KWIN animation that scales the window from minimized to full screen glitches. The full-screen application shows a stretched view (snapshot of the window in mid animation), and you can't interact with the window's content. 

* A windowed application cannot be set to full screen. It got stuck in windowed mode. 

But beyond this, I still experience OP's problem, I'm spamming (activating/deactivating) KDE's Overview mode:
4K 60hz monitor
```
274 frames in 5.0 seconds = 54.627 FPS 
284 frames in 5.0 seconds = 56.798 FPS 
163 frames in 5.0 seconds = 32.482 FPS - start
110 frames in 5.0 seconds = 21.909 FPS
137 frames in 5.0 seconds = 27.306 FPS
134 frames in 5.1 seconds = 26.084 FPS
141 frames in 5.0 seconds = 28.135 FPS
149 frames in 5.1 seconds = 29.503 FPS
141 frames in 5.0 seconds = 28.107 FPS - stop
233 frames in 5.0 seconds = 46.468 FPS
279 frames in 5.0 seconds = 55.797 FPS
261 frames in 5.0 seconds = 52.183 FPS
```
4K 120hz monitor
```
522 frames in 5.0 seconds = 104.285 FPS
544 frames in 5.0 seconds = 108.618 FPS
327 frames in 5.0 seconds = 65.159 FPS - start
196 frames in 5.0 seconds = 39.035 FPS
195 frames in 5.0 seconds = 38.839 FPS
191 frames in 5.0 seconds = 38.108 FPS
257 frames in 5.0 seconds = 51.372 FPS - stop
536 frames in 5.0 seconds = 107.191 FPS
```
That 50% GPU usage happened while I was doing the 4k@120 test. Both clocks are set to max. 

[image]

Nvidia Driver version: 575.64.03
Fedora 42 
ThinkBook 16p Gen 2 (16” AMD) 
Laptop is closed, and I'm only using two external 4k@60 and 4k@120 external monitors.

Kimiblock, 2025-08-19:
The 580 series driver seems to smooth things out a bit. Though it introduces some other bugs...

mtijanic, 2025-09-30:
Hi, sorry, I haven't been keeping up with what is happening here so best I can offer you is a digest of the audit log I just read, which is:

- The issues we were originally able to reproduce were fixed in some version of 575 (and 580)
- This was done by changing how display interrupts get delivered and serviced
- Some folks are reporting it is not fixed. I don't know if the last few messages above were seen specifically, so I'll forward them as well
  - Also reported at https://forums.developer.nvidia.com/t/nvidia-please-get-it-together-with-external-monitors-on-wayland/301684/74
- Second internal bug, 5547384 is tracking this issue
- We do not have internal repro yet, but someone is actively working on it.

gldio, 2025-10-01:
Things have improved but the issue is still present on 580.82.09. It’s been over a year now. If it’s still not fixed in early 2026, I’ll switch my laptop to one with an AMD GPU. It seems Nvidia doesn’t care about its customers.

RileyMathews, 2025-10-01:
@tm4ig when you say "without GSP Firmware" is that just saying that the proprietary driver does not have GSP firmware by default or do you have to do something extra to disable the GSP Firmware? Forgive my small ignorance in this area 😬. I will try the proprietary driver later but the last time I tried them a few weeks ago it had a similar problem for me 🙃.

SoloSaravanan, 2025-10-01:
Tried proprietary driver with gsp disabled issue still present.

[image]

RileyMathews, 2025-10-01:
> For me on the nvidia-open drivers at default settings on arch, forcing the bios setting to use the discrete gpu solved it perfectly on an asus laptop. I get stuttering audio on youtube but that may be from the recent news on asus' buggy bios for their recent laptops which is due for an update soon.

Unfortunately many of us don't have that ability. There are lots of modern laptops out there that ship dual GPUs without the ability to mux switch the main GPU on. Apparently my System76 laptop has a mux switch that would allow the nvidia GPU to drive the internal display as well but system76 never got around to updating their firmware to control it 🙃

SoloSaravanan, 2025-10-01:
> [@SoloSaravanan](https://github.com/SoloSaravanan)
> 
> How can you disable GSP firmware? I’m on 580.82.09, but the kernel argument isn’t working with it. On previous drivers, I could disable it, but not anymore. Also, I don’t have a MUX switch. I have an RTX 3060 laptop GPU on my Dell G15 5511, so I can’t force the dGPU. This issue is still being discussed on the NVIDIA forums, btw.

@gldio you have to install the proprietary driver and add "options nvidia NVreg_EnableGpuFirmware=0" to /etc/modprobe.d/nvidia.conf

mtijanic, 2025-10-01:
Hey folks, this is all great info and I'm very glad we're seeing it, but I think I'm gonna have to put my issue triage hat on for a second. Some of these posts seem to be a completely separate issue, so could we please keep this one relevant to the original issue which has a very particular signature of ***<ins>FPS being exactly half of the refresh rate</ins>*** (+/- 5% let's say). If you are still seeing this, it would be very helpful to post:
- Short description of the setup
- nvidia-bug-report.log, or at least ouptut of `nvidia-smi -q`
- Whether the issue goes away if the GPU runs at P0 or P2 (see https://github.com/NVIDIA/open-gpu-kernel-modules/issues/650#issuecomment-2500165851)

If the issue you are having is poor performance on external monitors in general (which seems to be the case with @SoloSaravanan above, IIUC), then maybe it might be better to open a new bug since the root cause is likely different.

aveyer3, 2025-10-01:
I'm having the same problem for ages on my Clevo PC70HR when using external monitor on Wayland.
It has Intel integrated graphics + Nvidia RTX 3070 laptop GPU.
Running Tuxedo OS, Nvidia mode.

Perhaps it is more of an issue with Wayland itself that needs fixing rather than the Nvidia driver, coz it seems like the Intel graphics are rendering the desktop etc, which is the cause of the poor performance.
X11 is fine, amazing performance.

[Nvidia-smi.txt](https://github.com/user-attachments/files/22649509/Nvidia-smi.txt)

[image]

Kimiblock, 2025-10-31:
I was convinced by that... Until I fired up apps like Firefox and stuttering is all over the place.

Still, way worse on GNOME tho.

[13 more replies]