NVIDIA/open-gpu-kernel-modules #651: Nvidia gpu not used

pr9543, 2024-06-04:
### NVIDIA Open GPU Kernel Modules Version

nvidia-driver-535-open

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [X] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Linux Mint 21.3

### Kernel Release

Linux debian 5.15.0-91-generic  x86_64 x86_64 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

GeForce RTX 2070 Mobile / Max-Q Refresh

### Describe the bug

I have a laptop with dgpu and igpu and only the igpu is being used in applications with nvidia-driver-535-open installed. I have set active profile to nvidia(performance mode) in the applet but applications still use my igpu instead of my nvidia dgpu.

### To Reproduce

install nvidia-driver-535-open and open any 3d application

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/15541895/nvidia-bug-report.log.gz)

robinhad, 2024-06-09:
Same issue on desktop 4090 on X11 as well on nvidia-driver-550-open

Binary-Eater, 2025-11-01:
@pr9543 according to your bug report, it looks like the driver was not installed correctly and you are running the display only with the X modesetting driver, so there is no provider to offload to for PRIME render offload.

```
xrandr --listproviders:

Providers: number : 1
Provider 0: id: 0x43 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
```