NVIDIA/open-gpu-kernel-modules #653: Please Update Changelog for huge  550.90.07 code dump

Qubitium, 2024-06-04:
### NVIDIA Open GPU Kernel Modules Version

 550.90.07

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

na

### Kernel Release

na

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

na

### Describe the bug

na

### To Reproduce

na

### Bug Incidence

Once

### nvidia-bug-report.log.gz

na

### More Info

@niv  Just saw a huge code dump for  `550.90.07`  but changelog was not updated with human readable counter part to the major changes/bug fixes. Thank you.

https://github.com/NVIDIA/open-gpu-kernel-modules/commit/e45d91de0224e17a6bf7c971d8f165090b3c07a6

niv, 2024-06-04:
Hi!

Thanks for checking in. We don't currently have a good process to maintain this in-tree changelog unfortunately. The driver release on nvidia.com will go up soon, along with the official release notes.

mtijanic, 2024-06-04:
To elaborate a bit, we actually have two sets of changelogs (for _reasons_) that mostly list different things. Most driver changes that get a changelog entry (and not many do) will get listed on the driver release page. For 550.90.07 that's https://www.nvidia.com/Download/driverResults.aspx/226768/en-us/ and it says:

> - Temporarily disabled the GLX_EXT_buffer_age extension on Xwayland to work around a bug that could cause corruption.
> - Fixed a bug that could cause corruption when the GLX_EXT_buffer_age extension is used on X.org with PRIME render offloading.
> - Improved the ability of nvidia-modprobe to detect whether kernel modules are already loaded. This corrects an issue that prevented nvidia-persistenced from setting persistence mode on some systems.
> - Fixed a bug that could cause additional X screens to claim displays that should have been assigned to a prior X screen based on the given X configuration.
> - Fixed a bug that could cause the X server to crash when graphics applications requested single-buffered drawables while certain features (such as Vulkan sharpening) are enabled.
> - Fixed a bug that could lead to a kernel panic, due to a failure to release a spinlock under some conditions.
> - Fixed a race condition which could lead to crashes when Xid errors occur concurrently on multiple GPUs.

These include entries that relate to any part of the driver stack. Then, we have https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/CHANGELOG.md which is mostly for entries referring to Issues or PRs made on this repo, or things that are otherwise tightly coupled with the kernel module source code.

I don't imagine this duplication will survive much longer once these open modules become the default driver mode, but until then the best way to get the idea of what changed (short of reading the diff) is to read both changelogs.