NVIDIA/open-gpu-kernel-modules #659: Periodic stutters and “NVRM: RmCheckForGcxSupportOnCurrentState” kernel warnings on Ubuntu 22.04 RTX 4070

edmcman, 2024-06-06:
### NVIDIA Open GPU Kernel Modules Version

550.54.15

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [X] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Ubuntu 22.04.4 LTS

### Kernel Release

6.5.0-28-generic

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 4070 Laptop GPU (UUID: GPU-02187fd8-22a1-3f71-cd52-22af54f42481)

### Describe the bug

I’ve been running into occasional visible “stutters” on my Ubuntu Linux 22.04 system. By stutter, I mean that for ~500ms there is no visible change to the screen. If there is a video playing, it freezes. If I am moving the mouse, the cursor will freeze.

At the same time, I get a ton of kernel messages such as:

```
Apr 24 14:59:16 banana kernel: NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0xffff
Apr 24 14:59:21 banana kernel: NVRM: _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 3d63e51c037800 >= 3d63e51c037800
Apr 24 14:59:21 banana kernel: NVRM: _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
Apr 24 14:59:21 banana kernel: NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0xffff
Apr 24 14:59:27 banana kernel: NVRM: _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 3d63e663d6cf00 >= 3d63e663d6cf00
Apr 24 14:59:27 banana kernel: NVRM: _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
Apr 24 14:59:27 banana kernel: NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0xffff
Apr 24 14:59:32 banana kernel: NVRM: _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 3d63e7abaa2600 >= 3d63e7abaa2600
Apr 24 14:59:32 banana kernel: NVRM: _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
Apr 24 14:59:32 banana kernel: NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0xffff
Apr 24 14:59:38 banana kernel: NVRM: _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 3d63e8f37d7d00 >= 3d63e8f37d7d00
```

Another concerning log entry is:
```
Apr 24 13:23:29 banana kernel: NVRM: _kgspLogXid119: ********************************* GSP Timeout **********************************
Apr 24 13:23:29 banana kernel: NVRM: _kgspLogXid119: Note: Please also check logs above.
Apr 24 13:23:29 banana kernel: NVRM: nvAssertFailedNoLog: Assertion failed: expectedFunc == pHistoryEntry->function @ kernel_gsp.c:1744
Apr 24 13:23:29 banana kernel: NVRM: GPU at PCI:0000:01:00: GPU-02187fd8-22a1-3f71-cd52-22af54f42481
Apr 24 13:23:29 banana kernel: NVRM: Xid (PCI:0000:01:00): 119, pid=1671935, name=kworker/1:3, Timeout after 1149s of waiting for RPC response from GPU0 GSP! Expected function 4097 (GSP_INIT_DONE) (0x0 0x0).
Apr 24 13:23:29 banana kernel: NVRM: GPU0 GSP RPC buffer contains function 4108 (UCODE_LIBOS_PRINT) and data 0x0000000000000000 0x0000000000000000.
Apr 24 13:23:29 banana kernel: NVRM: GPU0 RPC history (CPU -> GSP):
Apr 24 13:23:29 banana kernel: NVRM:     entry function                   data0              data1              ts_start           ts_end             duration actively_polling
Apr 24 13:23:29 banana kernel: NVRM:      0    47   UNLOADING_GUEST_DRIVE 0x0000000000000000 0x0000000000000000 0x000616daa95805e8 0x000616daa95d9f17    366 s y
Apr 24 13:23:29 banana kernel: NVRM:     -1    10   FREE                  0x00000000caf010bb 0x0000000000000000 0x000616daa958044a 0x000616daa95805e5    411us  
Apr 24 13:23:29 banana kernel: NVRM:     -2    76   GSP_RM_CONTROL        0x0000000020800ac3 0x0000000000000028 0x000616daa9580260 0x000616daa9580447    487us  
Apr 24 13:23:29 banana kernel: NVRM:     -3    4    ALLOC_MEMORY          0x0000000000000000 0x0000000000000000 0x000616daa957ff81 0x000616daa958025d    732us  
Apr 24 13:23:29 banana kernel: NVRM:     -4    10   FREE                  0x00000000caf010ba 0x0000000000000000 0x000616daa957fd61 0x000616daa957ff79    536us  
Apr 24 13:23:29 banana kernel: NVRM:     -5    76   GSP_RM_CONTROL        0x0000000020800ac3 0x0000000000000028 0x000616daa957fb78 0x000616daa957fd5f    487us  
Apr 24 13:23:29 banana kernel: NVRM:     -6    4    ALLOC_MEMORY          0x0000000000000000 0x0000000000000000 0x000616daa957f982 0x000616daa957fb75    499us  
Apr 24 13:23:29 banana kernel: NVRM:     -7    10   FREE                  0x00000000caf010b9 0x0000000000000000 0x000616daa957f7d9 0x000616daa957f97b    418us  
Apr 24 13:23:29 banana kernel: NVRM: GPU0 RPC event history (CPU <- GSP):
Apr 24 13:23:29 banana kernel: NVRM:     entry function                   data0              data1              ts_start           ts_end             duration during_incomplete_rpc
Apr 24 13:23:29 banana kernel: NVRM:      0    4108 UCODE_LIBOS_PRINT     0x0000000000000000 0x0000000000000000 0x000616daed814621 0x000616daed814622      1us  
Apr 24 13:23:29 banana kernel: NVRM:     -1    4108 UCODE_LIBOS_PRINT     0x0000000000000000 0x0000000000000000 0x000616daed8144ef 0x000616daed8144f0      1us  
Apr 24 13:23:29 banana kernel: NVRM:     -2    4128 GSP_POST_NOCAT_RECORD 0x0000000000000002 0x0000000000000027 0x000616daed812e49 0x000616daed812e4b      2us  
Apr 24 13:23:29 banana kernel: NVRM:     -3    4098 GSP_RUN_CPU_SEQUENCER 0x0000000000000628 0x0000000000003fe2 0x000616daed808c11 0x000616daed809d6e   4445us  
Apr 24 13:23:29 banana kernel: NVRM:     -4    4108 UCODE_LIBOS_PRINT     0x0000000000000000 0x0000000000000000 0x000616daa958d7c0 0x000616daa958d7c1      1us  
Apr 24 13:23:29 banana kernel: NVRM:     -5    4128 GSP_POST_NOCAT_RECORD 0x0000000000000002 0x0000000000000028 0x000616daa9585c1e 0x000616daa9585c20      2us  
Apr 24 13:23:29 banana kernel: NVRM:     -6    4111 PERF_BRIDGELESS_INFO_ 0x0000000000000000 0x0000000000000000 0x000616daa9585a33 0x000616daa9585a33           
Apr 24 13:23:29 banana kernel: NVRM:     -7    4128 GSP_POST_NOCAT_RECORD 0x0000000000000002 0x0000000000000001 0x000616daa853253d 0x000616daa8532544      7us  
Apr 24 13:23:29 banana kernel: NVRM: _kgspLogXid119: ********************************************************************************
Apr 24 13:23:29 banana kernel: NVRM: nvCheckOkFailedNoLog: Check failed: Call timed out [NV_ERR_TIMEOUT] (0x00000065) returned from rpcRecvPoll(pGpu, pRpc, NV_VGPU_MSG_EVENT_GSP_INIT_DONE) @ kernel_gsp.c:4074
Apr 24 13:23:29 banana kernel: NVRM: gpuPowerManagementResume: State load at resume for riscv/gsp failed: 0x65
Apr 24 13:23:35 banana kernel: NVRM: Xid (PCI:0000:01:00): 119, pid=1671935, name=kworker/1:3, Timeout after 6s of waiting for RPC response from GPU0 GSP! Expected function 76 (GSP_RM_CONTROL) (0x2080205b 0x4).
Apr 24 13:23:35 banana kernel: NVRM: _issueRpcAndWait: rpcRecvPoll timedout for fn 76!
Apr 24 13:23:35 banana kernel: NVRM: subdeviceCtrlCmdPerfSetPowerstate_KERNEL: NV2080_CTRL_CMD_PERF_SET_POWERSTATE RPC failed
Apr 24 13:23:46 banana kernel: NVRM: Xid (PCI:0000:01:00): 119, pid=854, name=nv_queue, Timeout after 6s of waiting for RPC response from GPU0 GSP! Expected function 76 (GSP_RM_CONTROL) (0x2080a7d7 0x2).
Apr 24 13:23:46 banana kernel: NVRM: _issueRpcAndWait: rpcRecvPoll timedout for fn 76!
Apr 24 13:23:46 banana kernel: NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x65
Apr 24 13:23:57 banana kernel: NVRM: Rate limiting GSP RPC error prints for GPU at PCI:0000:01:00 (printing 1 of every 30).  The GPU likely needs to be reset.
```

### To Reproduce

Unknown.  I just use machine for a while and it happens periodically.

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

I have emailed this to linux-bugs@nvidia.com on 4/24/2024

davelima, 2024-10-30:
I have the exact same problem. I have also noticed it tends to happend whenever I open a Chromium/Electron app, also it only happens under Wayland. Used X11 for a few weeks and it worked normally.

I'm using Fedora and a RTX 3050, Kernel 6.11.5-300.fc41.x86_64

snake-scaly, 2024-11-30:
I have an issue with very similar symptoms but which manifests much more severely. Whenever I install 560.35 or 565.57, my machine is guaranteed to hang. Can be before the login screen has a chance to render, or on the login screen, or on the desktop after I login. In rare cases when it does not hang, not all software is able to work. E.g. once Discord worked fine but Chrome couldn't render anything.

The latest stable driver for me is 545.29 which I'm using right now. This is probably related to the hardware I'm using (a Jiaolong laptop, RTX 3060 Mobile on Ryzen 7). This forces me to downgrade kernel to 6.7.5 because that's how Arch Nvidia drivers were built.

Please let me know if I can provide more information or run experiments to help you identify the issue.

<details>
<summary>Here's an example of what I see in the logs with 565.57 and kernel 6.12.1</summary>

```
Nov 30 17:11:16 archlinux kernel: NVRM: testIfDsmSubFunctionEnabled: GPS ACPI DSM called before _acpiDsmSupportedFuncCacheInit subfunction = 11.
[...]
Nov 30 17:11:17 archlinux kernel: NVRM: testIfDsmSubFunctionEnabled: GPS ACPI DSM called before _acpiDsmSupportedFuncCacheInit subfunction = 11.
[...]
Nov 30 17:11:38 mai kernel: NVRM: gpuWaitForGfwBootComplete_TU102: failed to wait for GFW_BOOT: (progress 0x9)
Nov 30 17:11:38 mai kernel: NVRM: kgspWaitForGfwBootOk_TU102: failed to wait for GFW boot complete: 0x55 VBIOS version 94.06.17.00.41
Nov 30 17:11:38 mai kernel: NVRM: kgspWaitForGfwBootOk_TU102: (the GPU may be in a bad state and may need to be reset)
Nov 30 17:11:38 mai kernel: NVRM: nvAssertFailedNoLog: Assertion failed: rmapiLockIsOwner() @ mem_mgr.c:465
[...]
Nov 30 17:11:44 mai kernel: NVRM: _kgspLogXid119: ********************************* GSP Timeout **********************************
Nov 30 17:11:44 mai kernel: NVRM: _kgspLogXid119: Note: Please also check logs above.
Nov 30 17:11:44 mai kernel: NVRM: GPU at PCI:0000:01:00: GPU-0b811689-1031-4db2-64b4-778d1b56fdd9
Nov 30 17:11:44 mai kernel: NVRM: Xid (PCI:0000:01:00): 119, pid=789, name=(udev-worker), Timeout after 6s of waiting for RPC response from GPU0 GSP! Expected function 76 (GSP_RM_CONTROL) (0x2080205b 0x4).
Nov 30 17:11:44 mai kernel: NVRM: GPU0 GSP RPC buffer contains function 76 (GSP_RM_CONTROL) and data 0x000000002080205b 0x0000000000000004.
Nov 30 17:11:44 mai kernel: NVRM: GPU0 RPC history (CPU -> GSP):
Nov 30 17:11:44 mai kernel: NVRM:     entry function                   data0              data1              ts_start           ts_end             duration actively_polling
Nov 30 17:11:44 mai kernel: NVRM:      0    76   GSP_RM_CONTROL        0x000000002080205b 0x0000000000000004 0x00062821e4dcd773 0x0000000000000000          y
Nov 30 17:11:44 mai kernel: NVRM:     -1    47   UNLOADING_GUEST_DRIVE 0x0000000000000000 0x0000000000000000 0x00062821e4caa699 0x00062821e4cf614e 309941us  
Nov 30 17:11:44 mai kernel: NVRM:     -2    10   FREE                  0x00000000c1e0000d 0x0000000000000000 0x00062821e4caa34d 0x00062821e4caa5e6    665us  
Nov 30 17:11:44 mai kernel: NVRM:     -3    10   FREE                  0x000000000000000a 0x0000000000000000 0x00062821e4ca9fb5 0x00062821e4caa34a    917us  
Nov 30 17:11:44 mai kernel: NVRM:     -4    10   FREE                  0x000000000000000b 0x0000000000000000 0x00062821e4ca9ce6 0x00062821e4ca9e90    426us  
Nov 30 17:11:44 mai kernel: NVRM:     -5    10   FREE                  0x0000000000000006 0x0000000000000000 0x00062821e4ca9895 0x00062821e4ca9cc8   1075us  
Nov 30 17:11:44 mai kernel: NVRM:     -6    10   FREE                  0x0000000000000002 0x0000000000000000 0x00062821e4ca89bd 0x00062821e4ca9769   3500us  
Nov 30 17:11:44 mai kernel: NVRM:     -7    10   FREE                  0x0000000000000005 0x0000000000000000 0x00062821e4ca818a 0x00062821e4ca89ad   2083us  
Nov 30 17:11:44 mai kernel: NVRM: GPU0 RPC event history (CPU <- GSP):
Nov 30 17:11:44 mai kernel: NVRM:     entry function                   data0              data1              ts_start           ts_end             duration during_incomplete_rpc
Nov 30 17:11:44 mai kernel: NVRM:      0    4108 UCODE_LIBOS_PRINT     0x0000000000000000 0x0000000000000000 0x00062821e4cb3d55 0x00062821e4cb3d57      2us  
Nov 30 17:11:44 mai kernel: NVRM:     -1    4128 GSP_POST_NOCAT_RECORD 0x0000000000000002 0x0000000000000028 0x00062821e4cafc3c 0x00062821e4cafc44      8us  
Nov 30 17:11:44 mai kernel: NVRM:     -2    4111 PERF_BRIDGELESS_INFO_ 0x0000000000000000 0x0000000000000000 0x00062821e4cafa77 0x00062821e4cafa79      2us  
Nov 30 17:11:44 mai kernel: NVRM:     -3    4108 UCODE_LIBOS_PRINT     0x0000000000000000 0x0000000000000000 0x00062821e38d545b 0x00062821e38d545c      1us  
Nov 30 17:11:44 mai kernel: NVRM:     -4    4108 UCODE_LIBOS_PRINT     0x0000000000000000 0x0000000000000000 0x00062821e38d5336 0x00062821e38d5338      2us  
Nov 30 17:11:44 mai kernel: NVRM:     -5    4128 GSP_POST_NOCAT_RECORD 0x0000000000000002 0x0000000000000027 0x00062821e38d2dfc 0x00062821e38d2e01      5us  
Nov 30 17:11:44 mai kernel: NVRM:     -6    4098 GSP_RUN_CPU_SEQUENCER 0x000000000000061c 0x0000000000003fe2 0x00062821e38c944b 0x00062821e38ca61f   4564us  
Nov 30 17:11:44 mai kernel: CPU: 15 UID: 0 PID: 789 Comm: (udev-worker) Tainted: G           OE      6.12.1-arch1-1 #1 33f4a68ee85c59cb5d6edb747af0349869779b24
Nov 30 17:11:44 mai kernel: Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[... cut ...]
Nov 30 17:14:46 mai kernel: NVRM: nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_client.c:843
Nov 30 17:14:46 mai kernel: NVRM: nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:258
[...]
Nov 30 17:14:52 mai kernel: NVRM: Rate limiting GSP RPC error prints for GPU at PCI:0000:01:00 (printing 1 of every 30).  The GPU likely needs to be reset.
Nov 30 17:14:52 mai kernel: NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x65
[... system hangs, no more logs ...]
```

</details>

CapoFantasma97, 2025-01-06:
Same on Fedora 41 using an NVidia RTX 3070 mobile, using Wayland. Haven't verified if the same behavior is shown under X11.

When this happens, the graphs on the system monitor and nvtop stop working entirely, nvidia-smi shows most entries as "ERR!" and some apps don't launch at all (e.g. Steam). OBS can't detect the GPU and reverts to software codec. Plus the constant stutters that freeze everything.

This happens at least on a daily occurrence, and may or may not be triggered more often by launching certain software as well (e.g. [Clementine Player](https://github.com/clementine-player/Clementine)).

Here are some of the lines that seem relevant in `journalctl`:
```
gen 06 15:50:04 fedora systemd[2540]: plasma-ksystemstats.service: Failed with result 'timeout'.
gen 06 15:50:04 fedora systemd[2540]: Failed to start plasma-ksystemstats.service - Track hardware statistics.
gen 06 15:50:07 fedora kernel: NVRM: _kgspRpcSendMessage: GspMsgQueueSendCommand failed on GPU0: 0x3
gen 06 15:50:07 fedora kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000003 for fn 76!
gen 06 15:50:07 fedora kernel: NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0xffff
gen 06 15:50:40 fedora kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: Failure: Generic Error [NV_ERR_GENERIC] (0x0000FFFF) returned from pRmApi->Control(pRmApi, pGpu->hInternalClient, pGpu->hInternalSu>
gen 06 15:50:41 fedora kernel: NVRM: _kgspRpcSendMessage: GspMsgQueueSendCommand failed on GPU0: 0x3
gen 06 15:50:41 fedora kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000003 for fn 76!
gen 06 15:50:53 fedora kernel: NVRM: nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:1867
```
I made sure everything is up to date and I tried rebuilding drivers with `akmods --force --rebuild`.

vanvugt, 2025-06-20:
Also tracking in Ubuntu: https://bugs.launchpad.net/bugs/2107653

cyril23, 2025-07-26:
Not sure if we have the same problem but here is my **GSP Timeout** log, on Proxmox 8.4.1:
```
2025-07-23T20:06:52.621437+02:00 ThreadripperIntern kernel: [612856.523956] NVRM: _kgspLogXid119: ********************************* GSP Timeout **********************************
2025-07-23T20:06:52.621455+02:00 ThreadripperIntern kernel: [612856.526094] NVRM: _kgspLogXid119: Note: Please also check logs above.
2025-07-23T20:06:52.625742+02:00 ThreadripperIntern kernel: [612856.528273] NVRM: GPU at PCI:0000:0a:00: GPU-701b524f-c979-7284-9737-0fecd8ca5f9b
2025-07-23T20:06:52.625751+02:00 ThreadripperIntern kernel: [612856.530413] NVRM: GPU Board Serial Number: 1791625003310
2025-07-23T20:06:52.628827+02:00 ThreadripperIntern kernel: [612856.531941] NVRM: Xid (PCI:0000:0a:00): 119, pid=1227848, name=llama-server, Timeout after 45s of waiting for RPC response from GPU0 GSP! Expected function 76 (GSP_RM_CONTROL) (0x20800a9a 0x8).
2025-07-23T20:06:52.630401+02:00 ThreadripperIntern kernel: [612856.533524] NVRM: GPU0 GSP RPC buffer contains function 76 (GSP_RM_CONTROL) and data 0x0000000020800a9a 0x0000000000000008.
2025-07-23T20:06:52.630405+02:00 ThreadripperIntern kernel: [612856.535071] NVRM: GPU0 RPC history (CPU -> GSP):
2025-07-23T20:06:52.633413+02:00 ThreadripperIntern kernel: [612856.536589] NVRM:     entry function                   data0              data1              ts_start           ts_end             duration actively_polling
2025-07-23T20:06:52.633420+02:00 ThreadripperIntern kernel: [612856.538086] NVRM:      0    76   GSP_RM_CONTROL        0x0000000020800a9a 0x0000000000000008 0x00063a9c902072c7 0x0000000000000000          y
2025-07-23T20:06:52.636405+02:00 ThreadripperIntern kernel: [612856.539595] NVRM:     -1    76   GSP_RM_CONTROL        0x0000000020800a9a 0x0000000000000008 0x00063a9c8f42c04c 0x00063a9c8f42c166    282us  
2025-07-23T20:06:52.636411+02:00 ThreadripperIntern kernel: [612856.541078] NVRM:     -2    76   GSP_RM_CONTROL        0x0000000020800a9a 0x0000000000000008 0x00063a9c8e5615de 0x00063a9c8e5634db   7933us  
2025-07-23T20:06:52.639324+02:00 ThreadripperIntern kernel: [612856.542562] NVRM:     -3    76   GSP_RM_CONTROL        0x0000000020800a9a 0x0000000000000008 0x00063a9c8d78404a 0x00063a9c8d78416c    290us  
2025-07-23T20:06:52.639331+02:00 ThreadripperIntern kernel: [612856.543998] NVRM:     -4    76   GSP_RM_CONTROL        0x0000000020800a9a 0x0000000000000008 0x00063a9c8c8b96de 0x00063a9c8c8bb538   7770us  
2025-07-23T20:06:52.642166+02:00 ThreadripperIntern kernel: [612856.545441] NVRM:     -5    76   GSP_RM_CONTROL        0x0000000020800a9a 0x0000000000000008 0x00063a9c8bade16c 0x00063a9c8bade286    282us  
2025-07-23T20:06:52.642171+02:00 ThreadripperIntern kernel: [612856.546840] NVRM:     -6    76   GSP_RM_CONTROL        0x0000000020800a9a 0x0000000000000008 0x00063a9c8ac1361f 0x00063a9c8ac15645   8230us  
2025-07-23T20:06:52.644903+02:00 ThreadripperIntern kernel: [612856.548221] NVRM:     -7    76   GSP_RM_CONTROL        0x0000000020800a9a 0x0000000000000008 0x00063a9c89e3859b 0x00063a9c89e386bd    290us  
2025-07-23T20:06:52.644908+02:00 ThreadripperIntern kernel: [612856.549574] NVRM: GPU0 RPC event history (CPU <- GSP):
2025-07-23T20:06:52.647634+02:00 ThreadripperIntern kernel: [612856.550962] NVRM:     entry function                   data0              data1              ts_start           ts_end             duration during_incomplete_rpc
2025-07-23T20:06:52.647642+02:00 ThreadripperIntern kernel: [612856.552307] NVRM:      0    4128 GSP_POST_NOCAT_RECORD 0x0000000000000005 0x000000000155d57a 0x00063a0de2cec85a 0x00063a0de2cec85a           
2025-07-23T20:06:52.650282+02:00 ThreadripperIntern kernel: [612856.553645] NVRM:     -1    4128 GSP_POST_NOCAT_RECORD 0x0000000000000005 0x000000000155d568 0x00063a0de2cec858 0x00063a0de2cec859      1us  
2025-07-23T20:06:52.650289+02:00 ThreadripperIntern kernel: [612856.554955] NVRM:     -2    4128 GSP_POST_NOCAT_RECORD 0x0000000000000005 0x000000000155d57a 0x00063a0de2cec856 0x00063a0de2cec857      1us  
2025-07-23T20:06:52.652861+02:00 ThreadripperIntern kernel: [612856.556259] NVRM:     -3    4128 GSP_POST_NOCAT_RECORD 0x0000000000000005 0x000000000155d568 0x00063a0de2cec850 0x00063a0de2cec855      5us  
2025-07-23T20:06:52.652868+02:00 ThreadripperIntern kernel: [612856.557534] NVRM:     -4    4124 GSP_LOCKDOWN_NOTICE   0x0000000000000000 0x0000000000000000 0x00063a0de200ad51 0x00063a0de200ad51           
2025-07-23T20:06:52.655376+02:00 ThreadripperIntern kernel: [612856.558808] NVRM:     -5    4124 GSP_LOCKDOWN_NOTICE   0x0000000000000001 0x0000000000000000 0x00063a0de200993d 0x00063a0de200993d           
2025-07-23T20:06:52.655383+02:00 ThreadripperIntern kernel: [612856.560049] NVRM:     -6    4124 GSP_LOCKDOWN_NOTICE   0x0000000000000000 0x0000000000000000 0x00063a0de200984b 0x00063a0de200984c      1us  
2025-07-23T20:06:52.657808+02:00 ThreadripperIntern kernel: [612856.561277] NVRM:     -7    4124 GSP_LOCKDOWN_NOTICE   0x0000000000000001 0x0000000000000000 0x00063a0de2009492 0x00063a0de2009493      1us  
2025-07-23T20:06:52.657814+02:00 ThreadripperIntern kernel: [612856.562481] CPU: 2 PID: 1227848 Comm: llama-server Tainted: P           OE      6.8.12-11-pve #1
2025-07-23T20:06:52.712534+02:00 ThreadripperIntern kernel: [612856.563673] Hardware name: Micro-Star International Co., Ltd. MS-7B09/X399 SLI PLUS (MS-7B09), BIOS A.89 08/11/2022
[... cut ...]
wget https://developer.download.nvidia.com/compute/cuda/12.9.1/local_installers/cuda-repo-debian12-12-9-local_12.9.1-575.57.08-1_amd64.deb
sudo dpkg -i cuda-repo-debian12-12-9-local_12.9.1-575.57.08-1_amd64.deb
sudo cp /var/cuda-repo-debian12-12-9-local/cuda-*-keyring.gpg /usr/share/keyrings/
sudo apt-get update
# I've installed the open driver:
sudo apt-get install -y nvidia-open
```

Now I'll try disabling GSP by adding `options nvidia NVreg_EnableGpuFirmware=0` into a file `/etc/modprobe.d/nvidia-gsp.conf` and then doing `update-initramfs -u -k all`, followed by a `reboot`. Hope this workaround helps.

francoism90, 2025-10-05:
```
...
[46869.864990] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46874.864837] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46879.864622] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46884.864405] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46889.864461] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46894.864352] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46899.864744] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46904.864585] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46909.864384] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46914.864245] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46919.864269] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46924.864082] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46929.863916] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46934.863743] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46939.863556] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46944.863354] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46949.863167] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46954.863005] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46959.862810] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46964.862610] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46969.862448] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46974.862271] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46979.862087] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46984.861911] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46989.861747] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46994.861542] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[46999.861578] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[47004.861383] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
[47009.861434] NVRM: RmCheckForGcxSupportOnCurrentState: NVRM, Failed to get GCx pre-requisite, status=0x62
```

The power package is installed as dep. Weird it's back.

francoism90, 2025-12-25:
After appending `nvidia-drm.modeset=1 nvidia-drm.fbdev=1` as kernel options, it now works fine. It seems both are disabled by default upstream.

thisisnotatrash, 2026-04-08:
> Not sure if we have the same problem but here is my **GSP Timeout** log, on Proxmox 8.4.1:

Same error on my side, on my laptop after opening Chrome. Chrome freeze for few seconds then back to normal