NVIDIA/open-gpu-kernel-modules #664: Failed to load module on boot

my4ng, 2024-06-17:
### NVIDIA Open GPU Kernel Modules Version

555.52.04

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [X] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Fedora Linux 40 (KDE Plasma)

### Kernel Release

Linux fedora 6.9.5-cb1.0.lto.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Jun 17 00:15:04 UTC 2024 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 4070 Ti

### Describe the bug

The open driver fails to load on the new linux kernel 6.9.5 consistently. This does not happen with either the 6.9.4 kernel or with the proprietary driver.

I have attached the filtered journalctl output: [error.log](https://github.com/user-attachments/files/15872083/error.log)

### To Reproduce

Pre-requisites:
- Fedora 40
- linux-cachyos LTO kernel 6.9.5 + akmods
```
dnf copr enable bieszczaders/kernel-cachyos-lto
dnf install kernel-cachyos-lto kernel-cachyos-lto-devel-matched akmods
```
- Negativo17's `fedora-nvidia` repo
```
dnf config-manager --add-repo=https://negativo17.org/repos/fedora-nvidia.repo
dnf install nvidia-driver nvidia-driver-libs
```
---

1. Set kernel module type to open `sed -i -e 's/kernel$/kernel-open/g' /etc/nvidia/kernel.conf`
2. `akmods --rebuild`
3. `reboot`

### Bug Incidence

Always

### nvidia-bug-report.log.gz

N/A

my4ng, 2024-07-31:
Can confirm still broken for 555.58.02 and 6.10.2-cb1.0.lto.fc40.x86_64

ptr1337, 2024-07-31:
This will not be fixed until https://github.com/dell/dkms/pull/417 is merged - but this is not a NVIDIA issue.

If you are using akmods, report to them the same as in the above PR

scaronni, 2024-10-23:
> This will not be fixed until [dell/dkms#417](https://github.com/dell/dkms/pull/417) is merged - but this is not a NVIDIA issue.
> 
> If you are using akmods, report to them the same as in the above PR

Merged, thanks.

my4ng, 2024-10-24:
> Merged, thanks.

I no longer owns an nvidia card, so unfortunately can't test if or when this is fixed. If someone can confirm it works, I will close the issue.