NVIDIA/open-gpu-kernel-modules #666: Failures to resume when sleeping (s0ix) on newer kernels

XutaxKamay, 2024-06-21:
### NVIDIA Open GPU Kernel Modules Version

550.90.07

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Gentoo

### Kernel Release

6.9.3-xutaxkamay-hardened (built it myself, but also breaks with official releases)

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce GTX 1650

### Describe the bug

I have a laptop with a dgpu (NVIDIA GeForce GTX 1650), I haven't tested on a lot of kernels yet, but it seems that I'm getting some failures to resume after sleeping.

I can't seem to get the card waking up after some time (Unable to change D3cold to D0). (see dmesg log)

I can SSH into my laptop though after a while by pressing the power button for some time and restart the display but the dgpu doesn't respond anymore.

It still work fine on older longterm kernels, like 6.1.x longterm, after longterm version, like 6.6.x it fails to resume.

### To Reproduce

It should be able to reproduce it with a laptop that has a 1650 GTX, a 11th Intel CPU and any kernel that is version >= 6.6.x.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/15934112/nvidia-bug-report.log.gz)
[dmesg.log](https://github.com/user-attachments/files/15934114/dmesg.log)

### More Info

Let me know if I can give more information.

Should I try the proprietary module ? (This will enforce me to disable Intel CET though which I don't really like the idea)

Not very sure what could affect it but it seems impossible to fix, I could try different kernel version between them and notice what kind of pull requests/patchs made this kernel module not working anymore.

closed 2024-06-28 by XutaxKamay

XutaxKamay, 2024-06-28:
> You could try following: sudo systemctl enable nvidia-resume nvidia-hibernate nvidia-suspend
> 
> Add following to /usr/lib/modprobe.d/nvidia-sleep.conf
> 
> ```
> options nvidia NVreg_PreserveVideoMemoryAllocations=1 NVreg_TemporaryFilePath=/var/tmp
> ```
> 
> After that regenerate your initramfs.
> 
> Also, I would suggest to check it without a hardned kernel
Just checked now,
Sadly same issue on both non-hardened and hardened kernel, I'm mostly thinking that it has to do with a kernel change, I'll likely need to test some kernel versions between 6.1 and 6.6 to find out what changed that could cause the issue.

XutaxKamay, 2024-06-28:
It is probably not NVIDIA fault

XutaxKamay, 2024-06-28:
Although right now I have a nice call trace, let me upload this.

ptr1337, 2024-06-28:
> Just checked now, Sadly same issue on both non-hardened and hardened kernel, I'm mostly thinking that it has to do with a kernel change, I'll likely need to test some kernel versions between 6.1 and [...]

Actually, I can not reproduce this on a 4070S on the 6.9.7 Kernel :/

XutaxKamay, 2024-06-28:
> Actually, I can not reproduce this on a 4070S on the 6.9.7 Kernel :/

I see, no worries though, I'll figure this out eventually, will just take some time, I'll come back to you once I have found the issue

XutaxKamay, 2024-06-30:
> Actually, I can not reproduce this on a 4070S on the 6.9.7 Kernel :/

I have found the commit that makes the problem appears: [34b62f186db9614e55d021f8c58d22fc44c57911](https://github.com/torvalds/linux/commit/34b62f186db9614e55d021f8c58d22fc44c57911#diff-2adf0b804ca9cdfe17639cad1ded3ed41119db7c66380a9de39eafbda34de61b)

I believe the issue is here:

   - Wait longer for devices to become ready after resume (as we do for
     reset) to accommodate Intel Titan Ridge xHCI devices

   - Extend D3hot delay for NVIDIA HDA controllers to avoid
     unrecoverable devices after a bus reset

This is probably some bugs that were introduced here.

I'll try to test few things by changing the code.

XutaxKamay, 2024-06-30:
Can be followed here if anyone is interested https://bugzilla.kernel.org/show_bug.cgi?id=219000