NVIDIA/open-gpu-kernel-modules #669: can't install cuda (530 | 535 | 555) in my system

maifeeulasad, 2024-06-30:
### NVIDIA Open GPU Kernel Modules Version

530

### Operating System and Version

Ubuntu 22.04.4 LTS

### Kernel Release

Linux maifee-desktop-ubuntu 6.5.0-41-generic #41~22.04.2-Ubuntu SMP PREEMPT_DYNAMIC Mon Jun  3 11:32:55 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Build Command

sudo apt install nvidia-driver-530 nvidia-dkms-530 

### Terminal output/Build Log

...
variable]
  103 |     nv_state_t *nv = NV_STATE_PTR(nvl);
      |                 ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-acpi.c:101:17: warning: unused variable ‘sp’ [-Wunused-variable]
  101 |     nv_stack_t *sp = NULL;
      |                 ^~
make[3]: *** [scripts/Makefile.build:251: /var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-pci.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [scripts/Makefile.build:251: /var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-acpi.o] Error 1
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c: In function ‘nvidia_mmap_numa’:
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:461:19: error: assignment of read-only member ‘vm_flags’
  461 |     vma->vm_flags |= VM_MIXEDMAP;
      |                   ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c: In function ‘nvidia_mmap_helper’:
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:606:23: error: assignment of read-only member ‘vm_flags’
  606 |         vma->vm_flags |= VM_IO | VM_PFNMAP | VM_DONTEXPAND;
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:673:23: error: assignment of read-only member ‘vm_flags’
  673 |         vma->vm_flags |= (VM_IO | VM_LOCKED | VM_RESERVED);
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:674:23: error: assignment of read-only member ‘vm_flags’
  674 |         vma->vm_flags |= (VM_DONTEXPAND | VM_DONTDUMP);
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:680:23: error: assignment of read-only member ‘vm_flags’
  680 |         vma->vm_flags &= ~VM_WRITE;
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:681:23: error: assignment of read-only member ‘vm_flags’
  681 |         vma->vm_flags &= ~VM_MAYWRITE;
...
As GitHub is telling `Comment is too long (maximum is 65536 characters)`, I'm sharing the rest as gist: https://gist.github.com/maifeeulasad/10d68d94fb83f16dfb4caf2e6e7cb982#file-cuda_530_20240630-txt

### More Info

I have tried installing in various way, but nothing works
 - cuda_12.1.1_530.30.02_linux.run  
 - cuda-repo-ubuntu2204-12-1-local_12.1.0-530.30.02-1_amd64.deb
 - cuda_12.2.0_535.54.03_linux.run
 - cuda_12.5.0_555.42.02_linux.run

gcc I'm using `gcc (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0`

ptr1337, 2024-06-30:
The issues for the open kernel modules is intended for the open kernel and not installation support. Check out the forum for help to install the nvidia driver
https://forums.developer.nvidia.com/c/gpu-graphics/linux/148

mtijanic, 2024-07-01:
This seems like a duplicate of #469, which is confirmed fixed as of 530.41.03 but your log says:
> Installing NVIDIA driver version 530.30.02.

You say you've tried with `cuda_12.5.0_555.42.02_linux.run` - can you post the log for that? That version should have the fix.

maifeeulasad, 2024-07-02:
@mtijanic tried with `12.1`, `12.4`, `12.5`. But nothing worked. I will share all the related logs files related to this last installation with `12.5`.