NVIDIA/open-gpu-kernel-modules #676: black background after certain time

Hyphaed, 2024-07-13:
### NVIDIA Open GPU Kernel Modules Version

555.52.04 

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Ubuntu 24.04 LTS

### Kernel Release

6.8.9

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 3070

### Describe the bug

after certain time background appears all black, everything keeps working

### To Reproduce

just work few hours under vmware, when I come back to gnome screen is all black

I'm using Gnome 46.2

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[Uploading nvidia-bug-report.log.gz…]()

emilnymann, 2024-07-18:
This happens on the proprietary driver as well, and is most likely not a direct issue with the driver but with the compositor. I can reproduce on Gnome with both the proprietary and open driver.

Hyphaed, 2024-07-18:
@emilnymann  you mean the problem is with mutter?

I had been all day without having this error, some days all the time the background is crashing

I do not understand why not today, I've done the same with the workstation as everyday