NVIDIA/open-gpu-kernel-modules #688: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read

taochenlove, 2024-08-02:
### NVIDIA Open GPU Kernel Modules Version

560.28.03

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Ubuntu 22.04 LTS

### Kernel Release

5.15.0-25-generic

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [X] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA A100-PCIE-40GB

### Describe the bug

When running nvidia-smi there are some exceptions printed below
(base) root@D11DJ-3410-01:~/chenct# nvidia-smi -L
[12919.827336] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88158,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827348] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88174,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827489] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x889d4,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827624] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e2c,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827628] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e30,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827632] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e34,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827636] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e38,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827639] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e3c,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827642] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e40,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827646] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e44,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827649] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e48,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827652] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e4c,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827656] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e50,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827659] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e54,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827662] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e58,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827665] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e5c,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827668] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e60,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827671] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e64,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827674] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e68,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827677] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e6c,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827680] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e70,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827682] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e74,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827685] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e78,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827689] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e7c,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827691] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e80,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827694] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e84,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827697] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e88,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827699] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e8c,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827703] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e90,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827705] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e94,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827708] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e98,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827711] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88e9c,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827714] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88ea0,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827717] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88ea4,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827720] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88ea8,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827722] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88eac,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827726] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88eb0,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827729] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88eb4,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827731] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88eb8,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827734] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88ebc,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827737] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88ec0,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[... cut ...]
[12919.827758] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88edc,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[12919.827761] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88fe4,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR

### To Reproduce

 "./cuda_12.6.0_560.28.03_linux.run -m=kernel-open" .Use this command after the installation will appear.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

none

gauravjuvekar, 2024-08-07:
Tracked internally as Bug 4290269

drastx, 2024-08-22:
I am seeing the same error code NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR on GH200, could these be related?
Ubuntu 24.04 and nvidia's ghvirt 6.5.3 based kernel, driver 550

[    5.868764] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920bc,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.870053] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920c0,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.871339] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920c4,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.872554] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920c8,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.873768] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920cc,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.875016] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920d0,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.876228] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920e4,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.877451] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920e8,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.878672] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920ec,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.879958] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920f0,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.881140] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920f4,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.882292] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920f8,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[    5.883533] NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920fc,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR

gauravjuvekar, 2024-08-22:
Yes, this is the same bug which affects release 550 and later.

drastx, 2024-08-23:
Thank you @gauravjuvekar for confirming, can we increase urgency for internal bug 4290269, we have about a thousand GH200 hosts with this issue. Is there a workaround, is there an ETA for the fix?

apoorvemohan, 2024-09-24:
We are seeing the following error on AMD system A100 40GB system with Nvidia Driver 550 and CUDA 12.4. 

```
[   37.333506] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88158,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
```

levipereira, 2024-10-16:
Same bug
Ubuntu 22.04 LTS
 
using NVIDIA-Linux-x86_64-560.35.03.run

GPU  RTX  4090

Architecture:             x86_64
  CPU op-mode(s):         32-bit, 64-bit
  Address sizes:          43 bits physical, 48 bits virtual
  Byte Order:             Little Endian
CPU(s):                   16
  On-line CPU(s) list:    0-15
Vendor ID:                AuthenticAMD
  Model name:             AMD Ryzen 7 3700X 8-Core Processor

jwatte, 2024-11-10:
Is this actually a problem, though? I see this in dmesg on bootup on many 8xH100 nodes (with quad xeon host) but it seems to work after this.

NVRM: gpuHandleSanityCheckRegReadError_GH100: Possible bad register read: addr: 0x920f0,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR

CMoench, 2024-12-03:
We are also observing similar error messages in _dmesg_ following the boot process on our 8xA100-80GB system.
Ubuntu 24.04, Kernel: 6.8.0-1018-nvidia-lowlatency
NVidia driver version: 565.57.01, CUDA 12.7
AMD Epyc CPUs

Approximately 50 error messages appear after system boot. These messages display various memory addresses with errors similar to the following:

`NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88fe4,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR`

We would greatly appreciate more information on whether this is indicative of an underlying issue or if we can continue to use the GPUs/server as normal.

gauravjuvekar, 2024-12-05:
This is not a functional issue, but we have seen minor perf impact in some cases due to printing so many error messages. You should be able to continue using the GPUs normally and ignore these prints.

It's fixed internally now and should show up in a future release.

twh, 2025-01-21:
On 565.57.01-0ubuntu0.22.04.4 I seeing these followed up with a shutdown, is this related or is this an actual issue with the memory register?

[Fri Jan 17 11:56:06 2025] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88ed4,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[Fri Jan 17 11:56:06 2025] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88ed8,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[Fri Jan 17 11:56:06 2025] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88edc,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[Fri Jan 17 11:56:06 2025] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88fe4,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[Fri Jan 17 12:07:02 2025] NVRM: knvlinkCoreShutdownDeviceLinks_IMPL: Need to shutdown all links unilaterally for GPU0

AGenchev, 2025-01-27:
I have this bug when in /etc/default/grub it is set iommu=pt
kernel is 6.12.10-arch1, nvidia is A100/80G

Abatpool, 2025-02-18:
HI @gauravjuvekar . **There is DGX A100 station which 4 A100 80 GB GPU. There was a automatic system reboot in DGX A100 station. Tried to see if kernel panic to cause this ,. To check kernel panic i used "sudo journalctl -k -b -1", i found the errors as below:-**

Feb 17 08:00:34 abggdna-DGX-Station-A100 kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: GPU lost from the bus [NV_ERR_GPU_IS_LOST] (0x0000000F) returned from rmStatus @ kernel_ce_gv100.c:106
Feb 17 08:00:34 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:00:34 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:00:34 abggdna-DGX-Station-A100 kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: GPU lost from the bus [NV_ERR_GPU_IS_LOST] (0x0000000F) returned from rmStatus @ kernel_ce_gv100.c:106
Feb 17 08:00:34 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:00:34 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:00:34 abggdna-DGX-Station-A100 kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: GPU lost from the bus [NV_ERR_GPU_IS_LOST] (0x0000000F) returned from rmStatus @ kernel_ce_gv100.c:106
Feb 17 08:00:42 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:04 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:04 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:04 abggdna-DGX-Station-A100 kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: GPU lost from the bus [NV_ERR_GPU_IS_LOST] (0x0000000F) returned from rmStatus @ kernel_ce_gv100.c:106
Feb 17 08:01:04 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:04 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:04 abggdna-DGX-Station-A100 kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: GPU lost from the bus [NV_ERR_GPU_IS_LOST] (0x0000000F) returned from rmStatus @ kernel_ce_gv100.c:106
Feb 17 08:01:04 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:04 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:04 abggdna-DGX-Station-A100 kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: GPU lost from the bus [NV_ERR_GPU_IS_LOST] (0x0000000F) returned from rmStatus @ kernel_ce_gv100.c:106
Feb 17 08:01:06 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:27 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:35 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:35 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:35 abggdna-DGX-Station-A100 kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: GPU lost from the bus [NV_ERR_GPU_IS_LOST] (0x0000000F) returned from rmStatus @ kernel_ce_gv100.c:106
Feb 17 08:01:35 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:35 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:35 abggdna-DGX-Station-A100 kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: GPU lost from the bus [NV_ERR_GPU_IS_LOST] (0x0000000F) returned from rmStatus @ kernel_ce_gv100.c:106
Feb 17 08:01:35 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:35 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:01:35 abggdna-DGX-Station-A100 kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: GPU lost from the bus [NV_ERR_GPU_IS_LOST] (0x0000000F) returned from rmStatus @ kernel_ce_gv100.c:106
Feb 17 08:01:42 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:02:05 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:02:05 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:02:05 abggdna-DGX-Station-A100 kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: GPU lost from the bus [NV_ERR_GPU_IS_LOST] (0x0000000F) returned from rmStatus @ kernel_ce_gv100.c:106
Feb 17 08:02:05 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
Feb 17 08:02:05 abggdna-DGX-Station-A100 kernel: NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x0000000f for fn 76!
[... cut ...]
[  201.522692] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88edc,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
[  201.522697] NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x88fe4,  regvalue: 0xbadf5040,  error code: NV_PPRIV_SYS_PRI_ERROR_CODE_FECS_PRI_CLIENT_ERR
"

CAn you let me know what cause system reboot and what does this error messages mean and how do i fix issues ? However nvidia-smi and coker kernel running smoothly post that

gauravjuvekar, 2025-02-18:
This bug should be fixed on 570.86.15 and later releases for Ampere architecture GPUs only. Fixes for other architectures will come in a later release.

gauravjuvekar, 2025-02-18:
@Abatpool  The system reboot `GPU lost from the bus [NV_ERR_GPU_IS_LOST]` seems to be unrelated to this issue. Please open a separate issue with nvidia-bug-report.sh result attached.

kylam-wevo, 2025-06-03:
Faced the same issue. When will this be fixed ?

francoism90, 2025-07-15:
Same issue over here:

```bash
... <just-flood-of-messages>
[33196.994580] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33198.994568] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33200.994532] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33202.994535] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33204.994527] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33206.994532] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33208.994572] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33210.994502] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33212.994525] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33214.994791] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33216.994594] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33218.994824] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33220.994545] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33222.994498] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33224.994772] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33226.994548] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
[33228.994617] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!
```

Any workaround? I'm on Fedora 42 with latest open nvidia drivers and supported card.

francoism90, 2025-12-27:
I really hope it's not a hardware issue, and only related to the firmware. :/

gauravjuvekar, 2026-01-16:
https://github.com/NVIDIA/open-gpu-kernel-modules/issues/688#issuecomment-2667062738

Closing.

The 580.119.02 log and `[33196.994580] NVRM: _issueRpcAndWait: rpcSendMessage failed with status 0x00000011 for fn 76!` seem like a different bugs.

closed 2026-01-16 by gauravjuvekar

Cyborgscode, 2026-01-16:
> OS: Fedora 42 Kernel: 6.17.13 Version : 580.119.02 Release : 1.fc42 Architecture: x86_64 Install Date: So 21 Dez 2025 12:03:19 CET
> 
> Errormessages(**1 Million+ in dmesg**):
> 
> Dec 27 12:28:45 eve kernel: NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x110100, regvalue: 0 xbadf5620, error code: Unknown SYS_PRI_ERROR_CODE Dec 27 12:28:45 eve kernel: NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x110100, regvalue: 0 xbadf5620, error code: Unknown SYS_PRI_ERROR_CODE Dec 27 12:28:45 eve kernel: NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0x110100, regvalue: 0 xbadf5620, error code: Unknown SYS_PRI_ERROR_CODE
> 
> Cause:
> 
> Power Failure on PCIE-Addition-Powerconnector
> 
> Solution:
> 
> Rework of all Power-Supply-Connections
> 
> REQUEST FOR ENHANCEMENT:
> 
> Limit count of messages to "a few" NOT A MILLION+!!!!

This was caused by an overpowered memory bank on the mainboard or a faulty connection on the pcie plug

Gfx card and memory working flawless for weeks now, after reconnecting all power connectors and setting memory bank voltage to auto.