NVIDIA/open-gpu-kernel-modules #700: akmods-nvidia-open requires adding rpmfusion-nonfree-release-tainted on RPM distro, and can't be concurrently installed for testing

westurner, 2024-08-29:
### NVIDIA Open GPU Kernel Modules Version

*

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Fedora 40 (doesn't have `lsb_release -d`, it has `(source /etc/os-release; echo "$PRETTY_NAME)")

### Kernel Release

*

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

*

### Describe the bug

akmods-nvidia-open requires adding rpmfusion-nonfree-release-tainted on RPM distro,
and can't be concurrently installed for testing

- https://rpmfusion.org/Howto/NVIDIA#Kernel_Open says to install the tainted repo first, and then install akmods-nvidia-open

### To Reproduce

```sh
sudo rpm-ostree install --apply-live rpmfusion-nonfree-release-tainted
sudo sh -c 'rpm-ostree update --uninstall akmod-nvidia --install akmod-nvidia-open && reboot'
```

### Bug Incidence

Once

### nvidia-bug-report.log.gz

There is no bug report because it's an issue with the downstream install UX, which isn't beyond the scope of this project. And downstream CI and keys too really. COPR OpenSUSE OBS may be able to build packages with a trusted MOK SecureBoot key.

### More Info

- This gh issue template asks: "I confirm that this does not happen with the proprietary driver package."
  - This would be much easier to test if it were possible to install both the proprietary NVIDIA module and the open NVIDIA module
    (which also loads a proprietary blob that *taints* the kernel hence the rpmfusion repo names,
    and so can be distributed by RPMfusion, and maybe Fedora COPR or OpenSuse OBS instead?)

The current Fedora Silverblue (`rpm-ostree`) install UX (for the proprietary module) also includes:

- https://github.com/fedora-silverblue/issue-tracker/issues/499 ; you must have akmods locally sign each gpu package with a key that you enroll in the bios for SecureBoot to work
  - https://github.com/CheariX/silverblue-akmods-keys#is-this-the-final-solution
- Bazzite is an rpm-ostree distro for gaming (\* with an fsync kernel and focused-app renice) which runs on HTPC, mobile Steam Deck and Rog Ally devices, and desktop systems with Gnome or KDE. They ship ProtonUp-QT and and the Steam client. Their README also describes steps for SecureBoot with Bazzite: https://github.com/ublue-os/bazzite?#secure-boot
- I got into this because Gnome Night Light isn't working but should be and neither is `xrandr --gamma`, and I don't want to install ddcutil and i2c-dev to emulate Gnome Night Light, which should be working with the new GAMMA_LUT support in the NVIDIA proprietary driver (too?); so now do I need to upgrade to fc41 today to drop the blue gamma channel like used to work with `xgamma --gamma 1:1:0.5` before they added Night Light support to Gnome and Wayland.

But to *safely* do this, I think I actually need to modify what's in `/etc/yum.repos.d/rpmfusion-nonfree-nvidia-driver` to have `includepkgs=akmods-nvidia-open` and are there additional package dependencies that require the additional package repo?

```sh
sudo rpm-ostree install rpmfusion-nonfree-release-tainted
reboot
sudo rpm-ostree update --uninstall akmod-nvidia --install akmod-nvidia-open
```

https://download1.rpmfusion.org/nonfree/fedora/tainted/40/x86_64/a/

Hopefully it's somehow possible to install akmods-nvidia and akmods-nvidia-open - without so many rpmfusion repos - for testing at least.

westurner, 2024-08-29:
> *can't be concurrently installed*

How true is this? The `rpm-ostree` command in the issue desc uninstalls akmods-nvidia first and then installs akmods-nvidia-open, but that's disadvantageous for differential testing as is already requested by the bug issue template

leigh123linux, 2024-08-30:
This is a rpm-ostree issue, it's a useless package installer.

```
leigh@leigh-pc:~$ sudo dnf swap akmod-nvidia akmod-nvidia-open
Updating and loading repositories:
Repositories loaded.
Package                                      Arch      Version                                      Repository                             Size
Removing:
 akmod-nvidia                                x86_64    3:560.35.03-1.fc42                           @commandline                       92.2 KiB
Installing:
 akmod-nvidia-open                           x86_64    3:560.35.03-1.fc41                           rpmfusion-nonfree-tainted          14.7 MiB

Transaction Summary:
 Installing:        1 packages
 Removing:          1 packages

Total size of inbound packages is 15 MiB. Need to download 15 MiB.
After this operation 15 MiB will be used (install 15 MiB, remove 92 KiB).
Is this ok [y/N]: y
[1/1] akmod-nvidia-open-3:560.35.03-1.fc41.x86_64                                                      100% |  52.7 MiB/s |  14.7 MiB |  00m00s
-----------------------------------------------------------------------------------------------------------------------------------------------
[1/1] Total                                                                                            100% |  38.4 MiB/s |  14.7 MiB |  00m00s
Running transaction
[1/4] Verify package files                                                                             100% |  25.0   B/s |   1.0   B |  00m00s
[2/4] Prepare transaction                                                                              100% |   6.0   B/s |   2.0   B |  00m00s
[3/4] Installing akmod-nvidia-open-3:560.35.03-1.fc41.x86_64                                           100% | 368.7 MiB/s |  14.7 MiB |  00m00s
>>> Running post-install scriptlet: akmod-nvidia-open-3:560.35.03-1.fc41.x86_64
>>> Stop post-install scriptlet: akmod-nvidia-open-3:560.35.03-1.fc41.x86_64
[4/4] Erasing akmod-nvidia-3:560.35.03-1.fc42.x86_64                                                   100% |   6.0   B/s |   2.0   B |  00m00s
>>> Running post-transaction scriptlet: akmod-nvidia-open-3:560.35.03-1.fc41.x86_64
>>> Stop post-transaction scriptlet: akmod-nvidia-open-3:560.35.03-1.fc41.x86_64
Complete!
```

westurner, 2024-08-30:
If the [rpm] packages could be installed concurrently, it would be possible
to test which module has a given but.

I don't think it's an rpm-ostree issue, I think it's an RPMfusion package
issue; that may be remedyable if:

- [ ] change file paths in akmods-nvidia-open
- [ ] host both modules in the same repo
- [ ] improve the RPMfusion docs in regards to the new open module

leigh123linux, 2024-08-30:
If dnf works fine, no issue, then the only other thing it could be is rpm-ostree.
If it was a rpmfusion packaging issue it would also fail with dnf, it doesn't.

P.S Why bug nvidia with this, go and bug rpm-ostree instead.

westurner, 2024-08-30:
dnf has the same problems: (1) you cannot install both akmods-nvidia and
akmods-nvidia-open and switch between them, and (2) there's yet another
third-party repo with a built copy to install and check with every `dnf
update` or `rpm-ostree upgrade`, and (3) you must ensure repo priorities
are correct and/or include_pkgs= or exclude_pkgs= in /etc/yum.repos.d so
that a kernel from another repo is not installed instead of whatever is
named 'kernel' from the updates repo whose gpg signature must be trusted.

rpm-ostree installs packages with dnf, on top of a built and signed Native
Containers root image, with a /etc and /usr overlays.

A fedora-toolbox podman rootless container can run regular dnf as root;
with a Containerfile to include the requisite `dnf install` commands
directly or with a shell script that calls a provisioner to provision an
updated, reproducible image.

Fedora Silverblue and derivatives ship fedora/toolbox for working with
local guest  containers. distrobox also mounts very many paths into a
rootless container automatically.

The Bazzite image already includes the akmods-nvidia rpmfusion repo which
it checks for updates every time

leigh123linux, 2024-08-30:
Install akmod-nvidia-open installs both and compiles both modules, I doubt there will be any method to switch them as the share the same name for the modules

```
leigh@leigh-pc:~$ sudo dnf install akmod-nvidia-open
Updating and loading repositories:
Repositories loaded.
Package                                    Arch     Version                                     Repository                                 Size
Installing:
 akmod-nvidia-open                         x86_64   3:560.35.03-1.fc41                          rpmfusion-nonfree-tainted              14.7 MiB
Installing dependencies:
 akmod-nvidia                              x86_64   3:560.35.03-1.fc41                          rpmfusion-nonfree-nvidia-driver        92.2 KiB
 akmods                                    noarch   0.5.10-6.fc41                               fedora                                 49.1 KiB
 egl-gbm                                   i686     2:1.1.2-1.fc41                              fedora                                 32.4 KiB
 egl-gbm                                   x86_64   2:1.1.2-1.fc41                              fedora                                 29.2 KiB
 egl-wayland                               i686     1.1.16-1.fc41                               fedora                                 86.3 KiB
 egl-wayland                               x86_64   1.1.16-1.fc41                               fedora                                 83.3 KiB
 kernel-devel-matched                      x86_64   6.11.0-0.rc5.43.fc41                        fedora                                  0.0   B
 kmodtool                                  noarch   1.1-11.fc41                                 fedora                                 27.5 KiB
 libglvnd-opengl                           i686     1:1.7.0-5.fc41                              fedora                                128.4 KiB
 nvidia-modprobe                           x86_64   3:560.35.03-1.fc41                          rpmfusion-nonfree-nvidia-driver        51.1 KiB
 nvidia-settings                           x86_64   3:560.35.03-1.fc41                          rpmfusion-nonfree-nvidia-driver         4.3 MiB
 xorg-x11-drv-nvidia                       x86_64   3:560.35.03-3.fc41                          rpmfusion-nonfree-updates-testing     190.4 MiB
 xorg-x11-drv-nvidia-kmodsrc               x86_64   3:560.35.03-3.fc41                          rpmfusion-nonfree-updates-testing      53.7 MiB
 xorg-x11-drv-nvidia-libs                  i686     3:560.35.03-3.fc41                          rpmfusion-nonfree-updates-testing     130.6 MiB
 xorg-x11-drv-nvidia-libs                  x86_64   3:560.35.03-3.fc41                          rpmfusion-nonfree-updates-testing     289.1 MiB
Installing weak dependencies:
 xorg-x11-drv-nvidia-cuda-libs             x86_64   3:560.35.03-3.fc41                          rpmfusion-nonfree-updates-testing     194.1 MiB
 xorg-x11-drv-nvidia-power                 x86_64   3:560.35.03-3.fc41                          rpmfusion-nonfree-updates-testing     227.4 KiB

Transaction Summary:
 Installing:       18 packages

Total size of inbound packages is 341 MiB. Need to download 341 MiB.
After this operation 878 MiB will be used (install 878 MiB, remove 0 B).
Is this ok [y/N]: y
[ 1/18] kmodtool-0:1.1-11.fc41.noarch                                                                  100% |  35.2 KiB/s |  15.8 KiB |  00m00s
[ 2/18] akmods-0:0.5.10-6.fc41.noarch                                                                  100% |  63.9 KiB/s |  29.1 KiB |  00m00s
[ 3/18] kernel-devel-matched-0:6.11.0-0.rc5.43.fc41.x86_64                                             100% | 289.9 KiB/s | 176.3 KiB |  00m01s
[ 4/18] nvidia-modprobe-3:560.35.03-1.fc41.x86_64                                                      100% | 457.1 KiB/s |  32.5 KiB |  00m00s
[ 5/18] akmod-nvidia-open-3:560.35.03-1.fc41.x86_64                                                    100% |  20.2 MiB/s |  14.7 MiB |  00m01s
[ 6/18] akmod-nvidia-3:560.35.03-1.fc41.x86_64                                                         100% | 968.8 KiB/s |  41.7 KiB |  00m00s
[ 7/18] libglvnd-opengl-1:1.7.0-5.fc41.i686                                                            100% |  93.7 KiB/s |  37.9 KiB |  00m00s
[ 8/18] nvidia-settings-3:560.35.03-1.fc41.x86_64                                                      100% |   8.8 MiB/s |   1.7 MiB |  00m00s
[ 9/18] egl-wayland-0:1.1.16-1.fc41.i686                                                               100% |  94.4 KiB/s |  46.5 KiB |  00m00s
[10/18] xorg-x11-drv-nvidia-power-3:560.35.03-3.fc41.x86_64                                            100% |   1.2 MiB/s | 103.1 KiB |  00m00s
[11/18] egl-gbm-2:1.1.2-1.fc41.i686                                                                    100% |  19.9 KiB/s |  21.7 KiB |  00m01s
[12/18] egl-gbm-2:1.1.2-1.fc41.x86_64                                                                  100% |  17.8 KiB/s |  20.6 KiB |  00m01s
[13/18] egl-wayland-0:1.1.16-1.fc41.x86_64                                                             100% |  33.7 KiB/s |  44.0 KiB |  00m01s
[14/18] xorg-x11-drv-nvidia-libs-3:560.35.03-3.fc41.i686                                               100% |  23.2 MiB/s |  31.2 MiB |  00m01s
[15/18] xorg-x11-drv-nvidia-libs-3:560.35.03-3.fc41.x86_64                                             100% |  26.9 MiB/s |  62.6 MiB |  00m02s
[... cut ...]
leigh@leigh-pc:~$ rpm -ql kmod-nvidia-6.11.0-0.rc5.43.fc41.x86_64-560.35.03-1.fc41.x86_64
/lib/modules/6.11.0-0.rc5.43.fc41.x86_64/extra
/lib/modules/6.11.0-0.rc5.43.fc41.x86_64/extra/nvidia
/lib/modules/6.11.0-0.rc5.43.fc41.x86_64/extra/nvidia/nvidia-drm.ko.xz
/lib/modules/6.11.0-0.rc5.43.fc41.x86_64/extra/nvidia/nvidia-modeset.ko.xz
/lib/modules/6.11.0-0.rc5.43.fc41.x86_64/extra/nvidia/nvidia-peermem.ko.xz
/lib/modules/6.11.0-0.rc5.43.fc41.x86_64/extra/nvidia/nvidia-uvm.ko.xz
/lib/modules/6.11.0-0.rc5.43.fc41.x86_64/extra/nvidia/nvidia.ko.xz
```

leigh123linux, 2024-08-30:
Using the /etc/rpm/macros.nvidia-kmod method is the easiest way to switch

https://rpmfusion.org/Howto/NVIDIA?highlight=%28%5CbCategoryHowto%5Cb%29#Kernel_Open

kwizart, 2024-08-30:
On ostree, you can install this package that will tune the option to build the kernel-open instead of the closed counterpart.

https://download.copr.fedorainfracloud.org/results/kwizart/nvidia-driver-rawhide/fedora-40-x86_64/07955716-akmod-nvidia-default-open/akmod-nvidia-default-open-1-1.fc40.x86_64.rpm

westurner, 2024-12-22:
Here's those as originally formatted with a quote indent `>`, due to github's regression to failing at newlines on inbound email to issues these days:

> If the [rpm] packages could be installed concurrently, it would be possible
> to test which module has a given but.
>
> I don't think it's an rpm-ostree issue, I think it's an RPMfusion package
> issue; [...]

...

> dnf has the same problems: (1) you cannot install both akmods-nvidia and
> akmods-nvidia-open and switch between them, and (2) there's yet another
> third-party repo with a built copy to install and [...]