NVIDIA/open-gpu-kernel-modules #703: Why is the proprietary driver so much bigger then the open source one?

birdie-github, 2024-09-08:
nvidia.ko compiled using the closed source blob is several times larger than nvidia.ko compiled from your open source.

Why is there such a huge discrepancy?

steven-bellock, 2024-09-08:
Much of the code in the proprietary driver is offloaded to the GSP microprocessor that resides on the GPU. This firmware is usually found in `/lib/firmware/nvidia/`.

birdie-github, 2024-09-08:
> Much of the code in the proprietary driver is offloaded to the GSP microprocessor that resides on the GPU. This firmware is usually found in `/lib/firmware/nvidia/`.

1. The proprietary driver also loads/uses the GSP firmware.
2. The GSP firmware itself is running on the GPU using the RISC ISA, so I don't understand how those could be related

steven-bellock, 2024-09-08:
>The proprietary driver also loads/uses the GSP firmware.

Sure, but it also needs to support GPUs that don't have GSP, and that code needs to reside in `nvidia.ko`, leading to its large size. The open source `nvidia.ko` only works with code offloaded to GSP, and as such can trim away code for the older GPUs, since GSP didn't exist pre-Turing.

birdie-github, 2024-09-09:
I get it, it has the firmware for the previous generation cards that are not covered by the external GSP files.

closed 2024-09-09 by birdie-github