NVIDIA/open-gpu-kernel-modules #706: intel gpu top allways killed after kernel  6.10.10 and causing a kernel panic

meduk0, 2024-09-17:
### NVIDIA Open GPU Kernel Modules Version

560.35.03 

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

6.11.0-2-cachyos

### Kernel Release

6.10.10 -> 6.11.0-2

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

rtx 3050 laptop 6gb 

### Describe the bug

```
> sudo intel_gpu_top                                            fish-1 | 0 (0.030s) < 09:34:39
[sudo] password for meduko:
fish: Job 1, 'sudo intel_gpu_top' terminated by signal SIGKILL (Forced quit)
```

```
> sudo  dmesg 
[ 2462.104713] BUG: kernel NULL pointer dereference, address: 0000000000000035
[ 2462.104726] #PF: supervisor read access in kernel mode
[ 2462.104731] #PF: error_code(0x0000) - not-present page
[ 2462.104735] PGD 0 P4D 0
[ 2462.104743] Oops: Oops: 0000 [#3] PREEMPT SMP NOPTI
[ 2462.104752] CPU: 15 UID: 0 PID: 25352 Comm: intel_gpu_top Tainted: G     UD    OE      6.11.0-2-cachyos #1 4e8f3da2cd7dfb196c69514fe8b3b64a033481c6
[ 2462.104765] Tainted: [U]=USER, [D]=DIE, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[ 2462.104768] Hardware name: Dell Inc. Dell G15 5530/04TT83, BIOS 1.19.0 08/16/2024
[ 2462.104773] Sched_ext: lavd (enabled+all), task: runnable_at=-3ms
[ 2462.104777] RIP: 0010:clkpm_show+0x47/0x70
[ 2462.104795] Code: 43 48 2d c8 00 00 00 48 8b 50 10 48 8b 42 10 48 85 c0 74 16 48 8b 42 38 48 85 c0 74 0d 80 78 6c 00 74 2a 48 8b 80 b8 00 00 00 <0f> b6 50 35 48 c7 c6 6c c4 a5 b5 83 e2 01 e8 46 ed d0 ff 48 98 c3
[ 2462.104800] RSP: 0018:ffffadf7ce943c48 EFLAGS: 00010202
[ 2462.104807] RAX: 0000000000000000 RBX: ffffffffb6350120 RCX: ffffffffb6350120
[ 2462.104811] RDX: ffff893dc20a4000 RSI: ffffffffb6350120 RDI: ffff893e4d481000
[ 2462.104817] RBP: ffffffffb555a010 R08: ffff893dc20ba0c8 R09: ffff893ed0422600
[ 2462.104821] R10: ffffadf7ce943c80 R11: 0000000000001000 R12: ffffadf7ce943d00
[ 2462.104824] R13: ffffadf7ce943cd8 R14: 0000000000000001 R15: 0000000000000001
[ 2462.104829] FS:  00007846fbe03740(0000) GS:ffff89413f780000(0000) knlGS:0000000000000000
[ 2462.104834] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 2462.104839] CR2: 0000000000000035 CR3: 000000026b7b2000 CR4: 0000000000f50ef0
[ 2462.104843] PKRU: 55555554
[ 2462.104847] Call Trace:
[ 2462.104853]  <TASK>
[ 2462.104859]  ? __die_body.cold+0x8/0x12
[ 2462.104870]  ? page_fault_oops+0x15a/0x2e0
[ 2462.104880]  ? exc_page_fault+0x81/0x190
[ 2462.104887]  ? asm_exc_page_fault+0x26/0x30
[ 2462.104899]  ? clkpm_show+0x47/0x70
[ 2462.104905]  dev_attr_show+0x19/0x40
[ 2462.104916]  sysfs_kf_seq_show+0xa8/0xf0
[ 2462.104925]  seq_read_iter+0x11e/0x470
[ 2462.104933]  vfs_read+0x344/0x470
[ 2462.104942]  __x64_sys_read+0x72/0xf0
[ 2462.104950]  do_syscall_64+0x82/0x190
[ 2462.104962]  ? __x64_sys_openat+0x1f5/0x230
[ 2462.104970]  ? syscall_exit_to_user_mode+0x10/0x1e0
[... 96 lines cut ...]
  System Temperatures: cpu: 34.0 C mobo: 34.0 C sodimm: SODIMM C
  Fan Speeds (rpm): cpu: 0 fan-1: 0
Info:
  Memory: total: 16 GiB note: est. available: 15.31 GiB used: 3.04 GiB (19.9%)
  Processes: 347 Uptime: 34m Init: systemd
  Packages: 1484 Compilers: clang: 18.1.8 gcc: 14.2.1 Shell: prime-run
    inxi: 3.3.35
```

### To Reproduce

boot with 6.10.10 or 6.11.0-2 on both arch linux  , endeavors os and cachyos and after proper intel and nvidia open driver installation just launch `sudo intel_gpu_top ` which cause the kernel panic after ` fish: Job 1, 'sudo intel_gpu_top' terminated by signal SIGKILL (Forced quit)` and it persist even after reboot or reinstall (it never happened before ) 

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/17025332/nvidia-bug-report.log.gz)

### More Info

other than this the peroformance is stable ( for now ) and i usually play battlefield  1 and use blender 
it is an old bug that return according to [intel_gpu_top](https://forums.developer.nvidia.com/t/loading-nvidia-drm-causes-intel-gpu-top-to-fail/288131)

mtijanic, 2024-09-17:
Hi! Forgive the obvious question, but what happens if the nvidia driver is not loaded (or even installed) and you run the tool?

meduk0, 2024-09-17:
sorry man but it turn out that the force_pcme ( a kernel parameter that is causing this issue (the nvidia card doesn't support it and forcing it cause stutter and driver panic for the hybrid power system + longer time to respond from the nvidia driver

closed 2024-09-17 by meduk0