NVIDIA/open-gpu-kernel-modules #708: drm_mode_config_funcs.output_poll_changed got removed in 6.12-rc1 and will prevent compiling of the nvidia driver

philmmanjaro, 2024-09-21:
### NVIDIA Open GPU Kernel Modules Version

560.35.03 (open and propritary)

### Operating System and Version

Manjaro Linux 24.1.0-pre1

### Kernel Release

6.12 (1868f9d0260e9)

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Build Command

`make -C kernel SYSSRC=/usr/lib/modules/"${_kernver}/build" module`

### Terminal output/Build Log

build  log: [nvidia_202409211314.log](https://github.com/user-attachments/files/17084306/nvidia_202409211314.log)

### More Info

Drivers implementing fbdev emulation use drm_kms_helper_hotplug_event() to call this hook to inform the fbdev helper of output changes. This hook is deprecated, drivers should instead implement fbdev support with struct drm_client, which takes care of any necessary hotplug event forwarding already without further involvement by the driver.

philmmanjaro, 2024-09-21:
Also reported at the forum: https://forums.developer.nvidia.com/t/error-const-struct-drm-mode-config-funcs-has-no-member-named-output-poll-changed/306078. Having a fix in early development cycle of 6.12 would be the right move here.

aritger, 2024-09-25:
Thanks.  We're aware of the issue, and a fix should be in a release shortly.  You can refer to this as NVIDIA internal bug 4854525.

ptr1337, 2024-09-26:
@philmmanjaro 
```diff
From 94ad29389cd5129401adea6925f987f4529abb1c Mon Sep 17 00:00:00 2001
From: Peter Jung <admin@ptr1337.dev>
Date: Thu, 26 Sep 2024 14:53:31 +0200
Subject: [PATCH 5/5] 6.12: drm_outpull_pill changed check

---
 kernel/conftest.sh                      | 23 ++++++++++++++++++++
 kernel/nvidia-drm/nvidia-drm-drv.c      |  4 ++++
 kernel/nvidia-drm/nvidia-drm-sources.mk |  1 +
 3 files changed, 28 insertions(+)

diff --git a/kernel/conftest.sh b/kernel/conftest.sh
index 4a239e63..4d6fff64 100755
--- a/kernel/conftest.sh
+++ b/kernel/conftest.sh
@@ -6631,6 +6631,29 @@ compile_test() {
             compile_check_conftest "$CODE" "NV_DRM_FBDEV_TTM_SETUP_PRESENT" "" "functions"
         ;;
 
+        drm_output_poll_changed)
+            #
+            # Determine whether drm_mode_config_funcs.output_poll_changed
+            # callback is present
+            #
+            # Removed by commit 446d0f4849b1 ("drm: Remove struct
+            # drm_mode_config_funcs.output_poll_changed") in v6.12. Hotplug
+            # event support is handled through the fbdev emulation interface
+            # going forward.
+            #
+            CODE="
+            #if defined(NV_DRM_DRM_MODE_CONFIG_H_PRESENT)
+            #include <drm/drm_mode_config.h>
+            #else
+            #include <drm/drm_crtc.h>
+            #endif
+            int conftest_drm_output_poll_changed_available(void) {
+                return offsetof(struct drm_mode_config_funcs, output_poll_changed);
+            }"
+
+            compile_check_conftest "$CODE" "NV_DRM_OUTPUT_POLL_CHANGED_PRESENT" "" "types"
+         ;;
+
         drm_aperture_remove_conflicting_pci_framebuffers)
             #
             # Determine whether drm_aperture_remove_conflicting_pci_framebuffers is present.
diff --git a/kernel/nvidia-drm/nvidia-drm-drv.c b/kernel/nvidia-drm/nvidia-drm-drv.c
index 50028c26..034dda66 100644
--- a/kernel/nvidia-drm/nvidia-drm-drv.c
+++ b/kernel/nvidia-drm/nvidia-drm-drv.c
@@ -126,6 +126,7 @@ static const char* nv_get_input_colorspace_name(
 
 #if defined(NV_DRM_ATOMIC_MODESET_AVAILABLE)
 
+#if defined(NV_DRM_OUTPUT_POLL_CHANGED_PRESENT)
 static void nv_drm_output_poll_changed(struct drm_device *dev)
 {
     struct drm_connector *connector = NULL;
@@ -169,6 +170,7 @@ static void nv_drm_output_poll_changed(struct drm_device *dev)
     nv_drm_connector_list_iter_end(&conn_iter);
 #endif
 }
+#endif /* NV_DRM_OUTPUT_POLL_CHANGED_PRESENT */
 
 static struct drm_framebuffer *nv_drm_framebuffer_create(
     struct drm_device *dev,
@@ -206,7 +208,9 @@ static const struct drm_mode_config_funcs nv_mode_config_funcs = {
     .atomic_check  = nv_drm_atomic_check,
     .atomic_commit = nv_drm_atomic_commit,
 
+    #if defined(NV_DRM_OUTPUT_POLL_CHANGED_PRESENT)
     .output_poll_changed = nv_drm_output_poll_changed,
+    #endif
 };
 
 static void nv_drm_event_callback(const struct NvKmsKapiEvent *event)
diff --git a/kernel/nvidia-drm/nvidia-drm-sources.mk b/kernel/nvidia-drm/nvidia-drm-sources.mk
index 247096b7..f5964a57 100644
--- a/kernel/nvidia-drm/nvidia-drm-sources.mk
+++ b/kernel/nvidia-drm/nvidia-drm-sources.mk
@@ -131,3 +131,4 @@ NV_CONFTEST_TYPE_COMPILE_TESTS += drm_aperture_remove_conflicting_pci_framebuffe
 NV_CONFTEST_TYPE_COMPILE_TESTS += drm_mode_create_dp_colorspace_property_has_supported_colorspaces_arg
 NV_CONFTEST_TYPE_COMPILE_TESTS += drm_syncobj_features_present
 NV_CONFTEST_TYPE_COMPILE_TESTS += drm_unlocked_ioctl_flag_present
+NV_CONFTEST_TYPE_COMPILE_TESTS += drm_output_poll_changed
-- 
2.46.2
``` 

Can you check this patch? Has been extracted from the 550 driver.

philmmanjaro, 2024-09-27:
@ptr1337 applying that to all drivers, 390xx, 470xx and 560xx made it compile. For the older drivers I had to rewrite the 550xx patch a little. Thx for letting me know.

ptr1337, 2024-09-27:
> [@ptr1337](https://github.com/ptr1337) applying that to all drivers, 390xx, 470xx and 560xx made it compile. For the older drivers I had to rewrite the 550xx patch a little. Thx for letting me know.

I think we can close this issue then, since its queued in 550.120 already, which is releated. The upcoming 560/565 should have it likely in

philmmanjaro, 2024-09-30:
Well, seems your issue #712 might be related to this one. Still have to get feedback if those drivers actually work ...

gilvbp, 2024-09-30:
> Well, seems your issue [#712](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/712) might be related to this one. Still have to get feedback if those drivers actually work ...

@ptr1337 @philmmanjaro  No, Gnome doesn't even start.

joanbm, 2024-10-01:
At least for nvidia-470xx, leaving out the `.output_poll_changed` member is insufficient, at the very least it makes the right modes not appear in `/sys/class/drm/card*/modes`, not sure if it actually breaks anything important. I have a workaround for it [here](https://gist.github.com/joanbm/a6d3f7f873a60dec0aa4a734c0f1d64e) which may be easy to port to the newer drivers.

mtijanic, 2024-10-22:
This should now be fixed in all live driver branches, closing.

ofcaah, 2025-01-29:
just tried installing freshly released 550.40.83 on kernel 6.12.11-amd64 and bumped into this. I'll try applying patch manually, but I'm reporting it here to let someone know they dropped the ball.

Binary-Eater, 2025-01-29:
@ofcaah Our latest 550 driver is 550.144.03.

https://www.nvidia.com/en-us/drivers/details/238858/

Binary-Eater, 2025-01-29:
Ah, I see where the confusion is coming from. @ofcaah that release comes from a Vulkan driver branch that is not the mainstream 550 driver. It's expected to not have kernel compat fixes.

ofcaah, 2025-01-29:
> Ah, I see where the confusion is coming from. [@ofcaah](https://github.com/ofcaah) that release comes from a Vulkan driver branch that is not the mainstream 550 driver. It's expected to not have kernel compat fixes.

I was investigating those drivers regarding broken mesh shaders, seeing a bugfix concerning those in changelog. More details here: https://github.com/ValveSoftware/Proton/issues/8408

Unfortunately, game's as broken on 550.40.83, as it is on 570.86.15.

Sorry for the noise and thank you for clarifying.

Binary-Eater, 2025-01-29:
@ofcaah good news is we have an internal bug for that issue and are working on the problem. Nvbug 5059946 for reference.

ryzendew, 2025-01-30:
> [@ofcaah](https://github.com/ofcaah) Our latest 550 driver is 550.144.03.
> 
> https://www.nvidia.com/en-us/drivers/details/238858/

nobody uses 550 we moved on long ago to 565 and now 570 as 550 lacks explicit sync and good wayland support

hmartinez82, 2025-05-22:
@hphilm
Would it be possible for you to provide the 470.xx compatible patch?  Or point me to a link to it? Thank you

ptr1337, 2025-05-22:
> [@hphilm](https://github.com/hphilm) Would it be possible for you to provide the 470.xx compatible patch? Or point me to a link to it? Thank you

https://github.com/CachyOS/CachyOS-PKGBUILDS/tree/master/nvidia/nvidia-470xx-utils

There you can find compat patches for 470.