NVIDIA/open-gpu-kernel-modules #709: Random boot fail to initialize with timeout or reset required

yamboo-efi, 2024-09-22:
### NVIDIA Open GPU Kernel Modules Version

560.35.03

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Gentoo Linux

### Kernel Release

6.6.52

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 4070 Ti SUPER

### Describe the bug

In a generally way nvidia driver failed to load properly. Sometimes with a error "RESET REQUIRED", other with timeout in communication with GPU.

I tried other kernels (6.8 and 6.11) with same results, and also propietary driver with/o GSP firmware loading, in all cases the result is the same.

It's strange and I cannot isolate the root  of cause, so I will try to explain the case.

I changed motherboard and CPU, from an Gigabyte X570 and Ryzen 3700X to an Gigabyte X670 adn Ryzen 9700X. The GPU are the same, 4070Ti Super. Previously the system works right with same OS and 560.35.03 driver.

After hardware upgrade, I cannot boot in a seamless way,  most of the times GPU initialization failed. Made a lot of trial and error (different kernel and nvidia driver config) with same result.

The driver produce two types of errors (I don't know why), one is a GPU communication timeout and other related to reset required.

Tried closed source driver (same issues) and disabling GSP firmware loading.

The most intrigguing thing is  that I can boot properly after some tries, generally 2 or 3, combining soft reset with cool boot. When I get a a "RESET ERROR" I do a cool boot and when get a "TIMEOUT" I do a soft reset. 

After get  a proper init, things go right.

I think that exists a issue in initialization order or logic in the combination of AGESA.

The most stable configuration (also  need some reboots to work) is using closed source with disabled GSP firmware.

If you  need a proper details or logs, please tell me what you need.

PD: I post here because as I understand, the open-source driver is the default option on 560 driver.

### To Reproduce

Almost, on any boot

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/17087923/nvidia-bug-report.log.gz)

### More Info

At least a consistent boot, but a full fix would be great.

ptr1337, 2024-09-22:
I think some AM5 Boards had problems on old BIOS Versions with 40xx gen. Did you try to upgrade to the latest available BIOS?

On my machine (9950X, Asrock x670E, 4070 Super) I do not see such issues.

yamboo-efi, 2024-10-07:
I'm in conversations with manufacturer because I'm very sure that is a hardware incompatibility between GPU and Motherboard that cause an improper initialization of GPU internals.

The issue also occurs in Windows, with latest drivers, taking sometimes a long time to boot resulting in a very very poor performance after booting in Windows.

The GPU is fine, tested on other system and check it with different tools (including nvidia MODS/MATS tools) without any error. Separated tested the Motherboard and CPU with other GPU and everything works as expected on Linux.

So I will close that issue because isn't related to a driver bug.

closed 2024-10-07 by yamboo-efi