NVIDIA/open-gpu-kernel-modules #719: WARNING: CPU: X PID: XXXX at include/linux/rwsem.h:80 follow_pte+0xf8/0x120 on resume

birdie-github, 2024-10-22:
### NVIDIA Open GPU Kernel Modules Version

565.57.01

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Fedora 40

### Kernel Release

Linux zen 6.11.4-zen3 #1 SMP PREEMPT_DYNAMIC Tue Oct 22 11:16:40 UTC 2024 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 4070 SUPER

### Describe the bug

The issue with resuming is **not** fixed in beta driver 565.57.01 :-(

There are even MORE dmesg errors than in the previous stable driver.

I'm only using + XFCE + no compositing.
```
options nvidia NVreg_EnableS0ixPowerManagement=1
options nvidia-drm modeset=1 fbdev=1
```

### To Reproduce

Suspend/resume.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[kernel-trace.txt](https://github.com/user-attachments/files/17479382/kernel-trace.txt)

aritger, 2024-10-22:
Can you please attach a full nvidia-bug-report.log.gz?

aritger, 2024-10-22:
Thanks for the log.  I've filed NVIDIA internal bug 4922186 for this.

Knowing this is specific to >= Linux kernel 6.10 helps; thanks for that isolation.
Any other isolation you can do will help a lot.  E.g.,
- Does this reproduce if forcing use of the closed kernel modules?
- Does this reproduce without NVreg_EnableS0ixPowerManagement=1?
- Does this reproduce without fbdev=1?

birdie-github, 2024-10-22:
>     * Does this reproduce if forcing use of the closed kernel modules?
Yes.
> 
>     * Does this reproduce without NVreg_EnableS0ixPowerManagement=1?
I'll try, and report later.
> 
>     * Does this reproduce without fbdev=1?
Yes.

birdie-github, 2024-10-22:
Removing `NVreg_EnableS0ixPowerManagement=1` fixes the issue.

No more dmesg spam (more than 60KB of messages).

aritger, 2024-10-22:
Thank you for those experiments.  That will help us focus our debugging.

tekstryder, 2024-10-22:
I'd been following https://github.com/NVIDIA/open-gpu-kernel-modules/issues/662 mistakenly this whole time when it's **_this_** issue that's been affecting every kernel since 6.10 release, as [mentioned](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/719#issuecomment-2429809337). I should have paid more attention to those stack traces! D'oh!

> Does this reproduce if forcing use of the closed kernel modules?

Yes

> Does this reproduce without NVreg_EnableS0ixPowerManagement=1?

I've never had `NVreg_EnableS0ixPowerManagement` set.

> Does this reproduce without fbdev=1?

Yes, and I'm unable to boot with `fbdev=1' anyhow. See [related Arch bug](https://gitlab.archlinux.org/archlinux/packaging/packages/nvidia-utils/-/issues/16)

Using the following kernel parameters:
`nvidia_drm.modeset=1 nvidia_drm.fbdev=0`

- Arch Linux | Kernel 6.10.14
- nVidia 560.35.03
- GeForce GTX 1050 Ti

The attached stacktrace repeats ~20ish times per suspend: [nvidia-sleep-stacktrace.txt](https://github.com/user-attachments/files/17480288/nvidia-sleep-stacktrace.txt)

_/usr/lib/modprobe.d/nvidia-sleep.conf:_

```
# https://download.nvidia.com/XFree86/Linux-x86_64/560.35.03/README/powermanagement.html#PreserveAllVide719f0
# Save and restore all video memory allocations.
options nvidia NVreg_PreserveVideoMemoryAllocations=1
#
# The destination should not be using tmpfs, so we prefer
# /var/tmp instead of /tmp
options nvidia NVreg_TemporaryFilePath=/var/tmp

```

birdie-github, 2024-10-22:
@tekstryder 

Please attach your nvidia-bug-report as well.

tekstryder, 2024-10-22:
> Please attach your nvidia-bug-report as well.

Edited my post and attached it.

I also just built and booted kernel 6.11.5, and suspended one time to reproduce this issue, so the `dmesg` is fresh.

[nvidia-bug-report.log](https://github.com/user-attachments/files/17481415/nvidia-bug-report.log)

ZulluBalti, 2024-10-23:
I'm also getting the same error on

[zullu@zorch ~]$ uname -a 
Linux zorch 6.10.10-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 12 Sep 2024 17:21:02 +0000 x86_64 GNU/Linux

ZulluBalti, 2024-10-23:
Happening to me also on the proprietary drivers as well on linux 6.10.10

birdie-github, 2024-10-23:
@aritger 

There's also #705 which looks similar.

I wonder if #662 #705 and this one are all somehow related because as far as I can see most people affected by them started to get issues with kernel 6.10 and later.

aritger, 2024-10-23:
If anyone is adventurous, could you retest this configuration with Linux kernel v6.12-rc1 or later?  It looks like -rc4 is currently the latest.

We're still investigating, but so far: I don't know if the 'WARNING: CPU: 6 PID: 7530 at include/linux/rwsem.h:80 follow_pte+0xf8/0x120' message is synonymous with the failure, or just coincidental.  We've seen that warning and backtrace in other scenarios where suspend/resume still worked.

That particular warning was introduced by this kernel commit:

    https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/mm/memory.c?id=c5541ba378e3d36ea88bf5839d5b23e33e7d1627

There was then some upstream kernel discussion about various drivers that were impacted by it:

    https://lore.kernel.org/linux-mm/20240712080414.GA47643@google.com/T/#u
    https://lore.kernel.org/all/20240712124359.3199-1-spasswolf@web.de/

This area was refactored more upstream:

    https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?id=b0a1c0d0edcd75a0f8ec5fd19dbd64b8d097f534
    https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?id=6da8e9634bb7e3fdad9ae0e4db873a05036c4343
    https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?id=6da8e9634bb7e3fdad9ae0e4db873a05036c4343

which is included in Linux kernel v6.12-rc1 and later.  In our testing, that resolved the warning message, but I don't have confirmation yet whether it impacted NVreg_EnableS0ixPowerManagement=1 configurations.

ptr1337, 2024-10-23:
6.12 will not work right now since its not able to open the display.
This patch on top of the open modules should fix it:
https://github.com/CachyOS/open-gpu-kernel-modules/commit/0140472cf5a3a4573d8f6b94f58bdef9e6a1ae88

aritger, 2024-10-23:
I /think/ the latest 565.xx beta should already contain compatibility fixes for 6.12-rc's, but let me know if that is not your experience.

letmeiiiin, 2024-10-24:
Can confirm you need this patch to open the display on 565.57.01 / linux 6.12-rc4.
Thx ptr1337 :)

cinnamonmatexfce, 2024-10-25:
Resume after suspend is working fine here, but I'm also having the same kind of trace.

 - KERNEL: `Linux 6.11.3-2-default GNU/Linux`

 - DISTRO: `openSUSE Tumbleweed`

 - PACKAGES:
```
zypper se -ids /^nvidia/

Loading repository data...
Reading installed packages...

S  | Name                          | Type    | Version                   | Arch   | Repository
---+-------------------------------+---------+---------------------------+--------+-----------
i+ | nvidia-compute-G06            | package | 550.127.05-27.1           | x86_64 | nvidia
i+ | nvidia-compute-G06-32bit      | package | 550.127.05-27.1           | x86_64 | nvidia
i+ | nvidia-compute-utils-G06      | package | 550.127.05-27.1           | x86_64 | nvidia
i+ | nvidia-driver-G06-kmp-default | package | 550.127.05_k6.11.3_2-27.1 | x86_64 | nvidia
i+ | nvidia-drivers-G06            | package | 550.127.05-27.1           | x86_64 | nvidia
i+ | nvidia-gl-G06                 | package | 550.127.05-27.1           | x86_64 | nvidia
i+ | nvidia-gl-G06-32bit           | package | 550.127.05-27.1           | x86_64 | nvidia
i+ | nvidia-utils-G06              | package | 550.127.05-27.1           | x86_64 | nvidia
i+ | nvidia-video-G06              | package | 550.127.05-27.1           | x86_64 | nvidia
i+ | nvidia-video-G06-32bit        | package | 550.127.05-27.1           | x86_64 | nvidia
```

```
zypper se -ids /^kernel\-/

Loading repository data...
Reading installed packages...

S  | Name                        | Type    | Version      | Arch   | Repository
---+-----------------------------+---------+--------------+--------+------------------------
i+ | kernel-default              | package | 6.11.3-1.1   | x86_64 | (System Packages)
i+ | kernel-default              | package | 6.11.3-2.1   | x86_64 | openSUSE-Tumbleweed-Oss
i  | kernel-default-devel        | package | 6.11.3-2.1   | x86_64 | openSUSE-Tumbleweed-Oss
i  | kernel-devel                | package | 6.11.3-2.1   | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-all         | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-amdgpu      | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-ath10k      | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-ath11k      | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-ath12k      | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-atheros     | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-bluetooth   | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-bnx2        | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-brcm        | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-chelsio     | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-dpaa2       | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-i915        | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-intel       | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-iwlwifi     | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-liquidio    | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-marvell     | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-media       | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-mediatek    | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-mellanox    | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-mwifiex     | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-network     | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-nfp         | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-nvidia      | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-platform    | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-prestera    | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-qcom        | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-qlogic      | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-radeon      | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-realtek     | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-serial      | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
i  | kernel-firmware-sound       | package | 20241018-1.1 | noarch | openSUSE-Tumbleweed-Oss
[... 735 lines cut ...]
ott 25 09:07:35 SUSNEPO kernel:  ? syscall_exit_to_user_mode+0x10/0x220
ott 25 09:07:35 SUSNEPO kernel:  ? do_syscall_64+0x8e/0x160
ott 25 09:07:35 SUSNEPO kernel:  ? exc_page_fault+0x73/0x170
ott 25 09:07:35 SUSNEPO kernel:  entry_SYSCALL_64_after_hwframe+0x76/0x7e
ott 25 09:07:35 SUSNEPO kernel: RIP: 0033:0x7fb48dd0fea4
ott 25 09:07:35 SUSNEPO kernel: Code: 84 00 00 00 00 00 90 90 90 90 90 90 90 90 90 90 90 90 90 90 f3 0f 1e fa 90 90 80 3d 05 ab 0f 00 00 74 13 b8 01 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 54 c3 0f 1f 00 48 83 ec 28 48 89 54 24 18 48
[... cut ...]
ott 25 09:07:35 SUSNEPO systemd-sleep[26493]: relying on kernel to perform the freeze.
ott 25 09:07:35 SUSNEPO systemd-sleep[26493]: This is a temporary downstream workaround for https://github.com/systemd/systemd/issues/33083.
ott 25 09:07:35 SUSNEPO systemd-sleep[26496]: INFO: Skip running /usr/lib/systemd/system-sleep/grub2.sleep for suspend
ott 25 09:07:35 SUSNEPO systemd-sleep[26493]: Performing sleep operation 'suspend'...
ott 25 09:07:35 SUSNEPO kernel: PM: suspend entry (deep)
ott 25 09:07:36 SUSNEPO kernel: Filesystems sync: 0.308 seconds
```

ethanh20009, 2024-10-31:
Getting the same kernel stack traces on suspend/resume on open kernel driver 565.

Causes intermittent behaviour for me. Sometimes one monitor out of 2 comes back. Sometimes it causes my monitors to swap relative positions (possibly DP-1 identifies as DP-2..)

Confirming all is solved when using an lts kernel (6.6)

SamoutL, 2024-10-31:
I still have this issue with linux lts (6.6) kernel and open kernel driver 565.

tekstryder, 2024-11-23:
> If anyone is adventurous, could you retest this configuration with Linux kernel v6.12-rc1 or later? It looks like -rc4 is currently the latest.

@aritger I wasn't adventurous enough and waited for stable kernel release.

I can confirm this issue is indeed resolved with the API changes in kernel 6.12.

- Arch Linux | Kernel 6.12.1
- nVidia 565.57.01

aritger, 2024-11-25:
Thanks for confirming.  I'm sorry it took us a bit to release a 6.12-compatible driver.

My post earlier in this issue still summarizes my understanding:
https://github.com/NVIDIA/open-gpu-kernel-modules/issues/719#issuecomment-2495666629

Maybe there are additional issues, but I think the originally reported symptom (the "WARNING: CPU: X PID: XXXX at include/linux/rwsem.h:80 follow_pte+0xf8/0x120 on resume") is fixed by Linux kernel 6.12.

If there are additional suspend/resume problems, let's file separate Issues for them.  Thanks.

closed 2024-11-25 by aritger