NVIDIA/open-gpu-kernel-modules #724: RTX 4060: stuck at 60Hz

Kimiblock, 2024-10-29:
### NVIDIA Open GPU Kernel Modules Version

560.35.03

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.11.5-arch1-1

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 4060 Laptop GPU (UUID: GPU-8329a1c6-f6a8-288a-79e5-65313cf3b82c)

### Describe the bug

After switching from proprietary 560 to open 560 drivers, GNOME no longer shows an option to enable 240Hz output.

[image]

### To Reproduce

```
pacman -S nvidia-open
```

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/17555728/nvidia-bug-report.log.gz)

### More Info

[image]

Kimiblock, 2024-10-29:
I did a little bit of experiment and it turns out that `NVreg_RegistryDwords="RMForcePstate=0"` was the root cause. But according to [this documentation](https://docs.nvidia.com/gameworks/content/gameworkslibrary/coresdk/nvapi/group__gpupstate.html), P0 should be "best performance"...

mtijanic, 2024-10-29:
P0 is maximum performance, yes. However, please note that the parameter above is not meant for end user use. It is an internal, test only knob that was never meant to be exposed (albeit trivial to discover by analyzing the driver; it's also not meant to be hidden). I can think of several reasons why it might mess with Is Mode Possible checks.

I really can't recommend using these `RegistryDwords` knobs (especially the undocumented ones) for anything other than temporary tests. We don't QA systems with these set, so you'll likely be hitting more issues where it's unclear what the root cause is.

Kimiblock, 2024-10-29:
That’s true, but open drivers currently have stuttering issues after idle. I’m looking for ways to mitigate it.--Sincerely, Kimiblock